ros-navigation/navigation2 #4155: Can MPPI follow the path with time stamps?

Chortine, 2024-02-29:
Hi team,

I have a global path planner that outputs the points with time stamps, and the car is supposed to be a the pose at the suggested exact time. I would like to know that if MPPI is able to do such local planning with strict time critics. 

I look into the PathFollowCritcs and PathAlignCritics, however, there seems to be no items encourging 'being at the pose at time'. Maybe I've missed something. Can we extend MPPI to do that?  Tons of thanks.

SteveMacenski, 2024-02-29:
At this moment, no, **but** MPPI is pluginized for a reason, you could easily create a new PathFollower/PathAlign critic that takes into account the timing in your pose messages. The MPPI samples themselves have a time forward, velocity, and pose information, so if you path contains the timing information, you'd have all of the information required in front of you - you just need to develop the algorithm to do so.

I think this would also be a great option to provide to MPPI users in the future as well, so if you created/adapted a path following critic for using path timestamps, I'd be more than happy to merge, review, and maintain it in Nav2 itself!

Chortine, 2024-02-29:
> At this moment, no, **but** MPPI is pluginized for a reason, you could easily create a new PathFollower/PathAlign critic that takes into account the timing in your pose messages. The MPPI samples [...]

Sure! I'll work on it these days, and when it passes our case, I will submit a merge request ;)

SteveMacenski, 2024-03-27:
@Chortine any updates or problems? :smile:

Chortine, 2024-03-28:
Indeed, I'm facing some problems. I have implemented my first version of time_critic, the car now drive faster than without this critic. However, it starts zig-zag around the path, instead of nicely aligning with the path like before, specially when the curve of the path is large. I'm currently working to solve this. Do you have any suggestions? 

https://github.com/ros-planning/navigation2/assets/107395103/c708181c-bc40-459c-9817-c5015c4934e9

SteveMacenski, 2024-03-28:
Its hard to say without looking at the code and the video is corrupted so its not viewable on my side. I'd suspect you need to run the controller faster to better converge

Chortine, 2024-03-29:
Hi, 

Here is my implementation of time_critic.cpp:

```
    using namespace xt::placeholders; // NOLINT
    using xt::evaluation_strategy::immediate;

    void TimeCritic::initialize()
    {
        auto getParam = parameters_handler_->getParamGetter(name_);
        getParam(path_dt_, "path_dt", 0.1f);
        getParam(power_, "cost_power", 1);
        getParam(weight_, "cost_weight", 10.0);
        getParam(weight_1_, "cost_weight_1", 10.0);
        getParam(weight_2_, "cost_weight_2", 10.0);

        RCLCPP_INFO(
            logger_,
            "TimeCritic instantiated with %d power and %f weight",
            power_, weight_);
    }

    void TimeCritic::score(CriticData &data)
    {
        std::cout << "============ goes into TimeCritic ============" << std::endl;
        // print the time of pose stamped in data
        std::cout << "============ pose time: " << data.state.pose.header.stamp.sec << "." << data.state.pose.header.stamp.nanosec << std::endl;
        float control_dt = data.model_dt;

        utils::setPathFurthestPointIfNotSet(data);
        const size_t path_segments_count = *data.furthest_reached_path_point; // up to furthest only

        if (path_segments_count < 2) // offset_from_furthest_
        {
            return;
        }
        // std::cout << "============ [TimeCritic] ckpt1" << std::endl;

        const auto P_x = xt::view(data.path.x, xt::range(_, -1));      // path points
        const auto P_y = xt::view(data.path.y, xt::range(_, -1));      // path points
        const auto P_yaw = xt::view(data.path.yaws, xt::range(_, -1)); // path points

        std::cout << "P_x shape: " << P_x.shape(0) << ", " << P_x.shape(1) << std::endl;

        std::cout << "============ P_x, P_y: " << std::endl;
        for (unsigned int i = 1; i < P_x.shape(0); i++)
        {
            std::cout << P_x(i) << ", " << P_y(i) << ";  ";
        }
        std::cout << "============ " << std::endl;

        const size_t batch_size = data.trajectories.x.shape(0);
        const size_t time_steps = data.trajectories.x.shape(1);
        auto &&cost = xt::xtensor<float, 1>::from_shape({data.costs.shape(0)});
        // std::cout << "============ [TimeCritic] ckpt2" << std::endl;
        // Find integrated distance in the path
        std::vector<float> path_integrated_distances(path_segments_count, 0.0f);
        float dx = 0.0f, dy = 0.0f, dyaw = 0.0f;

        for (unsigned int i = 1; i != path_segments_count; i++)
        {
            dx = P_x(i) - P_x(i - 1);
            dy = P_y(i) - P_y(i - 1);
            float curr_dist = sqrtf(dx * dx + dy * dy);
            path_integrated_distances[i] = path_integrated_distances[i - 1] + curr_dist;
        }
        // std::cout << "============ [TimeCritic] ckpt3" << std::endl;
        size_t step_interval = path_dt_ / control_dt; // 0.1 / 0.05
        float Tx = 0.0f, Ty = 0.0f;
[... 112 lines cut ...]
        max_path_occupancy_ratio: 0.05
        threshold_to_consider: 0.1
        offset_from_furthest: 20
        use_path_orientations: true
      PathFollowCritic:
        cost_weight: 50.0 # 50.0
        cost_power: 1
        threshold_to_consider: 0.1
        offset_from_furthest: 20
      PathAngleCritic:
        cost_weight: 0.0 # 5.0
        cost_power: 1
        threshold_to_consider: 0.5
        forward_preference: false
      TimeCritic:
        path_dt: 0.1
        cost_weight: 0.0 # 5.0
        cost_weight_1: 50.0 # 100.0
        cost_weight_2: 0.0 # 10.0
        cost_power: 1
```

And I reupload the video, where the car zig-zag when follow the path.

https://github.com/ros-planning/navigation2/assets/107395103/054072bd-89e8-4c6c-a37a-256407f23fce

SteveMacenski, 2024-03-29:
This looks like just the path alignment critic -- what is substantively different about this from the current path align critic? Wouldn't you rather be looking at the velocities of the trajectories instead? if you have some sequence of poses in a path with a timestamp, you can find the velocity between them that enables that easily. Then the trajectories you have their velocities, so it should be an easy subtraction to find the difference and then sum those up to average for each trajectory sample :smile: 

Also, I'd recommend pulling in the latest changes I made (after 45% increase in performance PR), some of the utils that you're using were changed. I don't recall if I fixed any bugs in them, but I think at least one had a bug that was resolved (the rest are just faster now).