ros-navigation/navigation2 #4157: user-misconfiguration may cause a heap-buffer-overflow bug in nav2_amcl

GoesM, 2024-02-29:
## Bug report

**Required Info:**

- Operating System: 
  - Ubuntu 22.04
- ROS2 Version: 
  -  humble
- Version or commit hash: 
  -  the latest
- DDS implementation: 
  -  default

#### Steps to reproduce issue

launch the navigation2 within defaulted `tb3_simulation_launch.py` as following commands
```
source install/setup.bash
source /opt/ros/humble/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
ros2 launch nav2_bringup tb3_simulation_launch.py params_file:=my_nav2_params.yaml
```
there's only one difference between `my_nav2_params.yaml`  and defaulted `nav2_params.yaml`,(we wrongly set `amcl->ros_parameters->z_rand` as a very large value ), as following:

```yaml
amcl:
  ros__parameters:
    ...
    ...
    z_rand: 54861240687936886832559362511872092700743926359323919340246415125412623597951915939539289081689902927585003914562122604.0  # (such value is derived from our fuzzy testing.)
    ...
    ...
``` 
then, just wait for the navigation2 launching. 

#### Expected behavior

Processes should not crash, or at least processes should not result in **heap buffer overflow**, thereby providing the possibility for hackers to inject malicious code.

#### Actual behavior
- we would find a asan report after (steps to reproduce issue) done:
```asan
#asan报告
=================================================================
==10254==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x63000000fe00 at pc 0x556909308687 bp 0x7f2f3d49c4b0 sp 0x7f2f3d49bc80
READ of size 24 at 0x63000000fe00 thread T11
    #0 0x7f2f45b7ca7c in pf_update_resample (/home/****/ROS_ws/nav2/install/nav2_amcl/lib/libpf_lib.so+0x8a7c) (BuildId: 73e413561c1948e30ca366fe29aeef8e601b70f9)
    #1 0x7f2f4617704b in nav2_amcl::AmclNode::laserReceived(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const>) (/home/****/ROS_ws/nav2/install/nav2_amcl/lib/libamcl_core.so+0x37704b) (BuildId: fcf524e51612a7063856b255b0200fa30f1c8a67)
    #2 0x7f2f463406a9 in void std::__invoke_impl<void, void (nav2_amcl::AmclNode::*&)(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const>), nav2_amcl::AmclNode*&, std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&>(std::__invoke_memfun_deref, void (nav2_amcl::AmclNode::*&)(std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const>), nav2_amcl::AmclNode*&, std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&) (/home/****/ROS_ws/nav2/install/nav2_amcl/lib/libamcl_core.so+0x5406a9) (BuildId: fcf524e51612a7063856b255b0200fa30f1c8a67)
    #3 0x7f2f4633fe35 in message_filters::CallbackHelper1T<std::shared_ptr<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&, sensor_msgs::msg::LaserScan_<std::allocator<void> > >::call(message_filters::MessageEvent<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&, bool) (/home/****/ROS_ws/nav2/install/nav2_amcl/lib/libamcl_core.so+0x53fe35) (BuildId: fcf524e51612a7063856b255b0200fa30f1c8a67)
    #4 0x7f2f462f342b in message_filters::Signal1<sensor_msgs::msg::LaserScan_<std::allocator<void> > >::call(message_filters::MessageEvent<sensor_msgs::msg::LaserScan_<std::allocator<void> > const> const&) (/home/****/ROS_ws/nav2/install/nav2_amcl/lib/libamcl_core.so+0x4f342b) (BuildId: fcf524e51612a7063856b255b0200fa30f1c8a67)
    #5 0x7f2f46338790 in tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>::transformReadyCallback(tf2_ros::TransformStampedFuture const&, unsigned long) (/home/****/ROS_ws/nav2/install/nav2_amcl/lib/libamcl_core.so+0x538790) (BuildId: fcf524e51612a7063856b255b0200fa30f1c8a67)
    #6 0x7f2f4633cc6e in std::_Function_handler<void (tf2_ros::TransformStampedFuture const&), std::_Bind<void (tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>::* (tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>*, std::_Placeholder<1>, unsigned long))(tf2_ros::TransformStampedFuture const&, unsigned long)> >::_M_invoke(std::_Any_data const&, tf2_ros::TransformStampedFuture const&) (/home/****/ROS_ws/nav2/install/nav2_amcl/lib/libamcl_core.so+0x53cc6e) (BuildId: fcf524e51612a7063856b255b0200fa30f1c8a67)
    #7 0x7f2f47620e8d  (/opt/ros/humble/lib/libtf2_ros.so+0x4fe8d) (BuildId: 5fe744f3037178d67abb96013375a405993407b9)
    #8 0x7f2f4759c5fe in tf2::BufferCore::testTransformableRequests() (/opt/ros/humble/lib/libtf2.so+0x105fe) (BuildId: 3fd150ee62f6512f752884e3ecc762fe9ad1e1e6)
    #9 0x7f2f4759f321 in tf2::BufferCore::setTransformImpl(tf2::Transform const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::chrono::time_point<std::chrono::_V2::system_clock, std::chrono::duration<long, std::ratio<1l, 1000000000l> > >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) (/opt/ros/humble/lib/libtf2.so+0x13321) (BuildId: 3fd150ee62f6512f752884e3ecc762fe9ad1e1e6)
    #100x7f2f4759fb69 in tf2::BufferCore::setTransform(geometry_msgs::msg::TransformStamped_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool) (/opt/ros/humble/lib/libtf2.so+0x13b69) (BuildId: 3fd150ee62f6512f752884e3ecc762fe9ad1e1e6)
    #11 0x7f2f47629bb0 in tf2_ros::TransformListener::subscription_callback(std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > const>, bool) (/opt/ros/humble/lib/libtf2_ros.so+0x58bb0) (BuildId: 5fe744f3037178d67abb96013375a405993407b9)
    #12 0x7f2f47629a89 in std::_Function_handler<void (std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > const>), std::_Bind<void (tf2_ros::TransformListener::* (tf2_ros::TransformListener*, std::_Placeholder<1>, bool))(std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > const>, bool)> >::_M_invoke(std::_Any_data const&, std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > const>&&) (/opt/ros/humble/lib/libtf2_ros.so+0x58a89) (BuildId: 5fe744f3037178d67abb96013375a405993407b9)
    #13 0x7f2f4763e601 in std::__detail::__variant::__gen_vtable_impl<std::__detail::__variant::_Multi_array<std::__detail::__variant::__deduce_visit_result<void> (*)(rclcpp::AnySubscriptionCallback<tf2_msgs::msg::TFMessage_<std::allocator<void> >, std::allocator<void> >::dispatch(std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > >, rclcpp::MessageInfo const&)::'lambda'(auto&&)&&, std::variant<std::function<void (tf2_msgs::msg::TFMessage_<std::allocator<void> > const&)>, std::function<void (tf2_msgs::msg::TFMessage_<std::allocator<void> > const&, rclcpp::MessageInfo const&)>, std::function<void (rclcpp::SerializedMessage const&)>, std::function<void (rclcpp::SerializedMessage const&, rclcpp::MessageInfo const&)>, std::function<void (std::unique_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> >, std::default_delete<tf2_msgs::msg::TFMessage_<std::allocator<void> > > >)>, std::function<void (std::unique_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> >, std::default_delete<tf2_msgs::msg::TFMessage_<std::allocator<void> > > >, rclcpp::MessageInfo const&)>, std::function<void (std::unique_ptr<rclcpp::SerializedMessage, std::default_delete<rclcpp::SerializedMessage> >)>, std::function<void (std::unique_ptr<rclcpp::SerializedMessage, std::default_delete<rclcpp::SerializedMessage> >, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > const>)>, std::function<void (std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > const>, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage const>)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage const>, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > const> const&)>, std::function<void (std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > const> const&, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage const> const&)>, std::function<void (std::shared_ptr<rclcpp::SerializedMessage const> const&, rclcpp::MessageInfo const&)>, std::function<void (std::shared_ptr<tf2_msgs::msg::TFMessage_<std::allocator<void> > >)>, [... cut ...]
However, bugs like **heap-buffer-overflow** may pose security issues within system software. Attackers can exploit heap buffer overflow to overwrite function pointers or program counters, thereby injecting malicious code into the program for execution, escalating to more serious security concerns.

a suggestion that may be helpful:
- necessary parameter checks, which may pose security risks, should be implemented. shall we add a param value check for `z_rand` in `nav2_amcl` to avoid such **heap-buffer-overflow** bug?

SteveMacenski, 2024-02-29:
That's an insane number, I wouldn't expect anything else to happen. 

Added to the #4005 - I think we discussed last time its a low-priority task to protect every parameter's potential reasonable bounds, since that would take weeks and you're only finding these by setting insane values. If you find more in the future, feel free to append to your ticket (#4005).

closed 2024-02-29 by SteveMacenski

GoesM, 2024-03-01:
> feel free to append to your ticket(#4005)

no problem, I got it  ^_^

SteveMacenski, 2024-03-01:
I will say that if this interests you to address, something I thought of was adding a Nav2 util eq of ‘get_parameter’ whos arguments can include min-max ranges for ints & doubles and throws if not met. That would be a relatively time consuming, but feasible for someone that wants it, action. 

Then the hard work comes to populating all those ranges. I figure one node at a time, once per week is a good way to break it into managable chunks. 

Dynamic parameter updates are another issue. Perhaps we could refactor the params for all nodes/servers into their own param handler util (some have already) so all that range checking is handled in a separate file to keep the algorithm implementations cleaner. If we wanted to go even a bridge further, we could change all member params into rclcpp::Parameters that have valid ranges in their descriptions.

GoesM, 2024-03-01:
We have some concerns about this security issue, while it is also a relatively low priority issue for us. Our submission of such issues mainly aims to provide a reminder and record of such issue for the navigation2 stack. Should we discover similar parameter issues in the future, we will add them directly to issue 4005 rather than creating a new one.

>   adding a Nav2 util eq of ‘get_parameter’ whos arguments can include min-max ranges for ints & doubles and throws if not met

it seems feasible, do you need us to open a PR for it?

SteveMacenski, 2024-03-01:
I try not do to do things half baked, since there may be some functional problems that require reworking the problem, but I was thinking something like 

```
rclcpp::Parameter get_parameter<TypeT>(Node & node, string param_name, TypeT min, TypeT max)
{
  rclcpp::Parameter param;
  if (node->get_parameter(param_name, param)) {
    if (param >= min && min <= max) {
      return param;
    }
    // throw exception, out of bounds 
  }
  // throw exception, did not obtain
}
```

Then the TypeT of this function have overrides for the different versions. The double/int use something like this, but the string/bool just do the normal get parameter logic. It would be expanded with the double/int arrays to also check those parameters' values in the array AND also check the array size. 

A couple of open questions:
- Which rclcpp get_parameters API makes most sense: the bool check or let it throw itself one if it doesn't exist?
- Even if we replace inline with this everywhere, how do we want to handle the dynamic parameters callback function

GoesM, 2024-03-04:
> I try not do to do things half baked
So do I, while It seems I need some time to delve deeper into this section of code and the API. I'd come back here to discuss further once I have more matured ideas.

GoesM, 2024-03-22:
I spent some time studying the code in the **ros2/rclcpp** repository, and I noticed that the current code for `get_parameter()` in rclcpp doesn't seem to provide an interface for setting valid boundaries such as maximum and minimum values. Do you suggest that I propose such a design to rclcpp first and then apply it to nav2 later?

SteveMacenski, 2024-03-22:
The parameter description has that!