ros-navigation/navigation2 #4168: Dynamic footprint support in collision monitor (humble)

ravi-kumaaar, 2024-03-07:
https://github.com/ros-planning/navigation2/issues/3196
Based on this thread it seems that it was decided to use topic interface to change the polygon footprint of the collision monitor. Which seems to be merged into main branch, however not backported to humble.

https://github.com/ros-planning/navigation2/pull/4056
Based on this PR someone tried to add collision monitor state topic to humble but it couldn't because of ABI compatibility.

Does this mean that dynamic footprint feature will not be available for humble?

SteveMacenski, 2024-03-07:
It is not possible to backport, it breaks ABI/API stability policies. That is available in Iron, Rolling (Jazzy) and newer.

robotcopper, 2024-04-10:
> It is not possible to backport, it breaks ABI/API stability policies. That is available in Iron, Rolling (Jazzy) and newer.

Hi, 
I haven't looked at the velocity_polygon code in detail yet, but I'm having trouble understanding why it wouldn't be portable to humble.
Indeed, in principle it doesn't require anything more than retrieving cmd_vel_raw and sensor data, or am I being too naive?

MichaelT64, 2024-04-11:
Hi, @robotcopper ,
I also want to use that in Humble and had the idea to make the main branch usable for Humble with some conditional compiling (CPP) and execution (Python). My fork is here [https://github.com/MichaelT64/navigation2/tree/rolling-humble](https://github.com/MichaelT64/navigation2/tree/rolling-humble). It builds and I can use the CollisionMonitor (my highest priority at the moment). Due to the lack of time and knowledge I a was not able to do a deeper test of navigation yet. But may be it is worth a try and I will be happy to hear about your experience with it.

robotcopper, 2024-04-11:
Thank you @MichaelT64 for reaching me out, I'll look into it and get back to you as soon as possible.