ros-navigation/navigation2 #4169: Source build failing due to missing packages

stinkyElias, 2024-03-07:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04, Docker version 25.0.3, build 4debf41
- ROS2 Version:
  - Rolling binaries
- Version or commit hash:
  - From source, main branch: 26363716536425ff6479610c18639c853e7537be
- DDS implementation:
  - Fast DDS (default)

#### Steps to reproduce issue
I'm building a Docker image from this Dockerfile https://github.com/athackst/dockerfiles/blob/main/ros2/rolling.Dockerfile and adding a new layer with nav2:

```
FROM gazebo AS nav2

ENV DEBIAN_FRONTEND=noninteractive

WORKDIR /home/ros/ros2_ws
RUN mkdir -p /home/ros/ros2_ws/src \
    && git clone https://github.com/ros-planning/navigation2.git --branch main ./src/navigation2 \
    && apt-get update && apt-get upgrade -y --no-install-recommends \
    && rosdep update \
    && rosdep install -y --ignore-src --from-paths src -r \
    && . /opt/ros/rolling/setup.sh \
    && colcon build --symlink-install

ENV DEBIAN_FRONTEND=
```
Building image...

```
docker build -t nav2:test .
```

#### Expected behavior
All rosdep keys resolved, no building error from either colcon or docker.

#### Actual behavior
Image building failing on ``` colcon build --symlink-install ``` due to missing packages.

Docker error
```
ERROR: failed to solve: process "/bin/sh -c mkdir -p /home/stinky/ros2_ws/src     && git clone https://github.com/ros-planning/navigation2.git --branch main ./src/navigation2     && apt-get update && apt-get upgrade -y --no-install-recommends     && rosdep update     && rosdep install -y --ignore-src --from-paths src -r     && . /opt/ros/rolling/setup.sh     && colcon build --symlink-install" did not complete successfully: exit code: 1
```

CMake error:
```
CMake Warning at CMakeLists.txt:11 (find_package):
By not providing "Findgeographic_msgs.cmake" in CMAKE_MODULE_PATH this
project has asked CMake to find a package configuration file provided by
"geographic_msgs", but CMake did not find one.
 
Could not find a package configuration file provided by "geographic_msgs"
with any of the following names:
 
geographic_msgsConfig.cmake
geographic_msgs-config.cmake

Add the installation prefix of "geographic_msgs" to CMAKE_PREFIX_PATH or
set "geographic_msgs_DIR" to a directory containing one of the above files.
If "geographic_msgs" provides a separate development package or SDK, be
sure it has been installed.

CMake Error at /opt/ros/rolling/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:192 (message):
rosidl_generate_interfaces() the passed dependency 'geographic_msgs' has
not been found before using find_package()
```

rosdep error:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
nav2_bringup: No definition of [turtlebot3_gazebo] for OS version [jammy]
nav2_msgs: No definition of [geographic_msgs] for OS version [jammy]
nav2_lifecycle_manager: No definition of [bondcpp] for OS version [jammy]
nav2_bt_navigator: No definition of [behaviortree_cpp] for OS version [jammy]
nav2_smac_planner: No definition of [ompl] for OS version [jammy]
nav2_behavior_tree: No definition of [behaviortree_cpp] for OS version [jammy]
nav2_util: No definition of [bond] for OS version [jammy]
nav2_waypoint_follower: No definition of [geographic_msgs] for OS version [jammy]

```
#### Additional information
I tried to install ```geographic_msgs``` from source and the error related to that package disappeared. However, I built a similar image just a couple of days ago that worked fine and this image shouldn't be any different. Given that I build nav2 the way it says in the docs, I find it weird that I have to install a bunch of other packages as well, especially when it worked some days ago.

I also tried to build from image ```osrf/ros:rolling-desktop-full``` and ```osrf/ros:rolling-desktop-full-jammy``` but getting the same error.

Is there an issue with the package or am I doing something wrong?

Any help would be appreciated. Thanks:)

SteveMacenski, 2024-03-07:
> However, I built a similar image just a couple of days ago that worked fine and this image shouldn't be any different

It actually is different unfortunately. You've come at a bad time - there's a massive migration underway for Rolling over to `22.04` so rosdep, binaries, etc are being migrated so things are in a bit of a wonky state. This is a transient issue out of our control until that migration is complete then should stabilize back with 22.04 for Rolling.

stinkyElias, 2024-03-07:
Thanks for letting me know. Do you know when the migration(s) is finished? If not, anywhere I can check for updates?

SteveMacenski, 2024-03-07:
In a week or two, not something in my control and is a topic of conversation with updates on Discourse :-) https://discourse.ros.org/t/preparing-ros-2-rolling-for-the-transition-to-ubuntu-24-04/35673/8

stinkyElias, 2024-03-07:
Thanks so much! I'll wait and try again at a later point:)

closed 2024-03-14 by SteveMacenski