ros-navigation/navigation2 #4172: MPPI generates undesirable paths

atinfinity, 2024-03-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
```shell
$ dpkg-query --show "ros-$ROS_DISTRO-navigation2"
ros-humble-navigation2  1.1.13-1jammy.20240217.105800
```
```shell
$ dpkg-query --show "ros-$ROS_DISTRO-navigation2"
ros-humble-navigation2  1.1.13-1jammy.20240217.105800
user@f5f26d76325f:~$ dpkg-query --show "ros-$ROS_DISTRO-nav2-*"
ros-humble-nav2-amcl    1.1.13-1jammy.20240217.085112
ros-humble-nav2-behavior-tree   1.1.13-1jammy.20240217.085125
ros-humble-nav2-behaviors       1.1.13-1jammy.20240217.093702
ros-humble-nav2-bringup 1.1.13-1jammy.20240217.110614
ros-humble-nav2-bt-navigator    1.1.13-1jammy.20240217.093705
ros-humble-nav2-collision-monitor       1.1.13-1jammy.20240217.092320
ros-humble-nav2-common  1.1.13-1jammy.20240217.071750
ros-humble-nav2-constrained-smoother    1.1.13-1jammy.20240217.092529
ros-humble-nav2-controller      1.1.13-1jammy.20240217.092532
ros-humble-nav2-core    1.1.13-1jammy.20240217.092311
ros-humble-nav2-costmap-2d      1.1.13-1jammy.20240217.085604
ros-humble-nav2-dwb-controller  1.1.13-1jammy.20240217.094145
ros-humble-nav2-lifecycle-manager       1.1.13-1jammy.20240217.085120
ros-humble-nav2-map-server      1.1.13-1jammy.20240217.084621
ros-humble-nav2-mppi-controller 1.1.13-1jammy.20240217.092539
ros-humble-nav2-msgs    1.1.13-1jammy.20240217.075125
ros-humble-nav2-navfn-planner   1.1.13-1jammy.20240217.092818
ros-humble-nav2-planner 1.1.13-1jammy.20240217.092854
ros-humble-nav2-regulated-pure-pursuit-controller       1.1.13-1jammy.20240217.092859
ros-humble-nav2-rotation-shim-controller        1.1.13-1jammy.20240217.093708
ros-humble-nav2-rviz-plugins    1.1.13-1jammy.20240217.104951
ros-humble-nav2-simple-commander        1.1.13-1jammy.20240217.075845
ros-humble-nav2-smac-planner    1.1.13-1jammy.20240217.092916
ros-humble-nav2-smoother        1.1.13-1jammy.20240217.092921
ros-humble-nav2-theta-star-planner      1.1.13-1jammy.20240217.093253
ros-humble-nav2-util    1.1.13-1jammy.20240217.083804
ros-humble-nav2-velocity-smoother       1.1.13-1jammy.20240217.085122
ros-humble-nav2-voxel-grid      1.1.13-1jammy.20240217.075307
ros-humble-nav2-waypoint-follower       1.1.13-1jammy.20240217.093251
```
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

I used <https://github.com/atinfinity/megarover_samples_ros2>.

```shell
$ mkdir -p ~/dev_ws/src
$ cd ~/dev_ws/src
$ git clone https://github.com/atinfinity/megarover_samples_ros2.git
$ cd ..
$ rosdep install -y -i --from-paths src/megarover_samples_ros2
$ colcon build --symlink-install
$ source ~/dev_ws/install/setup.bash
$ ros2 launch megarover_samples_ros2 vmegarover_with_sample_world.launch.py
$ ros2 launch megarover_samples_ros2 vmegarover_navigation.launch.py map:=$HOME/maps/vmegarover_samplemap.yaml
```

And, I set a goal in front of the robot via RViz2.

#### Expected behavior

I expect the robot to go straight.

#### Actual behavior

In fact, the robot moves around in a slight curve.
I referred to <https://navigation.ros.org/configuration/packages/configuring-mppic.html#example>.

[image]

#### Additional information

I tried to use DWB. The robot to go straight.

[image]

SteveMacenski, 2024-03-08:
I believe I know the root cause and this was actually fixed some weeks ago but just hasn't made it out to the Humble binaries yet. 

If you apply this patch https://github.com/ros-planning/navigation2/commit/b489f5b79baea6c9f6b91ecb23c5a2725317c140 does that fix your issue? If so, it'll be out in binaries in the next sync.

atinfinity, 2024-03-09:
@SteveMacenski Thank you for your help! After applying this patch, this problem was resolved.

closed 2024-03-09 by atinfinity

SteveMacenski, 2024-03-09:
My favorite kind of problem: the one I already solved 🙂