ros-navigation/navigation2 #4173: Mppi plugin running

LudovicaDanovaro, 2024-03-08:
Hello,
I would like to run your `nav2_mppi_controller` but I'm having problems on it.
Once I downloaded your navigation2 repository, I tried to modify the file `nav2_params.yaml` as indicated in the example here: https://navigation.ros.org/configuration/packages/configuring-mppic.html: I changed the lines after `FollowPath:` with the corresponding ones in my yaml file previously downloaded with the installation of nav2 (so it's path is `/opt/ros/foxy/share/nav2_bringup/params/av2_params.yaml`). Then, when I tried to run the simulation with the command `ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False`  it opens gazebo and rviz but I can't see the particle cloud mentioned in the getting started section as depicted in the screenshot below (while I could see it without changing the yaml file for the mppi plugin).

How can I solve this issues? What do I have to modify after downloading your navigation2 repository to run the mppi controller?
I'm using Ubuntu 20.04 with Ros2 foxy installed.

Thank you!

[image]

SteveMacenski, 2024-03-08:
Ask on robotics stack exchange or look through your logs, there's clearly something not loading properly that I'm sure the logs will point you to.

closed 2024-03-08 by SteveMacenski

donymorph, 2024-10-23:
you are using ros2 foxy you need to change to foxy-devel branch. but anyway on foxy there is no MPPI controller. you need to upgrade your ros2 from foxy to any other version (galactic, rolling, humble)