ros-navigation/navigation2 #4183: https://robotics.stackexchange.com/questions/109985/why-robot-not-moving-to-a-goal-pose-in-nav2-after-map-is-created-and-pose-is-est

astronaut71, 2024-03-15:
I have differential drive robot with Roboteq controller. After successfully created the map with slam_toolbox and localise the robot I tried to use the Nav2 to navigate the robot. Im using ROS2 humble. The controller try to navigate the robot to the goal pose but is stuck.

Here the launch file for mapping and localisation

```
from launch import LaunchDescription
from launch_ros.actions import Node
from launch.actions import IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import ThisLaunchFileDir, LaunchConfiguration, Command
from ament_index_python.packages import get_package_share_directory
import launch_ros
import os

def generate_launch_description():
    pkg_share = launch_ros.substitutions.FindPackageShare(package='amr_bot').find('amr_bot')
    #slam_params_file_path = os.path.join(get_package_share_directory('your_package_name'), 'config', 'mapper_params_online_async.yaml')

# Static Transform Publishers
    static_tf_base_to_laser = Node(
        package='tf2_ros',
        executable='static_transform_publisher',
        arguments=['-0.02', '0.02', '-0.02', '0', '0', '0', 'base_link', 'laser_link'],
    )

    static_tf_base_to_odom = Node(
        package='tf2_ros',
        executable='static_transform_publisher',
        arguments=['-0.002', '0', '-0.002', '0', '0', '0', 'base_link', 'odom'],
    )

    static_tf_base_to_footprint = Node(
        package='tf2_ros',
        executable='static_transform_publisher',
        arguments=['0', '0', '0', '0', '0', '0', 'base_link', 'base_footprint'],
    )

    camera2base_node = Node(
        package = 'tf2_ros',
        executable = 'static_transform_publisher',
        arguments = ['-0.04','0', '0', '0', '0', '0', 'base_link', 'camera_link'],
    )

    #ros2 run realsense2_camera realsense2_camera_node
    #rs_camera_node = Node(
        #package = 'realsense2_camera',
        #executable = 'realsense2_camera_node',
        #using unite_imu_method is important, or the imu messages will be separated
        #arguments = {'align_depth': True, 'linear_accel_cov': '1.0', 'unite_imu_method':'2' ,
        #'initial_reset':'true', 'enable_gyro':'true', 'enable_accel':'true'},
    #)

    #include_realsense2_camera = IncludeLaunchDescription(
        #PythonLaunchDescriptionSource(os.path.join(
        #get_package_share_directory('realsense2_camera'),
        #'launch',
        #'rs_launch.py' )),
        #launch_arguments={
            #'align_depth.enable': 'true',
            #'linear_accel_cov': '1.0',
            #'unite_imu_method': '2', 'initial_reset':'true', 'enable_gyro':'true', 'enable_accel':'true'

            # 'filters': 'pointcloud',
        #}.items()
    #)

    include_realsense2_camera = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(os.path.join(
        get_package_share_directory('realsense2_camera'),
        'launch',
[... 202 lines cut ...]
            #rgbd_odometry_node,
            #rgbd_odom_node,
            robot_localization_node,
            #gmapping_node,

            icp_odometry_node,
            points_xyzrgb_node,
            camera2base_node,
            #rs_camera_node,
            include_realsense2_camera,
            slam_toolbox_launch,
            slam_toolbox_node,
            

Then I run the nav2 with

  `  ros2 launch nav2_bringup bringup_launch.py use_sim_time:=False autostart:=False map:=/map_15_3_1.yaml params_file:=nav2_params.yaml`

and start the rviz2

Here the RVIZ 
![nav2](https://github.com/ros-planning/navigation2/assets/2849083/82b70dab-c940-457a-9650-aafc52d467c2)

```
    ])

alanxuefei, 2024-03-15:
**The `odom` to `base_link` Transformation**
Dynamically managed by the odometry system. For example, nodes like `imu_to_odom_node` play a significant role by providing essential odometry data.

**The `map` to `odom` Transformation**
Updated by the localization system, with nodes such as `robot_localization_node` serving as prime examples. This process involves calculating the transformation from `map` to `base_link` and then subtracting `odom` to `base_link` to update `map` to `odom`.

**The Static `odom` to `base_link` Transformation**
The provided code snippet initializes a static transformation between the `base_link` and `odom` frames. This static transformation can cause the robot to remain stationary because the dynamic transformations published by the odometry node are always overwritten by this static value.

```
static_tf_base_to_odom = Node(
    package='tf2_ros',
    executable='static_transform_publisher',
    arguments=['-0.002', '0', '-0.002', '0', '0', '0', 'base_link', 'odom'],
)
```

SteveMacenski, 2024-03-15:
Do not cross post with Stack Exchange https://robotics.stackexchange.com/questions/109985/why-robot-not-moving-to-a-goal-pose-in-nav2-after-map-is-created-and-pose-is-est

closed 2024-03-15 by SteveMacenski

astronaut71, 2024-03-16:
so means I dont need static_tf_base_to_odom node? and what about The odom -> base_link Transformation and The map -> odom Transformation??

astronaut71, 2024-03-16:
@SteveMacenski please dont close this question

alanxuefei, 2024-03-16:
@astronaut71 We can move the discussion to robotics.stackexchange.com.