ros-navigation/navigation2 #4187: Can't launch nav2_bringup. Error: No such file or directory: ''

arad2456, 2024-03-18:
I want to call nav2_bringup in my ros2 launch file. When I make the call I get the error message:

`
user:~/ros2_ws$ ros2 launch slam_train slam_algorithm.launch.py
[INFO] [launch]: All log files can be found below /home/arad2456/.ros/log/2024-03-18-02-17-32-921054-LAPTOP-HALVVF48-24760
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): [Errno 2] No such file or directory: ''
`

When I run it in debug I get:
`
user:~/ros2_ws$ ros2 launch --debug slam_train slam_algorithm.launch.py
[DEBUG] [launch.launch_context]: emitting event synchronously: 'launch.events.IncludeLaunchDescription'
[INFO] [launch]: All log files can be found below /home/arad2456/.ros/log/2024-03-18-02-17-01-363420-LAPTOP-HALVVF48-24214
[INFO] [launch]: Default logging verbosity is set to DEBUG
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f4afb07d270>'
[DEBUG] [launch]: processing event: '<launch.events.include_launch_description.IncludeLaunchDescription object at 0x7f4afb07d270>' ✓ '<launch.event_handlers.on_include_launch_description.OnIncludeLaunchDescription object at 0x7f4afb07cfa0>'
Executing <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:228> exception=FileNotFoundError(2, 'No such file or directory') created at /opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py:318> took 0.903 seconds
[DEBUG] [launch]: An exception was raised in an async action/event
[DEBUG] [launch]: Traceback (most recent call last):
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 336, in run_async
    raise completed_tasks_exceptions[0]
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 230, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_service.py", line 250, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  [Previous line repeated 4 more times]
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/node.py", line 490, in execute
    self._perform_substitutions(context)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/actions/node.py", line 445, in _perform_substitutions
    evaluated_parameters = evaluate_parameters(context, self.__parameters)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/utilities/evaluate_parameters.py", line 159, in evaluate_parameters
    output_params.append(param.evaluate(context))
  File "/opt/ros/humble/lib/python3.10/site-packages/launch_ros/parameter_descriptions.py", line 235, in evaluate
    param_file = perform_substitutions(context, self.__param_file)
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in perform_substitutions
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/utilities/perform_substitutions_impl.py", line 26, in <listcomp>
    return ''.join([context.perform_substitution(sub) for sub in subs])
  File "/opt/ros/humble/lib/python3.10/site-packages/launch/launch_context.py", line 240, in perform_substitution
    return substitution.perform(self)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/nav2_common/launch/rewritten_yaml.py", line 88, in perform
    data = yaml.safe_load(open(yaml_filename, 'r'))
FileNotFoundError: [Errno 2] No such file or directory: ''

[ERROR] [launch]: Caught exception in launch (see debug for traceback): [Errno 2] No such file or directory: ''
[DEBUG] [launch.launch_context]: emitting event: 'launch.events.Shutdown'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f4afccc3430>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f4afccc3430>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f4afade3d00>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f4afccc3430>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f4afac4fc10>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f4afccc3430>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f4afaad0520>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f4afccc3430>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f4afaad0be0>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f4afccc3430>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f4afac72c20>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f4afccc3430>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f4afaad2320>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f4afccc3430>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f4afae0bb50>'
[DEBUG] [launch]: processing event: '<launch.events.shutdown.Shutdown object at 0x7f4afccc3430>' ✓ '<launch.event_handlers.on_shutdown.OnShutdown object at 0x7f4afb07f340>'
`

My launch file is:
`
[... 84 lines cut ...]
    )

    
    nav2_toolbox_dir = os.path.join(get_package_share_directory('nav2_bringup'), 'launch')
 
    nav2_toolbox_launch = IncludeLaunchDescription(PythonLaunchDescriptionSource(os.path.join(nav2_toolbox_dir, 'navigation_launch.py')),
        launch_arguments={'use_sim_time': LaunchConfiguration('use_sim_time')}.items()
    )

    ld = LaunchDescription()

    ld.add_action(declare_use_sim_time)
    ld.add_action(declare_model_urdf)
    ld.add_action(gzserver_cmd)
    ld.add_action(gzclient_cmd)
    ld.add_action(robot_state_publisher_node)
    ld.add_action(joint_state_publisher_node)
    ld.add_action(spawn_entity)
    ld.add_action(declare_rviz_config)
    ld.add_action(rviz_node)
    ld.add_action(slam_toolbox_launch)
    ld.add_action(nav2_toolbox_launch)

    return ld
`

SteveMacenski, 2024-03-18:
Check out the standard inputs to the navigation launch file from our bringup package: https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/launch/bringup_launch.py#L183

arad2456, 2024-03-19:
Thanks! It works now.

SteveMacenski, 2024-03-19:
No problemo

closed 2024-03-19 by SteveMacenski