ros-navigation/navigation2 #4190: [AMCL] pf_z's default value seems wrong and is assigned directly into pop_z

Mygao, 2024-03-18:
https://github.com/ros-planning/navigation2/blob/db974ea91b1eb01d7abf094aad511c0f2306d55f/nav2_amcl/src/amcl_node.cpp#L160

Above, `pf_z` value is set to 0.99 and this is likely set with probability in mind.

Eventually `pf_z` value goes to pop_z value in pf.c and regarding the paper this value is quantile value: for 99% probability, appropriate value should be around 3(this is set right in pf_alloc function: `pop_err` as 0.01 and `pop_z` as 3).

I think default value should be 3.

SteveMacenski, 2024-03-18:
I yield to @mikeferguson on this one since I know it’s more topical for him, but in general I’m not in love with changing defaults inline in the code after all these years. Changing defaults in our provided configurations in nav2 bringup and the documentation is definitely possible though!

If he says you’re right and it’s a better global value, we can update it in the code. If he thinks it’s right but not objectively a better global value, we can update it in the docs and bringup so folks can leverage it.

SteveMacenski, 2024-07-03:
@nahueespinosa @hidmic I see that you have `3.0` in beluga. I assume that one of you have spent some time in the guts of AMCL recently for Beluga, can you verify that we should change this here too?

Also looking forward to some localization metrics for Beluga - I'd love to deprecate AMCL and have a new integration with Beluga if we can show it works better/faster/more reliably over long terms :wink:

mikeferguson, 2024-07-03:
Ok, so this latest comment reminded me to go look at this - my findings:

 - Original KLD paper and the blue book (Thrun et al) both agree that pop_z (as we call it) should be the "upper 1-d quantile of the standard normal distribution). Blue book further states that "values for pop_z for typical values of d are readily available in standard statistical tables".
 - Statistical table says 99 percentile should be ~2.33 (which is a bit lower than the 3 that is hard coded into the pf_alloc function or in beluga).

To me, that means a higher value for pop_z would be "more correct". It would also cause the KLD resampling to keep more particles (rather than less) - but just barely (running a few random selections for k, it looks like it is at most about 10-15% more particles). It is probably a safe change for all but the most resource constrained systems, but I also don't expect that it will have much, if any, noticeable change in the actual localization performance.

SteveMacenski, 2024-07-03:
In that case, I think we should err on the side of no change if its not going to make a big difference in exchange with changing the default behavior between distributions. Users can always change the parameters to their likings! 

Though, from this discussion, I made a note in the Configuration Guide for AMCL that 2.33 is 99% for folks to be aware of. I think that's a helpful note and closes of the intent of this ticket.

Thanks @mikeferguson !

hidmic, 2024-07-03:
It's 2.33 for 99% and 3.0 for 99.7%. The difference is likely negligible, as @mikeferguson says.