ros-navigation/navigation2 #4194: Inquiry on Plugin Naming Conventions in nav2_params.yaml

alanxuefei, 2024-03-19:
## Bug report

The [nav2_params.yaml](https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/params/nav2_params.yaml) file has two naming styles for plugins, specifically the usage of `::` and `/`.  
If the name is not declared in plugin.xml, the type assumes the role of the default name, adopting the `:: `style.
Should we standardize them? 

```
amcl:
  ros__parameters:
    robot_model_type: "nav2_amcl::DifferentialMotionModel"

bt_navigator:
  ros__parameters:
    navigate_to_pose:
      plugin: "nav2_bt_navigator/NavigateToPoseNavigator"
    navigate_through_poses:
      plugin: "nav2_bt_navigator/NavigateThroughPosesNavigator"

controller_server:
  ros__parameters:
    progress_checker:
      plugin: "nav2_controller::SimpleProgressChecker"
    general_goal_checker:
      plugin: "nav2_controller::SimpleGoalChecker"
    FollowPath:
      plugin: "dwb_core::DWBLocalPlanner"

local_costmap:
  local_costmap:
    ros__parameters:
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"
      voxel_layer:
        plugin: "nav2_costmap_2d::VoxelLayer"
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"

global_costmap:
  global_costmap:
    ros__parameters:
      obstacle_layer:
        plugin: "nav2_costmap_2d::ObstacleLayer"
      static_layer:
        plugin: "nav2_costmap_2d::StaticLayer"
      inflation_layer:
        plugin: "nav2_costmap_2d::InflationLayer"

planner_server:
  ros__parameters:
    GridBased:
      plugin: "nav2_navfn_planner/NavfnPlanner"

smoother_server:
  ros__parameters:
    simple_smoother:
      plugin: "nav2_smoother::SimpleSmoother"

behavior_server:
  ros__parameters:
    spin:
      plugin: "nav2_behaviors/Spin"
    backup:
      plugin: "nav2_behaviors/BackUp"
    drive_on_heading:
      plugin: "nav2_behaviors/DriveOnHeading"
    wait:
      plugin: "nav2_behaviors/Wait"
    assisted_teleop:
      plugin: "nav2_behaviors/AssistedTeleop"

waypoint_follower:
  ros__parameters:
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
```

alanxuefei, 2024-03-19:
Sure, will refactor them using double colons within this week.

SteveMacenski, 2024-04-02:
merged

closed 2024-04-02 by SteveMacenski