ros-navigation/navigation2 #4195: [Discussion] Can MPPI follow the global path that has multiple segments, each represents forward or reversing movement.

Chortine, 2024-03-19:
Hi Team,

I have a global path planner that outputs a global path, which is divided into multple segments, and between each segement, the car alternating its moving direction: either moving forward or moving reverse. My car model is ackermann car, and the reverse moving segment is not trival for the car to arrive at the target pose.

Given this context, my question is, can the current implementation of MPPI controller follow a global path like this? If not, what will be the best way to modify it to achieve this?

Thank you very much.

SteveMacenski, 2024-03-19:
That is sufficiently vague that it’s hard for me to say anything about anything with confidence 🙃

But yes, MPPI is being used and has been shown in Guillaume’s ROSCon 2023 talk on bidirectionality in Nav2 to be able to change direction with reversing segments of paths from the Smac Planner. The important path attribute is that you set the orientation vectors in the path so that it’s decipherable that you want an explicit change in orientation for direction of travel. 

MPPI’s critics then have settings where it’ll pick up on those and use them in their behavior scoring. I’d highly recommend @doisyg ’s talk: https://vimeo.com/879000809/b32f8ebec4

Chortine, 2024-03-21:
Hi that's exactly what I want! Here is what my global path looks like:
[image]

I set orientation in path and set `PathAlignCritic.use_path_orientations = true`, and it works!
Especially, the built-in removePosesAfterFirstInversion() function is very useful!

SteveMacenski, 2024-03-21:
Great! Sounds like its solved?

closed 2024-03-21 by SteveMacenski