ros-navigation/navigation2 #4198: [collision monitor] Transient local durability for subscription to polygon topic

anaelle-sw, 2024-03-19:
## Feature request

#### Feature description
The `collision_monitor` node can subscribe to topics defined in parameter `<polygon_name>.polygon_sub_topic`. When receiving a new message on this topic, the associated polygon is updated. It seems a reasonable design that this topic only published on change. In this case, the subscriber to this topic should have transient local durability QoS, in order to be sure to catch the first message published on topic.

#### Implementation considerations
The `collision_monitor` subscriptions to topics `<polygon_name>.polygon_sub_topic` could be changed to have a transient local durability QoS.
But this change can break QoS compatibility. If the publisher has a volatile durability, the `collision_monitor` won't receive the messages (if the subscription has transient local durability).

SteveMacenski, 2024-03-20:
> The collision_monitor subscriptions to topics <polygon_name>.polygon_sub_topic could be changed to have a transient local durability QoS.

I think adding a parameter setting option for this seems sensible and in-line with what we do in other parts of the system. Please submit a PR!

SteveMacenski, 2024-03-22:
PR open to be merged shortly

closed 2024-03-22 by SteveMacenski