ros-navigation/navigation2 #4202: cloned_multi_tb3_simulation, slam:=true

iOzcan1, 2024-03-21:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble

- DDS implementation:
  - cyclone

#### Steps to reproduce issue
I haven't changed anything. Installed nav2 packages including nav2_bringup and run the following command on the terminal:

ros2 launch nav2_bringup cloned_multi_tb3_simulation_launch.py robots:="robot1={x: 1.0, y: 1.0, yaw: 1.5707}; robot2={x: 1.0, y: 1.0, yaw: 1.5707}" headless:=False slam:=True

#### Expected behavior

I was expecting to see both of the robots creating their own map.

#### Actual behavior

There is no map creation at all. respective topics and nodes with namespaces "robot1" and "robot2" can be seen. When I check the rqt_graph:
[image]
I can see that both of the slam toolbox nodes subscribes to scan topic and not <robot_name>/scan topic. it should not be like this, right?

SteveMacenski, 2024-03-21:
Correct, the scan topic needs to be remapped or the slam toolbox sessions go under namespaces (https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/launch/slam_launch.py#L31). The latter makes more sense so that the SLAM Toolbox instances are not colliding in their names, not just topics. Though, we give `/scan` as a globally rooted name, so I think remapping in that launch file is also necessary https://github.com/SteveMacenski/slam_toolbox/blob/ros2/src/slam_toolbox_common.cpp#L354

So:
- Remap `/scan` to `namespace/scan`
- Set the namespace for the slam session to `namespace`

iOzcan1, 2024-03-22:
Sure! I'll need it for my thesis,  am happy to contribute. Thank you for your quick reply!

SteveMacenski, 2024-04-23:
merging soon

closed 2024-04-23 by SteveMacenski