ros-navigation/navigation2 #4217: [mppi] Improve testability to explore parameters for Ackermann steered Polaris

aosmw, 2024-03-26:
## Feature request

#### Feature description
Improve testability of nav2_mppi_controller to allow exploration of parameters under unit test like conditions.
In other words - help build intuition for the affect of parameter changes by sweeping over parameters of interest
and calculating the result of the generated commands without the heavy weight of a full simulation system.

This requires the ability to fully reset internal state, control the random noise seed, tweak critic parameters, apply a simulation function to roughly simulate control system lag/inertia.

I am particularly seeking to gain intuition on what parameters/critics costs/costmap settings are required to get the nav2_mppi_controller to output cmd_vel that reliably approach a target speed of 3.5m/s  for an Ackermann steered vehicle (turning radius 2.75m).

aosmw, 2024-03-27:
Maybe a "SpeedCritic" will be best way to explicitly desire trajectories/cmds that lead to a target speed... IOW vx_max?

SteveMacenski, 2024-03-27:
I'm not entirely sure how to read this ticket's request. If you'd like to make some kind of system for auto-tuning or testing, I'm all for it, but that is going to be something that probably makes sense as a separate repository if its going to be extremely involved. 

> Maybe a "SpeedCritic" will be best way to explicitly desire trajectories/cmds that lead to a target speed... IOW vx_max?

Sure, its plugin based for a reason, you can add any plugins your like that optimize for what's important to you. I did remark on that a bit in your other ticket #4219

SteveMacenski, 2024-04-22:
Hey - any updates / thoughts here?

SteveMacenski, 2024-05-03:
Closing without update

closed 2024-05-03 by SteveMacenski