ros-navigation/navigation2 #4219: [mppi] Add SpeedCritic to favour trajectories that create velocities closer to target speed.

aosmw, 2024-03-27:
## Feature request

#### Feature description
A SpeedCritic function 

#### Implementation considerations
* Target speed depends on speedLimit not just vx_max
* Experiment with noise generator to generate a normal distribution with a mean and a bias.   For the Akerman steering case, I have a gut feeling that the normal distribution around 0 leads to a kind of two steps forward 1 step back behaviour.  With a positive bias more noise would lead to options that increase velocity.  This may totally break the maths in Optimizer::updateControlSequence().  It may require new calculations to take the introduced noise bias it into account (My maths expertise breaks down at this point).

SteveMacenski, 2024-03-27:
I originally had such a critic but found it deeply problematic to enable reversing or even stopping at the end of paths. I suppose we could have the same kind of `threshold_to_consider` where its rejected when we're approaching the goal, but slowing down and stopping to reverse in cases where that's necessary is virtually impossible when you are incentivizing it to go full speed.

However, such a critic is really trivial to write. Its essentially the inverse of the simplest form of the Constraint Critic. I'd be fine merging this into Nav2 if you submitted a PR, but I'd suggest doing alot of testing to make sure you work out the kinks and edge cases. You don't always want to just travel as fast as possible.

aosmw, 2024-03-28:
Thanks for the insights.  I'll keep it in mind and I'll open a PR if I end up using it.
Happy for you to close this issue for now.

SteveMacenski, 2024-03-28:
Great, thanks!

closed 2024-03-28 by SteveMacenski