ros-navigation/navigation2 #4221: commander API hangs with amcl/get_state service not available

david-wb, 2024-03-27:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Iron
- Version or commit hash:
  - 1.2.6
- DDS implementation:
  - I don't know. How do I check? 

#### Steps to reproduce issue

I am launching a vanila nav2 stack with ros-ion/gazebo-fortress. Everything starts up properly and I can run waypoint following via Rviz. However, when I try to run waypoint following using the commander API it complains that the `amcl/get_state` service is unavailable. 

Here is what I am doing with the commander API.
```
nav = BasicNavigator()
nav.waitUntilNav2Active()
p = PoseStamped()
p.pose.position.x = 1
p.pose.position.y = 1
p.pose.position.z = 0
p.pose.orientation.x = 0
p.pose.orientation.y = 0
p.pose.orientation.z = 0
p.pose.orientation.w = 1
nav.followWaypoints([p])
```
Here are my nav2 params:
```
amcl:
  ros__parameters:
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
[... 432 lines cut ...]
            ),
            node_robot_state_publisher,
            spawn_entity,
            static_tf,
            bringup,
            rviz_node,
        ]
    )  # type: ignore

```
#### Expected behavior
The commander API should initialize and robot should follow waypoints.

#### Actual behavior
The `nav.waitForNav2Active()` call hangs.
```
[namoros-8] [INFO] [1711542259.476309392] [namoros.namoros_main]: amcl/get_state service not available, waiting...
[namoros-8] [INFO] [1711542260.580838297] [namoros.namoros_main]: amcl/get_state service not available, waiting...
[namoros-8] [INFO] [1711542261.599272221] [namoros.namoros_main]: amcl/get_state service not available, waiting...
[namoros-8] [INFO] [1711542262.654907867] [namoros.namoros_main]: amcl/get_state service not available, waiting...
[namoros-8] [INFO] [1711542263.684143388] [namoros.namoros_main]: amcl/get_state service not available, waiting...
```
#### Additional information

----

SteveMacenski, 2024-03-27:
Dont cross post with Stack Exchange.

closed 2024-03-27 by SteveMacenski

david-wb, 2024-03-27:
@SteveMacenski I posted here with additional context because I think there is a bug related to namespacing. Please read my comments in the "Additional Information" section above.

I think [this line](https://github.com/ros-planning/navigation.ros.org/blob/b3a5fef6240dd52001627470df65b66e4db52685/commander_api/index.rst?plain=1#L58) from the docs might have something to do with it.