ros-navigation/navigation2 #4232: [mppi] Investigate speed up potential of preallocated noise.

aosmw, 2024-04-02:
## Feature request

#### Feature description
Add a noise_benchmark to help explore potential speed up of using an xt::adapt(ed) vector of pre-generated noise rather than lazily evaluating it with `xt::random::randn`.

#### Implementation considerations
* Trade off memory usage for speed of operation?
* How much noise is enough?
* Acceptable startup delay to pre-generate noise?
* Option to reuse pre-generated noise during optimiser reset?
* Does it translate to an actual improvement?

#### Preliminary Work.
Code - https://github.com/aosmw/navigation2/blob/feature/mw/pregenerate_noise/nav2_mppi_controller/benchmark/noise_benchmark.cpp
Results - [noise_benchmark.txt](https://github.com/ros-planning/navigation2/files/14833272/noise_benchmark.txt)

SteveMacenski, 2024-04-02:
We are not lazily evaluating it with `xt::random::randn`, if you look at the data structures involved, its immediately evaluated into the `xtensor<float,2>`, not an evaluation expression.  But, happy for any contributions you might be able to provide to accelerate that process :-) Perhaps some different patterns here could be applied to other bottlenecks like trajectory rollouts. 

Note that we have the `regenerate_noises` parameter that lets you select whether to regenerate noises each iteration or use a static set noise distribution for all iterations. After speaking with some folks about this, its not the most ideal for small samples, but as the number of trajectories being evaluated increases, it is increasingly less important that it is resampled each time. There are several companies that use MPPI in automotive applications which I spoke with and did not regenerate noises each iteration in their certified implementations, so I took my inspiration from them and "good enough for them, good enough for me". 

But your vector 1k/3k being 10x faster seems really compelling, can you open a PR to introduce that and I can test it real quick on my side and merge it? I would mention that your numbers don't align with my experiences with computational time of these operations (they're a bit low) so I wonder in context if there's some cache breaks that result in some of that slow down. But hey, saving 0.7ms is a worthwhile activity!

SteveMacenski, 2024-04-22:
@aosmw any insights from your work?

SteveMacenski, 2024-05-03:
Closing without update. The other threads with Eigen seem more relevant

closed 2024-05-03 by SteveMacenski