ros-navigation/navigation2 #4233: maxRGB - alpha broke color logic

pank-su, 2024-04-02:
There are potential issues with handling transparency (alpha channel) in map server images.

- Black pixels (color #000000) with occupied_thresh 0.9: Black might be misinterpreted as "unknown" or "not occupied" if the background is transparent.
- White pixels on transparent backgrounds: White pixels might be considered less occupied than transparent areas.

Possible solutions:

- Remove transparency handling: This simplifies the code but might lead to unexpected behavior with transparent images.
- Document the issue: Clearly explain this potential problem in the documentation to avoid confusion.
- Configuration option for transparency: Implement a setting to control how the server handles transparency in images. This allows for more flexibility based on specific use cases.

https://github.com/ros-planning/navigation2/blob/2ecc91e47bb09485289531144ab9b038f4cfc432/nav2_map_server/src/map_io.cpp#L204C8-L205C99

SteveMacenski, 2024-04-02:
Quite frankly, this has been inherited and inherited since the dawn of modern robotics even pre-ROS' rise. I won't defend any particular element of it, honestly its not a place I spend alot of my time. I support any additional documentation you think would clarify this for users! Though, I don't think we're going to be making any changes to this given the wide reaching ramifications and the integrations with other SLAM and localization libraries that it could break in subtle ways for alot of users

SteveMacenski, 2024-04-22:
@pank-su any update?

SteveMacenski, 2024-05-03:
Closing without update

closed 2024-05-03 by SteveMacenski