ros-navigation/navigation2 #4238: [RP[P] Make new deceleration on cancelation an option

SteveMacenski, 2024-04-04:
Create a parameter for if we should cancel using deceleration (default true), then use that in the logic introduced in https://github.com/ros-planning/navigation2/pull/4136 to bypass if needed

Finally, add the new parameter to documentation & add the new param to the RPP deceleration migration guide entry

doisyg, 2024-04-04:
Maybe we could just use `cancel_deceleration` < 0.0 for disabling the behavior

tanmay-dhasade, 2024-05-11:
Hi @SteveMacenski, I am new to open source contributions and would like to work on this, wanted to check in if someone is already working on this? Thanks

jwdinius, 2024-05-11:
Let's say that a developer sets the value of the proposed parameter, call it `apply_cancel_deceleration`, to `false`, what should `linear_vel` be set to when `cancelling_` is `true` (see [here](https://github.com/nobleo/navigation2/blob/be2e9d2148e462e6c9384940793cf65982b947bc/nav2_regulated_pure_pursuit_controller/src/regulated_pure_pursuit_controller.cpp#L243))?  `0`?

doisyg, 2024-05-12:
> Let's say that a developer sets the value of the proposed parameter, call it `apply_cancel_deceleration`, to `false`, what should `linear_vel` be set to when `cancelling_` is `true` (see [...]

I would say 0.0, as before the `cancel_deceleration` was introduced

(also consider not using a new parameter but just `cancel_deceleration < 0.0`)

SteveMacenski, 2024-05-13:
@tanmay-dhasade no one is working on it yet, this would be a great first contribution!

SteveMacenski, 2024-06-18:
Merged!