ros-navigation/navigation2 #4242: Is there a method within the ros2 controller plugin to determine its termination status?

GPrathap, 2024-04-05:
**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble
- Version or commit hash:
- DDS implementation:
  - Fast-RTPS

Is there a method within the ros2 controller plugin to determine its termination status? I have multiple plugins loaded dynamically via the Behavior Tree. My objective is to clear certain messages in Rviz before transitioning to the next controller plugin. Currently, I'm continuously monitoring the distance to the goal, and upon reaching a close proximity to the goal tolerance, I clear the Rviz messages. Is there a more efficient approach to achieve this?

SteveMacenski, 2024-04-05:
I'm not sure I fully understand. The  controllers are isolated from each other, so they don't know anything about each other's status, no. 

What would you want to try to communicate between them and why? I don't understand what "clear certain messages in Rviz" means.

GPrathap, 2024-04-05:
Apologies, I don't require communication between them. Every controller plugin sends messages to rviz, such as markers, path, or custom messages. Yet, when transitioning one plugin to another, I must send empty messages from the preceding plugin. How can I determine when to send empty messages to rviz before it terminates?

SteveMacenski, 2024-04-05:
> I must send empty messages from the preceding plugin

If this is just for visualization: why?

SteveMacenski, 2024-04-05:
You can set the `decay` time in rviz to non-zero for those visualization displays and it'll delete displays after that time is expired! That should solve your concern without needing to modify any source code.

closed 2024-04-05 by SteveMacenski