ros-navigation/navigation2 #4247: Expanding the ~/ to the full home dir of user

Wiktor-99, 2024-04-08:
## Feature request
When lunching nav2 with following command:

`ros2 launch nav2_bringup tb3_simulation_launch.py  use_sim_time:=true map:=~/my_maps/house_map.yaml`

I get following error:

[image]

The **"~/"** is not evaluated to **/home/user** and map_server cannot find the yaml file. 
The exception is thrown but the error message is not very helpful.

### Feature description
Add additional info in logs about map path or expand the  `~/` to user home dir.

### Implementation considerations
What could be done is to improve error message e.g:
* add information if the `~/` was used and that this format is not supported [here](https://github.com/ros-planning/navigation2/blob/main/nav2_map_server/src/map_io.cpp#L288) 

Another option is to expand the  `~/` to user home dir. I believe it can be done easily with following code added [here](https://github.com/ros-planning/navigation2/blob/main/nav2_map_server/src/map_io.cpp#L118):
```
LoadParameters loadMapYaml(const std::string & yaml_filename)
{
    if (yaml_filename[0] == '~' and yaml_filename[1] == '/') 
    {   
        if (const char* home_dir = std::getenv("HOME"))
        {
            const std::string prefix{home_dir};
            yaml_filename.replace(0, 1, prefix);
        }
    } 
    YAML::Node doc = YAML::LoadFile(yaml_filename);
    ...
}
```
In this approach probably the error message should improved as well as we can get nullptr from the `std::getenv("HOME")`.
   
### Remarks
When one of approaches will be chosen I can prepare the PR.
It's small feature but i find it really helpful.

SteveMacenski, 2024-04-08:
A PR to add that would be appreciated :-) Obviously need to add the checks on functions as you mention (and make sure the path is > 1 long to be able to index from 0/1). 

You can also use the yaml files and use launch substitutions within that context, example:

> $(find-pkg-share my_package)/behavior_tree/my_nav_to_pose_bt.xml

This is a use case of relative path locations as long as its stored in a package structure.

Wiktor-99, 2024-04-11:
I'm aware of the launch substitutions, but when you want fiddle around and test something sometimes it could be faster to pass path form terminal.  Anyway |'ve add PR with fixed linting. 

Thanks for help :)

closed 2024-04-11 by SteveMacenski