ros-navigation/navigation2 #4248: Costmap resolution is not match

Liang-Chin, 2024-04-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble

#### Steps to reproduce issue

Set print log to show the costmap resolution at configure() and createPlan() in smac_planner_hybrid.cpp.

#### Expected behavior

It need to show the same value.
#### Actual behavior

[image]

#### Additional information
I think it will have an impact on generating the path, because the parameter "minimum_turning_radius" is calculated at configure(). And   it will be used to calculate together with the new costmap resolution in subsequent operations

Liang-Chin, 2024-04-10:
Thanks for your response! Now I use option 1 to pair these up.  
Actually I use 2cm map and without setting costmap resolution, but the default costmap resolution is 0.1m. So others will have the same problem if  they use the map other than 0.1m.

SteveMacenski, 2024-04-10:
Set the costmap resolution to 0.02 in your configuration file and that is fixed. I think the issue is that you didn't set it.

closed 2024-04-10 by SteveMacenski