ros-navigation/navigation2 #4250: BT navigator not recognizing a basic ActionNode

jesusramondovale, 2024-04-09:
**Required Info:**

- Operating System:
  - 
- ROS2 Version:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 

#### Steps to reproduce issue

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2024-04-09:
I don’t immediately see your issue, but you can find a full example in opennav_coverage https://github.com/open-navigation/opennav_coverage of custom external BT nodes for an external server. Hopefully you can find your mistake reviewing that!

jesusramondovale, 2024-04-10:
Thanks for your response! 

Actually I almost copied that exact example but I think there's a problem registering the node or compiling the library. 

I know that Nav2 is finding and loading the library "correctly" because if I misspell it on nav2_params.yaml, it says that library isn't found, so it must be with the building step.

SteveMacenski, 2024-04-10:
Sorry, please ask on robot stack exchange or look into the examples yourself in more detail. We don't have the bandwidth to debug every user's errors / misunderstandings. We have plenty of examples and references available so there's no shortage of recipes to look into to find your error. This isn't really a Nav2 issue but a BT.CPP problem understanding of how to build and work with custom plugins, it seems to me.

closed 2024-04-10 by SteveMacenski