ros-navigation/navigation2 #4251: HUMBLE: Undefined symbol when launching bt_navigator

EdMlt, 2024-04-09:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu 22.04
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
  - ros-humble-navigation2	1.1.13-1jammy.20240217.105800 
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
Firstly run turtlebot3 gazebo simulation:
```
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
```

In another terminal run nav2_bringup navigation_launch.py:
```
ros2 launch nav2_bringup navigation_launch.py use_sim_time:=True
```

#### Expected behavior
All nodes required to make navigation

#### Actual behavior
All nodes seems to launch, but the bt_navigator dies instantly:
```
[INFO] [bt_navigator-5]: process started with pid [66808]
[INFO] [waypoint_follower-6]: process started with pid [66810]
[INFO] [velocity_smoother-7]: process started with pid [66812]
[INFO] [lifecycle_manager-8]: process started with pid [66814]
[bt_navigator-5] /opt/ros/humble/lib/nav2_bt_navigator/bt_navigator: symbol lookup error: /opt/ros/humble/lib/nav2_bt_navigator/bt_navigator: undefined symbol: _ZN17nav2_bt_navigator11BtNavigatorC1EN6rclcpp11NodeOptionsE
[ERROR] [bt_navigator-5]: process has died [pid 66808, exit code 127, cmd '/opt/ros/humble/lib/nav2_bt_navigator/bt_navigator --ros-args --log-level info --ros-args -r __node:=bt_navigator --params-file /tmp/launch_params_4fxnlvrq -r /tf:=tf -r /tf_static:=tf_static'].
```

I then cannot give a goal to my robot as I don't have a BT.

Thanks for any help

SteveMacenski, 2024-04-09:
I tested the current humble binaries and don't have an issue - do you se this with `ros2 launch nav2_bringup tb3_simulation_launch.py` or only the TB3 launch files?

EdMlt, 2024-04-10:
Running ` ros2 launch nav2_bringup tb3_simulation_launch.py slam:=True` it seems ok:
```
[component_container_isolated-5] [INFO] [1712736784.869636928] [lifecycle_manager_navigation]: Activating bt_navigator
[component_container_isolated-5] [INFO] [1712736784.869841594] [bt_navigator]: Activating
[component_container_isolated-5] [INFO] [1712736784.959886682] [bt_navigator]: Creating bond (bt_navigator) to lifecycle manager.
[component_container_isolated-5] [INFO] [1712736785.064324979] [lifecycle_manager_navigation]: Server bt_navigator connected with bond.
```
Can send a goal to the robot and it goes by
The issue i reported earlier is also happening with my custom robot when trying to bringup nav2 with` ros2 launch nav2_bringup navigation_launch.py use_sim_time:=True`

EdMlt, 2024-04-10:
Also encountering the issue when trying to run `gps_waypoint_follower.launch.py`
[launch.log](https://github.com/ros-planning/navigation2/files/14935250/launch.log)

SteveMacenski, 2024-04-10:
Please review the package and try to find your error then :-) If this isn't an issue with Nav2 but rather your custom use of it, I can't provide much feedback or help beyond the fact that we have the `nav2_bringup` package & example for a reason :smile:

closed 2024-04-10 by SteveMacenski

bandasaikrishna, 2024-05-19:
Hi,
This problem occurs when nav2_bt_navigator node is run as a normal node and no issue when run as composable node.

In "tb3_simulation_launch.py" launch file, "use_composition" is set to True by default hence problem is not seen when  "tb3_simulation_launch.py" is run with default configuration. 

Use below command to see the issue with "tb3_simulation_launch.py".

$ ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False use_composition:=False

Hope this will be fixed.

SteveMacenski, 2024-05-20:
Using binaries

```
steve@reese:~$ apt info ros-humble-navigation2
Package: ros-humble-navigation2
Version: 1.1.14-1jammy.20240407.041637
```

I just tested your command and things work fine for me :-) 

```
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False use_composition:=False
```

aaronwilhelm1, 2024-10-28:
I have run into this issue as well. From what I can tell, it's related to opennav_coverage. I find that if I don't source the opennav_coverage overlay then I can use bt_navigator as usual but if I do source the overlay then the bt_navigator doesn't work outside of opennav_coverage examples because I get the undefined symbol error. I'm wondering if this has to do with the fact that for Humble opennav_coverage implements a backported bt_navigator and that may be interfering with the nav2 bt_navigator

rafa-oak, 2024-11-03:
> I have run into this issue as well. From what I can tell, it's related to opennav_coverage. I find that if I don't source the opennav_coverage overlay then I can use bt_navigator as usual but if I [...]

Thank you, this in in fact the case. I've been stuck trying to figure out what happened to my projects for a while and not sourcing the open_nav coverage solved it. 
However, I want to use open_nav coverage integrated into my navigation, is there any way to fix this?

aaronwilhelm1, 2024-11-03:
You won't be able to run the turtlebot demos directly after sourcing opennav_coverage. So copy over their launch file and param files and then just use the backported bt_navigator from open_nav coverage. You can set it as the the plugin to use for the navigate_to_pose navigator in the param file. Here's an example from my param file:

```
bt_navigator:
  ros__parameters:
    use_sim_time: False
    global_frame: map
    robot_base_frame: base_link
    odom_topic: odom
    bt_loop_duration: 10
    default_server_timeout: 20
    navigators: ['navigate_complete_coverage', 'navigate_to_pose']
    navigate_complete_coverage:
      plugin: "opennav_coverage_navigator/CoverageNavigator"
    navigate_to_pose:
      plugin: "backported_bt_navigator/NavigateToPoseNavigator"
    plugin_lib_names:
      - nav2_compute_path_to_pose_action_bt_node
      - nav2_compute_path_through_poses_action_bt_node
      - nav2_smooth_path_action_bt_node
      - nav2_follow_path_action_bt_node
      - nav2_spin_action_bt_node
      - nav2_wait_action_bt_node
      - nav2_assisted_teleop_action_bt_node
      - nav2_back_up_action_bt_node
      - nav2_drive_on_heading_bt_node
      - nav2_clear_costmap_service_bt_node
      - nav2_is_stuck_condition_bt_node
      - nav2_goal_reached_condition_bt_node
      - nav2_goal_updated_condition_bt_node
      - nav2_globally_updated_goal_condition_bt_node
      - nav2_is_path_valid_condition_bt_node
      - nav2_initial_pose_received_condition_bt_node
      - nav2_reinitialize_global_localization_service_bt_node
      - nav2_rate_controller_bt_node
      - nav2_distance_controller_bt_node
      - nav2_speed_controller_bt_node
      - nav2_truncate_path_action_bt_node
      - nav2_truncate_path_local_action_bt_node
      - nav2_goal_updater_node_bt_node
      - nav2_recovery_node_bt_node
      - nav2_pipeline_sequence_bt_node
      - nav2_round_robin_node_bt_node
      - nav2_transform_available_condition_bt_node
      - nav2_time_expired_condition_bt_node
      - nav2_path_expiring_timer_condition
      - nav2_distance_traveled_condition_bt_node
      - nav2_single_trigger_bt_node
      - nav2_goal_updated_controller_bt_node
      - nav2_is_battery_low_condition_bt_node
      - nav2_navigate_through_poses_action_bt_node
      - nav2_navigate_to_pose_action_bt_node
      - nav2_remove_passed_goals_action_bt_node
      - nav2_planner_selector_bt_node
      - nav2_controller_selector_bt_node
      - nav2_goal_checker_selector_bt_node
      - nav2_controller_cancel_bt_node
      - nav2_path_longer_on_approach_bt_node
      - nav2_wait_cancel_bt_node
      - nav2_spin_cancel_bt_node
      - nav2_back_up_cancel_bt_node
      - nav2_assisted_teleop_cancel_bt_node
      - nav2_drive_on_heading_cancel_bt_node
      - opennav_compute_complete_coverage_action_bt_node
      - opennav_cancel_complete_coverage_action_bt_node
    error_code_names: ["compute_path_error_code", "follow_path_error_code", "compute_coverage_error_code"]
 ```

SteveMacenski, 2024-11-04:
Sorry, I was gone for ROSCon, ROS-I, and a short vacation. Getting back to things now.

This is the current answer from @aaronwilhelm1. The BT Navigator had to be backported and changed to support the Coverage Navigator type for Humble so you need to use that new version in conjunction with opennav_coverage. We could not back those ABI breaking changes in Nav2 itself for Humble. 

If you do not need the Coverage Navigator at all (just using the task server), you can safely delete the backported bt navigator from the opennav coverage branch (and probably delete at BT node or two) so it doesn't compile and you should be good to go to use the task server alone. If you need the coverage BT navigator API, then you should adjust your launch files to use that new BT Navigator instead of the Humble default BT navigator server. Once you upgrade to Iron or newer, you will not need to do this anymore.

rafa-oak, 2024-11-18:
> You won't be able to run the turtlebot demos directly after sourcing opennav_coverage. So copy over their launch file and param files and then just use the backported bt_navigator from open_nav [...]
Hi, 

I followed these exact instructions and I still get the same undefined symbol error if I have opennav_coverage sourced. This happens when i run this command:
`ros2 launch nav2_bringup navigation_launch.py use_composition:=True  params_file:=/home/rafael/ros2_ws/src/scout_nav2_gz/config/nav2_no_map_params_backported.yaml`

The parameter file is [nav2_no_map_params.yaml](https://github.com/rafa-oak/novamob/blob/new_gazebo/scout_nav2_gz/config/nav2_no_map_params_backported.yaml), with the changes to use the backported_bt_navigator from opennav_coverage, and it still doesnt work. Did you make any other changes, like to the backported_bt_navigator itself?