ros-navigation/navigation2 #4260: cloned_multi_tb3_simulation_launch map is not being published

Unknown-Student, 2024-04-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22-04 running in WSL on Windows 10
- ROS2 Version:
  - Humble 
- Version or commit hash:
  - ros-humble-navigation2  1.1.13-1jammy.20240217.105800

- DDS implementation:
  - Cyclone

#### Steps to reproduce issue

I launched the ```cloned_multi_tb3_simulation_launch.py``` launchfile as follows:
```
ros2 launch nav2_bringup cloned_multi_tb3_simulation_launch.py robots:="robot1={x: 1.0, y: 2.0, yaw: 1.5707}; robot2={x: 1.0, y: -2.0, yaw: 1.5707}" slam:=False 
```

#### Expected behavior
Start of Gazebo and two Rviz windows, one for each robot with the map loaded.

#### Actual behavior
No map is loaded and the map server is also not started. The RQT Graph and the ``` ros2 topic list``` do not show a map topic.
Do I need to remap a topic manually ? I would expect the map to be published to <robot_name>/map respectively.

The Gazebo Simulation starts with the two robots and two Rviz windows open one for each robot:
[image]

RQT Graph:
[image]

output of ```ros2 topic list```
```
/clock
/diagnostics
/parameter_events
/performance_metrics
/robot1/cmd_vel
/robot1/global_costmap/costmap
/robot1/global_costmap/costmap_updates
/robot1/imu
/robot1/initialpose
/robot1/intel_realsense_r200_depth/camera_info
/robot1/intel_realsense_r200_depth/depth/camera_info
/robot1/intel_realsense_r200_depth/depth/image_raw
/robot1/intel_realsense_r200_depth/depth/image_raw/compressed
/robot1/intel_realsense_r200_depth/depth/image_raw/compressedDepth
/robot1/intel_realsense_r200_depth/depth/image_raw/theora
/robot1/intel_realsense_r200_depth/image_raw
/robot1/intel_realsense_r200_depth/image_raw/compressed
/robot1/intel_realsense_r200_depth/image_raw/compressedDepth
/robot1/intel_realsense_r200_depth/image_raw/theora
/robot1/intel_realsense_r200_depth/points
/robot1/joint_states
/robot1/local_costmap/costmap
/robot1/local_costmap/costmap_updates
/robot1/local_costmap/published_footprint
/robot1/local_plan
/robot1/map
/robot1/map_updates
/robot1/mobile_base/sensors/bumper_pointcloud
/robot1/odom
/robot1/particle_cloud
/robot1/plan
/robot1/robot_description
/robot1/scan
/robot1/tf
/robot1/tf_static
/robot1/waypoints
/robot2/amcl/transition_event
/robot2/behavior_server/transition_event
/robot2/bt_navigator/transition_event
/robot2/cmd_vel
/robot2/controller_server/transition_event
/robot2/global_costmap/costmap
/robot2/global_costmap/costmap_updates
/robot2/global_costmap/global_costmap/transition_event
/robot2/imu
/robot2/initialpose
/robot2/intel_realsense_r200_depth/camera_info
/robot2/intel_realsense_r200_depth/depth/camera_info
/robot2/intel_realsense_r200_depth/depth/image_raw
/robot2/intel_realsense_r200_depth/depth/image_raw/compressed
/robot2/intel_realsense_r200_depth/depth/image_raw/compressedDepth
/robot2/intel_realsense_r200_depth/depth/image_raw/theora
/robot2/intel_realsense_r200_depth/image_raw
/robot2/intel_realsense_r200_depth/image_raw/compressed
/robot2/intel_realsense_r200_depth/image_raw/compressedDepth
/robot2/intel_realsense_r200_depth/image_raw/theora
/robot2/intel_realsense_r200_depth/points
/robot2/joint_states
/robot2/local_costmap/costmap
/robot2/local_costmap/costmap_updates
/robot2/local_costmap/local_costmap/transition_event
/robot2/local_costmap/published_footprint
/robot2/local_plan
/robot2/map
/robot2/map_server/transition_event
/robot2/map_updates
/robot2/mobile_base/sensors/bumper_pointcloud
/robot2/odom
/robot2/particle_cloud
/robot2/plan
/robot2/planner_server/transition_event
/robot2/robot_description
/robot2/scan
/robot2/smoother_server/transition_event
/robot2/tf
/robot2/tf_static
/robot2/velocity_smoother/transition_event
/robot2/waypoint_follower/transition_event
/robot2/waypoints
/rosout 
```

SteveMacenski, 2024-04-12:
Hit "startup" - Nav2 starts in the inactive state in the multirobot bringup launch files :smile:

closed 2024-04-12 by SteveMacenski

Unknown-Student, 2024-04-13:
Thank you for the quick response. However when I hit "startup" I get the following error output. 

[component_container_isolated-9] [INFO] [1712997793.523376375] [robot2.lifecycle_manager_navigation]: Configuring behavior_server
[component_container_isolated-9] [INFO] [1712997793.523622913] [robot2.behavior_server]: Configuring
[component_container_isolated-9] [INFO] [1712997793.529405218] [robot2.behavior_server]: Creating behavior plugin spin of type nav2_behaviors/Spin
[component_container_isolated-9] [INFO] [1712997793.530493930] [robot2.behavior_server]: Configuring spin
[component_container_isolated-9] [INFO] [1712997793.535834517] [robot2.behavior_server]: Creating behavior plugin backup of type nav2_behaviors/BackUp
[component_container_isolated-9] [INFO] [1712997793.536937255] [robot2.behavior_server]: Configuring backup
[component_container_isolated-9] [INFO] [1712997793.541945830] [robot2.behavior_server]: Creating behavior plugin drive_on_heading of type nav2_behaviors/DriveOnHeading
[component_container_isolated-9] [INFO] [1712997793.542991402] [robot2.behavior_server]: Configuring drive_on_heading
[component_container_isolated-9] [INFO] [1712997793.546849067] [robot2.behavior_server]: Creating behavior plugin assisted_teleop of type nav2_behaviors/AssistedTeleop
[component_container_isolated-9] [INFO] [1712997793.548907324] [robot2.behavior_server]: Configuring assisted_teleop
[component_container_isolated-9] [INFO] [1712997793.555642773] [robot2.behavior_server]: Creating behavior plugin wait of type nav2_behaviors/Wait
[component_container_isolated-9] [INFO] [1712997793.556352253] [robot2.behavior_server]: Configuring wait
[component_container_isolated-9] [INFO] [1712997793.560182417] [robot2.lifecycle_manager_navigation]: Configuring bt_navigator
[component_container_isolated-9] [INFO] [1712997793.560444449] [robot2.bt_navigator]: Configuring
[component_container_isolated-9] [ERROR] [1712997793.594457838] [robot2.bt_navigator]: Caught exception in callback for transition 10
[component_container_isolated-9] [ERROR] [1712997793.594532808] [robot2.bt_navigator]: Original error: Could not load library: libnav2_are_error_codes_active_condition_bt_node.so: cannot open shared object file: No such file or directory
[component_container_isolated-9] [WARN] [1712997793.594556423] [robot2.bt_navigator]: Error occurred while doing error handling.
[component_container_isolated-9] [FATAL] [1712997793.594568726] [robot2.bt_navigator]: Lifecycle node bt_navigator does not have error state implemented
[component_container_isolated-9] [ERROR] [1712997793.594961383] [robot2.lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[component_container_isolated-9] [ERROR] [1712997793.595012168] [robot2.lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.

SteveMacenski, 2024-04-13:
Make sure to use an appropriate configuration file for your distribution. Don’t use the main branch’s for humble.