ros-navigation/navigation2 #4263: Nav2 Robot only rotating

alboron18, 2024-04-16:
## Bug report
Hi! As stated in the title, I'm currently working on a school project where we are using Nav2 to get a differential drive robot to autonomously navigate a given area. We are successfully mapping the environment using the SLAM toolbox and can generate a costmap, but when we use the 2D goal pose the robot just turns in place. We are echoing the cmd_vel output from Nav2 and it is consistent with this, just a constant output rotational velocity. Also, in the nav2 plugin for Rviz the display states that navigation is active but localization is inactive. Any help is much appreciated thanks! We have been modifying the mapping param file to decrease the cost map inflation layers and increase the robot speed.

- Operating System:
  Ubuntu 20.04
- ROS2 Version:
  Foxy
- DDS implementation:
  Fast-RTPS

#### Steps to reproduce issue
To create this issue, we launch our LiDAR driver in ROS1 because that is the only driver support it has, use a ROS bridge to get the data over to ROS2. Then we launch our robot state publisher and camera node which gives us odometry data. From there we launch the slam toolbox and nav2 after that. all runs well up to the costmaps for nav2. The costmaps generate but when we use a goal pose the robot just turns.

#### Expected behavior
We expect to get some x direction movement out of the robot, not just angular velocities.

#### Actual behavior
As mentioned above, we only see rotational velocity commands being sent from Nav2.

Thanks for the help! This is my first robotics project so I'm eager to learn how to fix this for now and in the future.

padhupradheep, 2024-04-19:
Hey Alboron, I really appreciate your enthusiasm. However, I would recommend you to ask your question in robotics stack exchange where you will get help. This is an issue tracker to report bugs and feature requests for Nav2 stack. I hope you understand. 

Also note that ROS 2 Foxy is not supported and it already reached End of Life. The new version is ROS 2 Iron. 

But please keep going.. you will learn for sure..

SteveMacenski, 2024-04-21:
Ditto what @padhupradheep said!

closed 2024-04-21 by SteveMacenski