ros-navigation/navigation2 #4264: Docker build fails

stevedanomodolor, 2024-04-17:
## Bug report

**Required Info:**

- Operating System:
  - ubunut 22 
- ROS2 Version:
  - ros rolling docker
- Version or commit hash:
  - current main
- DDS implementation:
  - default 

#### Steps to reproduce issue

```
export $ROS_DISTRO=rolling
docker build --tag navigation2:$ROS_DISTRO \ 
  --build-arg FROM_IMAGE=ros:$ROS_DISTRO \
  --build-arg OVERLAY_MIXINS="release ccache lld" \
  --cache-from ghcr.io/ros-planning/navigation2:main \
  ./navigation2

```

#### Expected behavior
It should build
#### Actual behavior
It does not build and it gives and error 
```
 [internal] load build definition from Dockerfile                                                                                                                                     0.0s
 => => transferring dockerfile: 6.63kB                                                                                                                                                   0.0s
 => [internal] load metadata for docker.io/library/ros:rolling                                                                                                                           0.8s
 => [internal] load .dockerignore                                                                                                                                                        0.0s
 => => transferring context: 229B                                                                                                                                                        0.0s
 => importing cache manifest from ghcr.io/ros-planning/navigation2:main                                                                                                                  1.3s
 => => inferred cache manifest type: application/vnd.docker.distribution.manifest.v2+json                                                                                                0.0s
 => CACHED [builder  1/11] FROM docker.io/library/ros:rolling@sha256:831fa3f0bb3667c510d945bccfa86847f67d807aaae74342a61bc3a983719f48                                                    0.0s
 => [internal] load build context                                                                                                                                                        0.9s
 => => transferring context: 126.19MB                                                                                                                                                    0.9s
 => [builder  2/11] RUN echo 'APT::Install-Recommends "0";\nAPT::Install-Suggests "0";\n' > /etc/apt/apt.conf.d/01norecommend                                                            0.3s
 => ERROR [builder  3/11] RUN apt-get update &&     apt-get upgrade -y --with-new-pkgs &&     apt-get install -y       ccache       lcov       lld       python3-pip       ros-rolling  76.3s
 => CACHED [cacher 2/8] WORKDIR /opt/underlay_ws/src                                                                                                                                     0.0s
 => CACHED [cacher 3/8] COPY ./tools/underlay.repos ../                                                                                                                                  0.0s
 => [cacher 4/8] RUN vcs import ./ < ../underlay.repos                                                                                                                                   0.5s
 => [cacher 5/8] WORKDIR /opt/overlay_ws/src                                                                                                                                             0.0s
 => [cacher 6/8] COPY ./ ./navigation2                                                                                                                                                   0.3s
 => [cacher 7/8] WORKDIR /opt                                                                                                                                                            0.0s
 => [cacher 8/8] RUN find . -name "src" -type d       -mindepth 1 -maxdepth 2 -printf '%P\n'       | xargs -I % mkdir -p /tmp/opt/% &&     find . -name "package.xml"       | xargs cp   0.5s
------                                                                                                                                                                                        
 > [builder  3/11] RUN apt-get update &&     apt-get upgrade -y --with-new-pkgs &&     apt-get install -y       ccache       lcov       lld       python3-pip       ros-rolling-rmw-fastrtps-cpp       ros-rolling-rmw-connextdds       ros-rolling-rmw-cyclonedds-cpp     && pip3 install       fastcov       git+https://github.com/ruffsl/colcon-cache.git@a937541bfc496c7a267db7ee9d6cceca61e470ca       git+https://github.com/ruffsl/colcon-clean.git@a7f1074d1ebc1a54a6508625b117974f2672f2a9     && rosdep update     && colcon mixin update     && colcon metadata update     && rm -rf /var/lib/apt/lists/*:                                                                                                                                                                  
0.604 Get:1 http://archive.ubuntu.com/ubuntu noble InRelease [255 kB]                                                                                                                         
0.606 Get:2 http://security.ubuntu.com/ubuntu noble-security InRelease [90.7 kB]                                                                                                              
1.057 Get:3 http://packages.ros.org/ros2/ubuntu noble InRelease [4658 B]                                                                                                                      
1.158 Get:4 http://archive.ubuntu.com/ubuntu noble-updates InRelease [89.7 kB]
1.284 Get:5 http://archive.ubuntu.com/ubuntu noble-backports InRelease [90.8 kB]
1.334 Get:6 http://packages.ros.org/ros2/ubuntu noble/main amd64 Packages [344 kB]
1.412 Get:7 http://archive.ubuntu.com/ubuntu noble/restricted amd64 Packages [138 kB]
1.461 Get:8 http://archive.ubuntu.com/ubuntu noble/main amd64 Packages [1806 kB]
1.745 Get:9 http://archive.ubuntu.com/ubuntu noble/universe amd64 Packages [19.4 MB]
3.019 Get:10 http://archive.ubuntu.com/ubuntu noble/multiverse amd64 Packages [295 kB]
3.811 Fetched 22.6 MB in 3s (6575 kB/s)
3.811 Reading package lists...
4.405 Reading package lists...
5.281 Building dependency tree...
5.489 Reading state information...
5.560 Calculating upgrade...
5.855 The following packages will be upgraded:
5.856   bash bzip2 diffutils findutils hostname libacl1 libassuan0 libattr1
5.856   libbz2-1.0 libdatrie1 libdeflate0 libffi8 libgcrypt20 libgpg-error0
[... 558 lines cut ...]
  58 | >>>       ros-$ROS_DISTRO-rmw-connextdds \
  59 | >>>       ros-$ROS_DISTRO-rmw-cyclonedds-cpp \
  60 | >>>     && pip3 install \
  61 | >>>       fastcov \
  62 | >>>       git+https://github.com/ruffsl/colcon-cache.git@a937541bfc496c7a267db7ee9d6cceca61e470ca \
  63 | >>>       git+https://github.com/ruffsl/colcon-clean.git@a7f1074d1ebc1a54a6508625b117974f2672f2a9 \
  64 | >>>     && rosdep update \
  65 | >>>     && colcon mixin update \
  66 | >>>     && colcon metadata update \
  67 | >>>     && rm -rf /var/lib/apt/lists/*
  68 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c apt-get update &&     apt-get upgrade -y --with-new-pkgs &&     apt-get install -y       ccache       lcov       lld       python3-pip       ros-$ROS_DISTRO-rmw-fastrtps-cpp       ros-$ROS_DISTRO-rmw-connextdds       ros-$ROS_DISTRO-rmw-cyclonedds-cpp     && pip3 install       fastcov       git+https://github.com/ruffsl/colcon-cache.git@a937541bfc496c7a267db7ee9d6cceca61e470ca       git+https://github.com/ruffsl/colcon-clean.git@a7f1074d1ebc1a54a6508625b117974f2672f2a9     && rosdep update     && colcon mixin update     && colcon metadata update     && rm -rf /var/lib/apt/lists/*" did not complete successfully: exit code: 1
 ✘  ~/nav2_work  
 ✘  ~/nav2_work  docker build --tag navigation2:$ROS_DISTRO \ 
  --build-arg FROM_IMAGE=ros:$ROS_DISTRO \
  --build-arg OVERLAY_MIXINS="release ccache lld" \
  --cache-from ghcr.io/ros-planning/navigation2:main \
  ./navigation2

```
#### Additional information
* This behavior happens in main also

----

closed 2024-04-19 by stevedanomodolor

Ayush1285, 2024-04-24:
Hi @stevedanomodolor, I'm also facing the same issue on the main branch. How did you resolve it?

stevedanomodolor, 2024-04-26:
I merged with latest main and made sure I had the latest dockers

andywot, 2024-05-09:
@Ayush1285 @stevedanomodolor Hi, I had the latest docker and latest commit on main but I still have this issue. Any idea on how to solve it?

Ayush1285, 2024-05-09:
> @Ayush1285 @stevedanomodolor Hi, I had the latest docker and latest commit on main but I still have this issue. Any idea on how to solve it?

Can you please check version of Ubuntu in your docker  env? Currently due to migration of ROS Rolling from 22.04 to 24.04, we are facing issues regarding packages not available in rosdep. You can try building your own docker image of Nav2 main branch from the ROS Jazzy image, and see if it works or not.

stevedanomodolor, 2024-05-11:
root@stevedan:/opt/overlay_ws# lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.4 LTS
Release:	22.04
Codename:	jammy
ARG FROM_IMAGE=ros:rolling
ARG UNDERLAY_WS=/opt/underlay_ws
ARG OVERLAY_WS=/opt/overlay_ws