ros-navigation/navigation2 #4269: Rviz doesn't load any saved maps with Nav2 bringup

HyydenJakcz, 2024-04-19:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble binaries
- Version or commit hash:
  -ros-humble-nav2-amcl	1.1.13-1jammy.20240217.085112
ros-humble-nav2-behavior-tree	1.1.13-1jammy.20240217.085125
ros-humble-nav2-behaviors	1.1.13-1jammy.20240217.093702
ros-humble-nav2-bringup	1.1.13-1jammy.20240217.110614
ros-humble-nav2-bt-navigator	1.1.13-1jammy.20240217.093705
ros-humble-nav2-collision-monitor	1.1.13-1jammy.20240217.092320
ros-humble-nav2-common	1.1.13-1jammy.20240217.071750
ros-humble-nav2-constrained-smoother	1.1.13-1jammy.20240217.092529
ros-humble-nav2-controller	1.1.13-1jammy.20240217.092532
ros-humble-nav2-core	1.1.13-1jammy.20240217.092311
ros-humble-nav2-costmap-2d	1.1.13-1jammy.20240217.085604
ros-humble-nav2-dwb-controller	1.1.13-1jammy.20240217.094145
ros-humble-nav2-lifecycle-manager	1.1.13-1jammy.20240217.085120
ros-humble-nav2-map-server	1.1.13-1jammy.20240217.084621
ros-humble-nav2-mppi-controller	1.1.13-1jammy.20240217.092539
ros-humble-nav2-msgs	1.1.13-1jammy.20240217.075125
ros-humble-nav2-navfn-planner	1.1.13-1jammy.20240217.092818
ros-humble-nav2-planner	1.1.13-1jammy.20240217.092854
ros-humble-nav2-regulated-pure-pursuit-controller	1.1.13-1jammy.20240217.092859
ros-humble-nav2-rotation-shim-controller	1.1.13-1jammy.20240217.093708
ros-humble-nav2-rviz-plugins	1.1.13-1jammy.20240217.104951
ros-humble-nav2-simple-commander	1.1.13-1jammy.20240217.075845
ros-humble-nav2-smac-planner	1.1.13-1jammy.20240217.092916
ros-humble-nav2-smoother	1.1.13-1jammy.20240217.092921
ros-humble-nav2-theta-star-planner	1.1.13-1jammy.20240217.093253
ros-humble-nav2-util	1.1.13-1jammy.20240217.083804
ros-humble-nav2-velocity-smoother	1.1.13-1jammy.20240217.085122
ros-humble-nav2-voxel-grid	1.1.13-1jammy.20240217.075307
ros-humble-nav2-waypoint-follower	1.1.13-1jammy.20240217.093251

- DDS implementation:
  - cyclone dds

#### Steps to reproduce issue
```
1. launch the gazebo world 'playground'. I included the world file below.
2. ros2 launch slam_toolbox online_async_launch.py use_sime_time:=true
3. drive around and save the map
4. ros2 launch nav2_bringup bringup_launch.py map:={path_to_file} use_sim_time:=true
```
#### Expected behavior
The saved map is loaded into Rviz and asks me to set the initial estimate pose
#### Actual behavior
there is no map frame in Rviz, nor any saved map, meanwhile everything seemingly runs fine in Rviz. If I try to set an initial estimate pose anyway, then the costmaps appear as expected but still no map
#### Additional information

[playground.txt](https://github.com/ros-planning/navigation2/files/15040083/playground.txt)

HyydenJakcz, 2024-04-19:
Here are the log files from the terminal:
[INFO] [launch]: All log files can be found below /home/shananda/.ros/log/2024-04-19-14-48-17-497025-shananda-Legion-5-15ACH6A-14847
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_isolated-1]: process started with pid [14862]
[component_container_isolated-1] [INFO] [1713530898.546034113] [nav2_container]: Load Library: /opt/ros/humble/lib/libmap_server_core.so
[component_container_isolated-1] [INFO] [1713530898.557968865] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::CostmapFilterInfoServer>
[component_container_isolated-1] [INFO] [1713530898.558006440] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapSaver>
[component_container_isolated-1] [INFO] [1713530898.558018732] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapServer>
[component_container_isolated-1] [INFO] [1713530898.558029767] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapServer>
[component_container_isolated-1] [INFO] [1713530898.561407100] [map_server]: 
[component_container_isolated-1] 	map_server lifecycle node launched. 
[component_container_isolated-1] 	Waiting on external lifecycle transitions to activate
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/map_server' in container '/nav2_container'
[component_container_isolated-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[component_container_isolated-1] [INFO] [1713530898.561466326] [map_server]: Creating
[component_container_isolated-1] [INFO] [1713530898.563699443] [nav2_container]: Load Library: /opt/ros/humble/lib/libamcl_core.so
[component_container_isolated-1] [INFO] [1713530898.566314455] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_amcl::AmclNode>
[component_container_isolated-1] [INFO] [1713530898.566341623] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_amcl::AmclNode>
[component_container_isolated-1] [INFO] [1713530898.569075784] [amcl]: 
[component_container_isolated-1] 	amcl lifecycle node launched. 
[component_container_isolated-1] 	Waiting on external lifecycle transitions to activate
[component_container_isolated-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[component_container_isolated-1] [INFO] [1713530898.569543514] [amcl]: Creating
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/amcl' in container '/nav2_container'
[component_container_isolated-1] [INFO] [1713530898.571920505] [nav2_container]: Load Library: /opt/ros/humble/lib/libnav2_lifecycle_manager_core.so
[component_container_isolated-1] [INFO] [1713530898.572425321] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_lifecycle_manager::LifecycleManager>
[component_container_isolated-1] [INFO] [1713530898.572445225] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_lifecycle_manager::LifecycleManager>
[component_container_isolated-1] [INFO] [1713530898.574608291] [lifecycle_manager_localization]: Creating
[component_container_isolated-1] [INFO] [1713530898.576029499] [lifecycle_manager_localization]: [34m[1mCreating and initializing lifecycle service clients[0m[0m
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/lifecycle_manager_localization' in container '/nav2_container'
[component_container_isolated-1] [INFO] [1713530898.577023067] [lifecycle_manager_localization]: [34m[1mStarting managed nodes bringup...[0m[0m
[component_container_isolated-1] [INFO] [1713530898.577064134] [lifecycle_manager_localization]: [34m[1mConfiguring map_server[0m[0m
[component_container_isolated-1] [INFO] [1713530898.577190058] [map_server]: Configuring
[component_container_isolated-1] [INFO] [map_io]: Loading yaml file: /home/shananda/slick_ws/src/slick_navigation/maps/playground.yaml
[component_container_isolated-1] [DEBUG] [map_io]: resolution: 0.05
[component_container_isolated-1] [DEBUG] [map_io]: origin[0]: -10.8
[component_container_isolated-1] [DEBUG] [map_io]: origin[1]: -11.5
[component_container_isolated-1] [DEBUG] [map_io]: origin[2]: 0
[component_container_isolated-1] [DEBUG] [map_io]: free_thresh: 0.25
[component_container_isolated-1] [DEBUG] [map_io]: occupied_thresh: 0.65
[component_container_isolated-1] [DEBUG] [map_io]: mode: trinary
[component_container_isolated-1] [DEBUG] [map_io]: negate: 0
[component_container_isolated-1] [INFO] [map_io]: Loading image_file: /home/shananda/slick_ws/src/slick_navigation/maps/playground.pgm
[component_container_isolated-1] [DEBUG] [map_io]: Read map /home/shananda/slick_ws/src/slick_navigation/maps/playground.pgm: 434 X 416 map @ 0.05 m/cell
[component_container_isolated-1] [INFO] [1713530898.590219718] [lifecycle_manager_localization]: [34m[1mConfiguring amcl[0m[0m
[component_container_isolated-1] [INFO] [1713530898.590325878] [amcl]: Configuring
[component_container_isolated-1] [INFO] [1713530898.590423446] [amcl]: initTransforms
[component_container_isolated-1] [INFO] [1713530898.593312028] [amcl]: initPubSub
[component_container_isolated-1] [INFO] [1713530898.593878234] [amcl]: Subscribed to map topic.
[component_container_isolated-1] [INFO] [1713530898.594750626] [lifecycle_manager_localization]: [34m[1mActivating map_server[0m[0m
[... cut ...]
[component_container_isolated-1] [INFO] [1713530965.742841020] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
[component_container_isolated-1] [INFO] [1713530966.242842016] [global_costmap.global_costmap]: Timed out waiting for transform from base_link to map to become available, tf error: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist

padhupradheep, 2024-04-19:
Hey there, 
I cannot reproduce the issue, unfortunately this is a question for robotics stack exchange.

SteveMacenski, 2024-04-20:
Agreed 👍

closed 2024-04-20 by SteveMacenski