ros-navigation/navigation2 #4277: footprint_clearing_enabled for more (or all) layers

tonynajjar, 2024-04-25:
Spinoff from https://github.com/ros-planning/navigation2/issues/4271#issuecomment-2073467118

## Feature request

#### Feature description
`footprint_clearing_enabled` is only a parameter of the obstacle layer. it's also in the voxel layer but doesn't seem to be used. I think it would make sense if this parameter was also implemented for the static layer. Going further, we can also argue for it to be at top-level costmap such that it applies to all layers/ 

#### Implementation considerations

SteveMacenski, 2024-04-25:
>  it's also in the voxel layer but doesn't seem to be used

The voxel layer derives from the obstacle layer and they share the same `updateFootprint` function which uses the parameter, so the voxel layer should work :-) Do you see it not working or is that based on code review?

Since its a param, I have no issue with it being applied to the static layer as well! :smile: 

Please submit a PR and that should be easy to merge

tonynajjar, 2024-04-25:
> Do you see it not working or is that based on code review?

I'm wrong then, was purely code review (after I found 'clearing_layers' not used so I assumed there is more 😬)

> I have no issue with it being applied to the static layer as well

What do you think about a param to apply it to the whole costmap instead of by layer? I haven't checked if that's easily feasible but I assume so

SteveMacenski, 2024-04-25:
I think we should leave it on a layer by layer basis for granularity of controls. I suppose we could offer both though like we have for some settings. That would be fine!

closed 2024-05-03 by SteveMacenski