ros-navigation/navigation2 #4279: The speed of the differentially driven AMR robot in the DWB does not decrease near the cost map

frkngltk, 2024-04-25:
**Required Info:**

- Operating System:
  - INTEL NUC i7 Gen13, 16 GB RAM, 1TB SSD, UBUNTU 22.04 LTS
- ROS2 Version:
  - ROS2 HUMBLE

#### Information about the problem

Hello, I wish everyone a good day. I have an AMR(differential drive) robot that I work with in real time. In accordance with the main system
`<link>` : <https://navigation.ros.org/configuration/packages/configuring-dwb-controller.html>
from this site and this parameter page 
`<link>` : <https://github.com/ros-planning/navigation2/blob/main/nav2_bringup/params/nav2_params.yaml> 
I made adjustments according to my own robot. You can see my tuning file below.

```yaml
amcl:
  ros__parameters:
    use_sim_time: False
    alpha1: 0.9
    alpha2: 0.1
    alpha3: 0.05
    alpha4: 0.01
    alpha5: 0.04
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.02
    pf_z: 0.85
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 2
    robot_model_type: "nav2_amcl::DifferentialMotionModel"
    save_pose_rate: 0.5
    sigma_hit: 0.02
    tf_broadcast: true
    transform_tolerance: 0.5
    update_min_a: 0.06
    update_min_d: 0.025
    z_hit: 0.7
    z_max: 0.001
    z_rand: 0.059
    z_short: 0.24
    scan_topic: scan_front_filtered

    set_initial_pose: True
    initial_pose.x: 0.0
    initial_pose.y: 0.0
    initial_pose.z: 0.0
    initial_pose.yaw: 0.0

amcl_map_client:
  ros__parameters:
    use_sim_time: False

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: False
[... 313 lines cut ...]
    min_rotational_vel: 0.05
    rotational_acc_lim: 0.3

robot_state_publisher:
  ros__parameters:
    use_sim_time: False

waypoint_follower:
  ros__parameters:
    loop_rate: 20
    stop_on_failure: false
    waypoint_task_executor_plugin: "wait_at_waypoint"
    wait_at_waypoint:
      plugin: "nav2_waypoint_follower::WaitAtWaypoint"
      enabled: True
      waypoint_pause_duration: 200
```

According to this file, the robot moves from point A to point B according to acceleration and deceleration data. My main question is this: When it encounters any obstacle on the way to the target I sent, it reaches the point by bypassing this obstacle, but no slowdown is observed. In short, even if the robot detects the costmaps formed next to it while it is moving, it does not reduce the max_vel_x value and goes to the target point at high speed. I also demonstrated this in the video below.
I would appreciate it if you could help me with this problem.

https://github.com/ros-planning/navigation2/assets/46001550/9786787c-8f0d-45b7-8a27-3ac94608a734

###### Note
*I have just started working on ROS2 and NAV2. I am trying to transfer the AMR robot from the ROS1 system to the ROS2 system.*

SteveMacenski, 2024-04-25:
You likely need to tune DWB to get your desired behavior. There are many critic plugins and relative weights you can set to get the behavior that you want. Since DWB is a scoring-based local trajectory planner, fine tuning is required to balance the different objectives (and costmap settings).

frkngltk, 2024-04-26:
Teşekkürler @SteveMacenski. DWB yapısını tam olarak kavradım ve herbir parametrenin ne işe yaradığına dair forum sitelerinde ve [bu](https://navigation.ros.org/configuration/packages/configuring-dwb-controller.html) kanaldaki bir çok yapıları incelendim. OscillationCritic, BaseObstacleCritic, ObstacleFootprintCritic vb. yapıların bu soruna destek vereceğini düşündüm. Ayrıca local_costmap parametrelerinden olan cost_scaling_factor gibi yapıların sistemi etkilemesi gerekiyor. Asıl sorun şu ki bu parametreler benim kendi ayarladığım haricinde Turtlebot gibi simülasyon ortamlarında da çalışmıyor. NAV2 sisteminde robota yakın bir konumda hareket eden bir cisim veya obje yaklaşında oluşan cost_map'den dolayı otomatik olarak yavaşlaması gerekir diye düşünüyorum. Buradan dediklerinize göre her bir parametre bu durumu tetiklediğini anlıyorum doğru mu ? Kısaca bu sorunu hangi parametre ile yada nasıl bir teknikle çözebilirim. NAV2 sisteminde otomatik olarak bu durumun ( AMR cihazlarla kullanıldığı için) aktif olması  lazım değil midir?

frkngltk, 2024-04-26:
Thanks @SteveMacenski. I fully understood the DWB structure and examined many structures on forum sites and [this](https://navigation.ros.org/configuration/packages/configuring-dwb-controller.html) channel to see what each parameter does. OscillationCritic, BaseObstacleCritic, ObstacleFootprintCritic etc. I thought the structures would support this problem. In addition, structures such as cost_scaling_factor, which are among the local_costmap parameters, should affect the system. The main problem is that these parameters do not work in simulation environments such as Turtlebot, other than the ones I set myself. I think that in the NAV2 system, when an object or object moves close to the robot, it should slow down automatically due to the cost_map created. According to what you said, I understand that each parameter triggers this situation, is it true? Briefly, with which parameter or with what technique can I solve this problem? Shouldn't this situation be automatically active in the NAV2 system (since it is used with AMR devices)?

SteveMacenski, 2024-04-26:
That is what the critic objective function are for, you can define that behavior.

SteveMacenski, 2024-05-03:
I believe this was answered, can we close this out?

frkngltk, 2024-05-13:
I can't say it's fully resolved, but I'm trying the parameters of your specification. I will also try the Collision monitor structure. Thank you.

frkngltk, 2024-05-20:
Hello @SteveMacenski. I am trying the structures I mentioned. But I couldn't reach any conclusion. Here you can use critic plugins, cost_map settings, or Smoother Server, Collision Monitor, Velocity Smoother etc. in the Configuration Guide [link](https://docs.nav2.org/configuration/index.html#configuration-guide). I used structures. According to my observation, there is no change. Frankly, when I run ROS1 with the same parameters on the same robot, I observe a speed slowdown in narrow areas, while the speed increases in wide areas. In the ROS2 nav2 system, there is no speed reduction in narrow or wide areas. I examined in detail the locations of the Critics parameters. I cannot observe any changes based on real-time observation as a result of the adjustment. Do you have any suggestions regarding this subject? I would be glad if you could help.

SteveMacenski, 2024-05-20:
DWB and DWA from ROS 1 are not the same package, though they are related. Like I mentioned above, you will need to tune it to your application desires.

However, I might recommend trying out MPPI. I believe its a pretty big step function up in capabilities and I think it does more or less what you want out of the box how I tuned it.