ros-navigation/navigation2 #4283: UAF bug occurs during `nav2_navfn_planner::NavFn` 's calculation work

GoesM, 2024-04-28:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu22.04
- ROS2 Version:
  -  humble
- Version or commit hash:
  -  the latest
- DDS implementation:
  -  defaulted

#### Steps to reproduce issue

I encounter this UAF-bug many times when using the Nav2Goal feature 

Launch the navigation2 normally, as following steps:
```sh
#!/bin/bash
export ASAN_OPTIONS=halt_on_error=0:new_delete_type_mismatch=0:detect_leaks=0:log_pah=asan
source install/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=True use_rviz:=False use_composition:=False 
```

Keep sending `Nav2Goal` in rviz2-displayer or sending goal by command `ros2 action send_goal` or send msm into topic `/goal_pose` , in which the goal_pose is random.

Finally sent Ctrl+C to shutdown navigation2, which is before stop the action-sending and even before the latest action-goal finished.

An ASAN report file was discovered in my execution environment.

#### Expected behavior

#### Actual behavior
The ASAN reporting a **use-after-free** bug to me, as following:
```
=================================================================
==924435==ERROR: AddressSanitizer: heap-use-after-free on address 0x75dd9a983f48 at pc 0x75dd9fe736b9 bp 0x75dd9e1a8670 sp 0x75dd9e1a8668
READ of size 4 at 0x75dd9a983f48 thread T34
    #0 0x75dd9fe736b8 in nav2_navfn_planner::NavFn::updateCell(int) (/home/***/nav2/install/nav2_navfn_planner/lib/libnav2_navfn_planner.so+0x326b8) (BuildId: 1ac9c097881f207fa259d7749bdafc5eb1e98cf3)
    #1 0x75dd9fe6fa42 in nav2_navfn_planner::NavFn::propNavFnDijkstra(int, bool) (/home/***/nav2/install/nav2_navfn_planner/lib/libnav2_navfn_planner.so+0x2ea42) (BuildId: 1ac9c097881f207fa259d7749bdafc5eb1e98cf3)
    #2 0x75dd9fe5f08f in nav2_navfn_planner::NavfnPlanner::makePlan(geometry_msgs::msg::Pose_<std::allocator<void> > const&, geometry_msgs::msg::Pose_<std::allocator<void> > const&, double, nav_msgs::msg::Path_<std::allocator<void> >&) (/home/***/nav2/install/nav2_navfn_planner/lib/libnav2_navfn_planner.so+0x1e08f) (BuildId: 1ac9c097881f207fa259d7749bdafc5eb1e98cf3)
    #3 0x75dd9fe5d649 in nav2_navfn_planner::NavfnPlanner::createPlan(geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&, geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&) (/home/***/nav2/install/nav2_navfn_planner/lib/libnav2_navfn_planner.so+0x1c649) (BuildId: 1ac9c097881f207fa259d7749bdafc5eb1e98cf3)
    #4 0x75ddac6a4ee5 in nav2_planner::PlannerServer::getPlan(geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&, geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (/home/***/nav2/install/nav2_planner/lib/libplanner_server_core.so+0xa4ee5) (BuildId: aef58e89824c7a18a2835f121508ebf22b22877a)
    #5 0x75ddac690ed9 in nav2_planner::PlannerServer::computePlan() (/home/***/nav2/install/nav2_planner/lib/libplanner_server_core.so+0x90ed9) (BuildId: aef58e89824c7a18a2835f121508ebf22b22877a)
    #6 0x75ddac7994f4 in nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose>::work() (/home/***/nav2/install/nav2_planner/lib/libplanner_server_core.so+0x1994f4) (BuildId: aef58e89824c7a18a2835f121508ebf22b22877a)
    #7 0x75ddac798956 in std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::ComputePathToPose> >)::'lambda'()> >, void>::operator()() const (/home/***/nav2/install/nav2_planner/lib/libplanner_server_core.so+0x198956) (BuildId: aef58e89824c7a18a2835f121508ebf22b22877a)
    #8 0x75ddac79874f in std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> (), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::ComputePathToPose> >)::'lambda'()> >, void> >::_M_invoke(std::_Any_data const&) (/home/***/nav2/install/nav2_planner/lib/libplanner_server_core.so+0x19874f) (BuildId: aef58e89824c7a18a2835f121508ebf22b22877a)
    #9 0x75ddac79837f in std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) (/home/***/nav2/install/nav2_planner/lib/libplanner_server_core.so+0x19837f) (BuildId: aef58e89824c7a18a2835f121508ebf22b22877a)
    #10 0x75ddaa499ee7 in __pthread_once_slow nptl/./nptl/pthread_once.c:116:7
    #11 0x75ddac796ca6 in std::__future_base::_Async_state_impl<std::thread::_Invoker<std::tuple<nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose>::handle_accepted(std::shared_ptr<rclcpp_action::ServerGoalHandle<nav2_msgs::action::ComputePathToPose> >)::'lambda'()> >, void>::_M_run() (/home/***/nav2/install/nav2_planner/lib/libplanner_server_core.so+0x196ca6) (BuildId: aef58e89824c7a18a2835f121508ebf22b22877a)
    #12 0x75ddaa8dc252  (/lib/x86_64-linux-gnu/libstdc++.so.6+0xdc252) (BuildId: e37fe1a879783838de78cbc8c80621fa685d58a2)
    #13 0x75ddaa494ac2 in start_thread nptl/./nptl/pthread_create.c:442:8
    #14 0x75ddaa52684f  misc/../sysdeps/unix/sysv/linux/x86_64/clone3.S:81

0x75dd9a983f48 is located 329544 bytes inside of 589824-byte region [0x75dd9a933800,0x75dd9a9c3800)
freed by thread T0 here:
    #0 0x5af36f5bc9dd in operator delete[](void*) (/home/***/nav2/install/nav2_planner/lib/nav2_planner/planner_server+0xdf9dd) (BuildId: a53c0b4c8e364d2ddd89807e504fe220fd79ef2e)
    #1 0x75dd9fe6bc19 in nav2_navfn_planner::NavFn::~NavFn() (/home/***/nav2/install/nav2_navfn_planner/lib/libnav2_navfn_planner.so+0x2ac19) (BuildId: 1ac9c097881f207fa259d7749bdafc5eb1e98cf3)
    #2 0x75dd9fe524f3 in nav2_navfn_planner::NavfnPlanner::~NavfnPlanner() (/home/***/nav2/install/nav2_navfn_planner/lib/libnav2_navfn_planner.so+0x114f3) (BuildId: 1ac9c097881f207fa259d7749bdafc5eb1e98cf3)

previously allocated by thread T34 here:
    #0 0x5af36f5bc18d in operator new[](unsigned long) (/home/***/nav2/install/nav2_planner/lib/nav2_planner/planner_server+0xdf18d) (BuildId: a53c0b4c8e364d2ddd89807e504fe220fd79ef2e)
    #1 0x75dd9fe6b5ae in nav2_navfn_planner::NavFn::setNavArr(int, int) (/home/***/nav2/install/nav2_navfn_planner/lib/libnav2_navfn_planner.so+0x2a5ae) (BuildId: 1ac9c097881f207fa259d7749bdafc5eb1e98cf3)
    #2 0x75dd9fe5ec88 in nav2_navfn_planner::NavfnPlanner::makePlan(geometry_msgs::msg::Pose_<std::allocator<void> > const&, geometry_msgs::msg::Pose_<std::allocator<void> > const&, double, nav_msgs::msg::Path_<std::allocator<void> >&) (/home/***/nav2/install/nav2_navfn_planner/lib/libnav2_navfn_planner.so+0x1dc88) (BuildId: 1ac9c097881f207fa259d7749bdafc5eb1e98cf3)
    #3 0x75dd9fe5d649 in nav2_navfn_planner::NavfnPlanner::createPlan(geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&, geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&) (/home/***/nav2/install/nav2_navfn_planner/lib/libnav2_navfn_planner.so+0x1c649) (BuildId: 1ac9c097881f207fa259d7749bdafc5eb1e98cf3)
    #4 0x75ddac6a4ee5 in nav2_planner::PlannerServer::getPlan(geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&, geometry_msgs::msg::PoseStamped_<std::allocator<void> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (/home/***/nav2/install/nav2_planner/lib/libplanner_server_core.so+0xa4ee5) (BuildId: aef58e89824c7a18a2835f121508ebf22b22877a)
    #5 0x75ddac690ed9 in nav2_planner::PlannerServer::computePlan() (/home/***/nav2/install/nav2_planner/lib/libplanner_server_core.so+0x90ed9) (BuildId: aef58e89824c7a18a2835f121508ebf22b22877a)
[... 43 lines cut ...]
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==924435==ABORTING
```

#### Additional information

----

Accroding to the ASAN report , 

During the function `nav2_navfn_planner::NavFn::propNavFnDijkstra` running , the resources in need for this function were freed by the destructor of `nav2_planner`

However, it seems that here's already a `action_server_pose->deactivate()` and `action_server_pose.reset()` in `on_deactivate()` and `on_cleanup()` of nav2_planner

GoesM, 2024-04-28:
### Additional check
Finally, I confirmed the reason for this issue, it's also caused by the `error-thrown` mechanism defect in `nav2_util`  (in humble branch )

https://github.com/ros-planning/navigation2/blob/150dd6beb50aee502ada8935cee23d7160a40f57/nav2_util/include/nav2_util/simple_action_server.hpp#L294

It seems like another ticket for #4175 and #4180, and has been fixed in main-branch in our last work.

GoesM, 2024-04-29:
### Additional check
Through my expeirments( by insert thread-sleep in the function `nav2_navfn_planner::NavfnPlanner::makePlan` and insert ros log)
 
Finally, I confirmed the reason for this issue, it's also caused by the `error-thrown` mechanism defect in `nav2_util`  (in humble branch )

https://github.com/ros-planning/navigation2/blob/150dd6beb50aee502ada8935cee23d7160a40f57/nav2_util/include/nav2_util/simple_action_server.hpp#L294

It seems like another ticket for #4175 and #4180, and my experiments are very similar to how I do it in #4175.

This bug seems fixed in main-branch in our last work, and just not updated into the humble-branch.

SteveMacenski, 2024-04-29:
Can you submit a PR to backport to Humble?

GoesM, 2024-05-03:
It seems there's a restriction on "only being able to submit to the main branch," and I'm not particularly familiar with **cherry-pick** operations... T_T   

Would you be willing to help me?

SteveMacenski, 2024-05-03:
Sure, if you find the git hash that you're interested in either through `git log` or the GitHub UI, you can apply that hash to another branch via

```
git checkout humble
git checkout -b my_branch
git cherry-pick <hash>
```

If there are issues with conflicts, then it'll give you errors that you need to resolve in line of the code. Else, it should be smooth. Then open a PR against that branch. You can ignore the `only being able to submit to the main branch` bot, that is a general rule for general situations. When we're talking about approved specific backports, its OK. That more for people that want to add a new feature or fix a new bug in some branch but that isn't in `main` (so then any new distributions wouldn't get it :cry: ). Since this is already in `main`, this is the backport, which is fine.

GoesM, 2024-05-07:
Thanks a lot ! ^_^   I'd have a try on it ~

closed 2024-05-07 by SteveMacenski