ros-navigation/navigation2 #4291: Launch navigation with namespace

ruibartsegura, 2024-05-01:
## Bug report

**Required Info:**

- Operating System:
  - <Ubuntu 22-04-->
- ROS2 Version:
  - <Humble-->
- Version or commit hash:
  - <ros-humble-nav2-amcl	1.1.14-1jammy.20240407.024740
ros-humble-nav2-behavior-tree	1.1.14-1jammy.20240407.024820
ros-humble-nav2-behaviors	1.1.14-1jammy.20240407.033140
ros-humble-nav2-bringup	1.1.14-1jammy.20240407.042212
ros-humble-nav2-bt-navigator	1.1.14-1jammy.20240407.033140
ros-humble-nav2-collision-monitor	1.1.14-1jammy.20240407.031918
ros-humble-nav2-common	1.1.14-1jammy.20240407.022742
ros-humble-nav2-constrained-smoother	1.1.14-1jammy.20240407.032128
ros-humble-nav2-controller	1.1.14-1jammy.20240407.032126
ros-humble-nav2-core	1.1.14-1jammy.20240407.031918
ros-humble-nav2-costmap-2d	1.1.14-1jammy.20240407.025649
ros-humble-nav2-dwb-controller	1.1.14-1jammy.20240407.033138
ros-humble-nav2-lifecycle-manager	1.1.14-1jammy.20240407.024841
ros-humble-nav2-map-server	1.1.14-1jammy.20240407.024841
ros-humble-nav2-mppi-controller	1.1.14-1jammy.20240407.032145
ros-humble-nav2-msgs	1.1.14-1jammy.20240407.023105
ros-humble-nav2-navfn-planner	1.1.14-1jammy.20240407.032142
ros-humble-nav2-planner	1.1.14-1jammy.20240407.032129
ros-humble-nav2-regulated-pure-pursuit-controller	1.1.14-1jammy.20240407.032148
ros-humble-nav2-rotation-shim-controller	1.1.14-1jammy.20240407.032807
ros-humble-nav2-rviz-plugins	1.1.14-1jammy.20240407.025255
ros-humble-nav2-simple-commander	1.1.14-1jammy.20240407.023753
ros-humble-nav2-smac-planner	1.1.14-1jammy.20240407.032151
ros-humble-nav2-smoother	1.1.14-1jammy.20240407.032225
ros-humble-nav2-theta-star-planner	1.1.14-1jammy.20240407.032248
ros-humble-nav2-util	1.1.14-1jammy.20240407.023837
ros-humble-nav2-velocity-smoother	1.1.14-1jammy.20240407.024846
ros-humble-nav2-voxel-grid	1.1.14-1jammy.20240407.023142
ros-humble-nav2-waypoint-follower	1.1.14-1jammy.20240407.032248
-->
- DDS implementation:
  - 

#### Steps to reproduce issue
<Launch localization with a namespace and then navigation and rviz``` -->
```

```

#### Expected behavior
The map shows in the rvis

#### Actual behavior
The map doesn't show because it doesn't find the rout to the map

Original error: parameter 'yaml_filename' is not initialized

SteveMacenski, 2024-05-01:
Please properly fill in the issue template, include your precise commands and error messages.

ruibartsegura, 2024-05-02:
# Required Info:

    - Operating System:
       Ubuntu 22.04

    - ROS2 Version:
        Humble

    - Version or commit hash:
        ros-humble-nav2-amcl	1.1.14-1jammy.20240407.024740
        ros-humble-nav2-behavior-tree	1.1.14-1jammy.20240407.024820
        ros-humble-nav2-behaviors	1.1.14-1jammy.20240407.033140
        ros-humble-nav2-bringup	1.1.14-1jammy.20240407.042212
        ros-humble-nav2-bt-navigator	1.1.14-1jammy.20240407.033140
        ros-humble-nav2-collision-monitor	1.1.14-1jammy.20240407.031918
        ros-humble-nav2-common	1.1.14-1jammy.20240407.022742
        ros-humble-nav2-constrained-smoother	1.1.14-1jammy.20240407.032128
        ros-humble-nav2-controller	1.1.14-1jammy.20240407.032126
        ros-humble-nav2-core	1.1.14-1jammy.20240407.031918
        ros-humble-nav2-costmap-2d	1.1.14-1jammy.20240407.025649
        ros-humble-nav2-dwb-controller	1.1.14-1jammy.20240407.033138
        ros-humble-nav2-lifecycle-manager	1.1.14-1jammy.20240407.024841
        ros-humble-nav2-map-server	1.1.14-1jammy.20240407.024841
        ros-humble-nav2-mppi-controller	1.1.14-1jammy.20240407.032145
        ros-humble-nav2-msgs	1.1.14-1jammy.20240407.023105
        ros-humble-nav2-navfn-planner	1.1.14-1jammy.20240407.032142
        ros-humble-nav2-planner	1.1.14-1jammy.20240407.032129
        ros-humble-nav2-regulated-pure-pursuit-controller	1.1.14-1jammy.20240407.032148
        ros-humble-nav2-rotation-shim-controller	1.1.14-1jammy.20240407.032807
        ros-humble-nav2-rviz-plugins	1.1.14-1jammy.20240407.025255
        ros-humble-nav2-simple-commander	1.1.14-1jammy.20240407.023753
        ros-humble-nav2-smac-planner	1.1.14-1jammy.20240407.032151
        ros-humble-nav2-smoother	1.1.14-1jammy.20240407.032225
        ros-humble-nav2-theta-star-planner	1.1.14-1jammy.20240407.032248
        ros-humble-nav2-util	1.1.14-1jammy.20240407.023837
        ros-humble-nav2-velocity-smoother	1.1.14-1jammy.20240407.024846
        ros-humble-nav2-voxel-grid	1.1.14-1jammy.20240407.023142
        ros-humble-nav2-waypoint-follower	1.1.14-1jammy.20240407.032248

    DDS implementation:
        ros-humble-rmw-dds-common    1.6.0-2jammy.20240217.051355     amd64

# Steps to reproduce issue

I have two launcher and with both I have the same issue, these are the launchers:

```python
# Copyright (c) 2024 Intelligent Robotics Lab (URJC)
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

# Modified by Juan Carlos Manzanares Serrano
import os

from ament_index_python.packages import get_package_share_directory
from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument, IncludeLaunchDescription
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import SetRemap
[... 451 lines cut ...]
/robot1/global_costmap/costmap_updates
/robot1/initialpose
/robot1/joint_states
/robot1/local_costmap/costmap
/robot1/local_costmap/costmap_updates
/robot1/local_costmap/published_footprint
/robot1/local_plan
/robot1/map
/robot1/map_updates
/robot1/mobile_base/sensors/bumper_pointcloud
/robot1/odom
/robot1/particle_cloud
/robot1/plan
/robot1/robot_description
/robot1/scan
/robot1/sensors/battery_state
/robot1/sensors/core
/robot1/sensors/dock_ir
/robot1/sensors/imu_data
/robot1/sensors/imu_data_raw
/robot1/tf
/robot1/tf_static
/robot1/version_info
/robot1/waypoints
/rosout

padhupradheep, 2024-05-02:
Hey @ruibartsegura, Please check your `multi_robot` directory if your map exists and you have given the correct path to the map in your launch file. This seems to be a customized version of the Nav2 launch and I see this to be not originating at the Nav2 end. Note that I also cannot reproduce the issue locally as well. Therefore closing the issue.

closed 2024-05-02 by padhupradheep

ruibartsegura, 2024-05-02:
The route to the map is correct because when I use the same launcher but without the namespace it work rightly.

padhupradheep, 2024-05-02:
I understand, could you please ask your question at https://robotics.stackexchange.com/ ?

I think you will get the necessary help there.