ros-navigation/navigation2 #4297: VoxelLayer and ObstacleLayer should raycast for invalid pointcloud points padded inf or 0

cemonem, 2024-05-04:
## Feature request

VoxelLayer and ObstacleLayer should optionally raycast for invalid pointcloud points padded with inf or 0. These invalid points can come from sensors (depth cameras etc.) when they do not hit an obstacle (Gazebo's depth camera plugin pads invalid points with inf, for example, or D435 depth camera optionally pads these points with 0)  I don't know if this is already a feature for laserscans with inf_is_valid parameter, but that parameter is not supported for pointcloud data.

#### Feature description
The observation cloud is the filtered version of the pointcloud with global z value between min_obstacle_height and max_obstacle_height here 

https://github.com/open-navigation/navigation2/blob/07fbc50edb75ee97b3650716227caa6925874e84/nav2_costmap_2d/src/observation_buffer.cpp#L135C17-L135C18

The clearing raycast feature only raycasts for points in the observation cloud here, which results in absent obstacles only being cleared if they were previously in front of an existing obstacle. Obviously it should not behave like this. 

https://github.com/open-navigation/navigation2/blob/07fbc50edb75ee97b3650716227caa6925874e84/nav2_costmap_2d/plugins/obstacle_layer.cpp#L652

#### Implementation considerations
This feature could be added as an additional parameter.

SteveMacenski, 2024-05-06:
>  (Gazebo's depth camera plugin pads invalid points with inf, for example, or D435 depth camera optionally pads these points with 0) 

So those points are invalid? They shouldn't be used to raycast then. We need some kind of "hit" to know an obstacle is there to raytrace up-to. Laser scanners have the option for inf is valid because those are active laser sensors and no return is better-chance-than-not total clear space past the bounds of its power rating. 3D sensors are frequently passive or with an abundant reasons why you wouldn't get data back for a particular pixel, so the assumption that `Inf` is a meaningful measurement is mostly untrue in my experience.

cemonem, 2024-05-07:
> 3D sensors are frequently passive or with an abundant reasons why you wouldn't get data back for a particular pixel, so the assumption that `Inf` is a meaningful measurement is mostly untrue in my experience.

Depth cameras might perform worse than laser scanners in some environments, but this is not a reason to exclude such an important feature. People might have to rely on these solutions due to budget reasons. I am not sure it the correct approach to dismiss a feature simply because it might generate false positives due to low quality sensors. This should be up to the developers of the application, they can simply test it and disable it if it performs worse.

cemonem, 2024-05-08:
I really am not convinced this justifies the omission of this feature, but as I am relatively new, I respect your viewpoint. Thank you for the explanation.

closed 2024-05-08 by cemonem

StetroF, 2024-05-09:
Here is the part where obstacle layer handle the inf laserscan data:
`  // Filter positive infinities ("Inf"s) to max_range.
  float epsilon = 0.0001;  // a tenth of a millimeter
  sensor_msgs::msg::LaserScan message = *raw_message;
  for (size_t i = 0; i < message.ranges.size(); i++) {
    float range = message.ranges[i];
    if (!std::isfinite(range) && range > 0) {
      message.ranges[i] = message.range_max - epsilon;
    }
  }`
You can have a try to subscribe the laserscan and camera's pointcloud data. And you will observe that camera will return a empty list rather than a list full of inf value when camera is facing a empty space. Comparing with camera.Laserscan's return data will full of inf value. The code upon show up that you must have a value so that you can change their inf value to max_range.But if the list is empty.It's meaningless to handle them.

franckmunar-bit, 2026-11-02:
We are investigating a general 3D occupancy update / clearing issue with Nav2 VoxelLayer on ROS 2 Jazzy (Raspberry Pi 5, Astra depth camera, PointCloud2). This is **not** an issue limited to eight persistent voxels.

In controlled experiment AST-03.16E.48, the robot remained stationary:
- 0–15 s: baseline, mean 733.82 MARKED voxels
- 15–35 s: suspended object present, mean 734.90
- 35–75 s: object removed, mean 822.67
- 225 VoxelGrid publications; one consecutive change from 883 to 814 MARKED voxels.

Local costmap XY resolution: 0.025 m; voxel Z resolution: 0.05 m, 16 vertical voxels. The PointCloud2 source has marking and clearing enabled, observation_persistence: 0, expected_update_rate: 0, raytrace_max_range: 3.0 m, obstacle_max_range: 2.5 m. The depth-processing path discards invalid returns and subsamples pixels.

We have **not** spatially attributed the extra voxels to the suspended object, so these counts indicate unexpected occupancy evolution, not proof of a specific clearing defect.

We read the discussion here about invalid depth returns not being evidence of free space. We do not propose blindly treating invalid points as free space.

Could maintainers advise on:
1. How to instrument per-update clearing-ray traversal and voxel state transitions before clearing, after clearing, and after marking?
2. How to distinguish insufficient valid clearing rays from immediate re-marking or reuse of stale observations?
3. Whether observation_persistence: 0 can still reuse the most recent cloud in update cycles without a new message?
4. Whether sparse depth-camera sampling can explain persistent marks at this resolution, and what tests best discriminate this?
5. Whether there are known Jazzy-specific fixes or recommended alternatives for reliable dynamic-obstacle removal with depth cameras?

We can share a minimal reproducible dataset, effective YAML and exact package versions after verification. We are seeking guidance for a safe, reproducible diagnosis, not claiming a Nav2 bug. Thank you.

franckmunar-bit, 2026-11-16:
Follow-up on our AST-03.16E investigation (November 2026): We implemented **SpatioTemporalVoxelLayer (STVL)** as an alternative to our original Nav2 VoxelLayer setup on a Raspberry Pi 5 / ROS 2 Jazzy AGV with an Astra depth camera. In supervised stationary tests, the new configuration has worked satisfactorily: the local occupancy representation recovers after the test obstacle is removed, rather than showing the persistent/increasing occupancy that prompted our original question. We are pausing the recurring investigation.

Important limits: we have **not established the exact root cause** of the original E.48 measurements or spatially attributed its additional marks to the suspended object. Full dynamic-navigation and safety validation remains pending. This is a configuration-specific outcome, **not** a claim of a universal fix or a recommendation to treat invalid depth returns as free space.

Thank you to the maintainers and contributors for the technical discussion. We wanted to report the outcome for others investigating similar symptoms.