ros-navigation/navigation2 #4299: Local costmap doesn't detect obstacles untill the robot is too close to the obstacle

HazemAshraf11, 2024-05-05:
**Required Info:**

- Operating System:
  - ubuntu 20.04
- ROS2 Version:
  - Foxy
- Version or commit hash:
  - ros-foxy-navigation2	0.4.7-1focal.20230606.053636
- DDS implementation:
  - rmw_cyclonedds_cpp

#### Steps to reproduce issue

The navigation works with the static map but when I introduce a new obstacle, the LaserScan sees it but the local costmap doesn't react until the robot is too close and it hits the obstacle. the width = 15 and height = 15 so the obstacle should be visible in the local costmap. Here are the parameters I am using. I used an obstacle layer instead of a voxel layer because for some reason I get an error saying (Sensor origin out of map bounds. The costmap can't raytrace for it.) and the local planner doesn't work. The only solution I found was to replace the voxel layer with an obstacle layer.
[image]
[image]

Params File

`amcl:
  ros__parameters:
    use_sim_time: True
    alpha1: 0.2
    alpha2: 0.2
    alpha3: 0.2
    alpha4: 0.2
    alpha5: 0.2
    base_frame_id: "base_footprint"
    beam_skip_distance: 0.5
    beam_skip_error_threshold: 0.9
    beam_skip_threshold: 0.3
    do_beamskip: false
    global_frame_id: "map"
    lambda_short: 0.1
    laser_likelihood_max_dist: 2.0
    laser_max_range: 100.0
    laser_min_range: -1.0
    laser_model_type: "likelihood_field"
    max_beams: 60
    max_particles: 2000
    min_particles: 500
    odom_frame_id: "odom"
    pf_err: 0.05
    pf_z: 0.99
    recovery_alpha_fast: 0.0
    recovery_alpha_slow: 0.0
    resample_interval: 1
    robot_model_type: "differential"
    save_pose_rate: 0.5
    sigma_hit: 0.2
    tf_broadcast: true
    transform_tolerance: 1.0
    update_min_a: 0.2
    update_min_d: 0.25
    z_hit: 0.5
    z_max: 0.05
    z_rand: 0.5
    z_short: 0.05
    scan_topic: scan

amcl_map_client:
  ros__parameters:
    use_sim_time: True

amcl_rclcpp_node:
  ros__parameters:
    use_sim_time: True

bt_navigator:
  ros__parameters:
[... 227 lines cut ...]
    footprint_topic: local_costmap/published_footprint
    cycle_frequency: 10.0
    recovery_plugins: ["back_up", "wait"]
    back_up:
      plugin: "nav2_recoveries/BackUp"
    wait:
      plugin: "nav2_recoveries/Wait"
    global_frame: odom
    robot_base_frame: base_link
    transform_timeout: 0.1
    use_sim_time: true
    simulate_ahead_time: 2.0
    max_rotational_vel: 0.1
    min_rotational_vel: 0.4
    rotational_acc_lim: 3.2

robot_state_publisher:
  ros__parameters:
    use_sim_time: True`

#### Expected behavior
The obstacles within the 15*15 m^2 square around the robot should be visible in the local costmap

#### Actual behavior
They are not visible untill the robot is too close. If they appear and the robot leaves, they become visible in the costmap within that square until the robot get far away.

tonynajjar, 2024-05-05:
You said that the problem was when you use the voxel layer right? I only see the config with the obstacle layer. Can you add the problematic config?

HazemAshraf11, 2024-05-05:
The local costmap doesn't even work when I use a voxel layer, and it gives me this warning "Sensor origin out of map bounds. The costmap can't raytrace for it."
When I use an obstacle layer, the obstacles don't appear in the local costmap until it's too late as you can see in the two pictures.

I am going to use a jetson nano later and it doesn't support Ubuntu 22.04 so I can't use humble.

HazemAshraf11, 2024-05-05:
I am pretty sure the problem is related to the ( <obstacle layer>.<data source>.obstacle_max_range ) parameter. It's default value is 2.5 and indeed the obstacles don't appear in the local costmap if they are more than 2.5 m far from the sensor. After they appear, they stay in this map untill the robot moves and they become out of the 15*15 m^2 square. But when I change this parameter, it doesn't do any thing. it's like it only works with the defined default value.

HazemAshraf11, 2024-05-05:
I found the problem. It's "obstacle_range" not "obstacle_max_range"

padhupradheep, 2024-05-06:
Yepp, you are right @HazemAshraf11 

You can see the PR here: https://github.com/open-navigation/navigation2/pull/2126/files 

It can be also found in foxy to galactic migration guide: https://navigation.ros.org/migration/Foxy.html

can we close the issue noe @HazemAshraf11 ?

closed 2024-05-06 by HazemAshraf11