ros-navigation/navigation2 #4308: Rewritten Yaml nested parameters

T-Lemmel, 2024-05-07:
**Hello, i am trying to use Rewritten yaml inside a launch file for my nav2_parameters, as such :** 

[image]

the robot_base_frame, global_frame and odom_topic are correctly assigned, but **not the nested ones**.
  
#### Expected behavior

```
$ ros2 param get /local_costmap/local_costmap obstacle_layer.scan.topic
String value is: robot_name/scan
```

#### Actual behavior

```
$ ros2 param get /local_costmap/local_costmap obstacle_layer.scan.topic
String value is: /scan
```

#### Additional information

i tried using the root_key argument, like that : 

[image]

**but then my others parameters are not assigned properly and i get this error :** 

[controller_server-8] [ERROR] [1715082825.036608858] [controller_server]: Couldn't load critics! Caught exception: No critics defined for FollowPath

**here is my the relevant part of my param file :** 
```

controller_server:
    ros__parameters:
      use_sim_time: False
      controller_frequency: 20.0
      min_x_velocity_threshold: 0.001
      min_y_velocity_threshold: 0.5
      min_theta_velocity_threshold: 0.001
      failure_tolerance: 0.3
      odom_topic: "odometry"
      scan.topic: "scan"
      progress_checker_plugin: "progress_checker"
      goal_checker_plugin: "goal_checker"
      controller_plugins: ["follow_path"]
     # Progress checker parameters
      progress_checker:
        plugin: "nav2_controller::SimpleProgressChecker"
        required_movement_radius: 0.5
        movement_time_allowance: 10.0
      # Goal checker parameters
      goal_checker:
        plugin: "nav2_controller::SimpleGoalChecker"
        xy_goal_tolerance: 0.25
        yaw_goal_tolerance: 0.25
        stateful: True
      # basic MPC with DWB
      follow_path:
        plugin: "dwb_core::DWBLocalPlanner"
        debug_trajectory_details: True
        min_vel_x: 0.0
        min_vel_y: 0.0
        max_vel_x: 0.26
        max_vel_y: 0.0
        max_vel_theta: 1.0
        min_speed_xy: 0.0
        max_speed_xy: 0.26
        min_speed_theta: 0.0
        acc_lim_x: 2.5
        acc_lim_y: 0.0
        acc_lim_theta: 3.2
        decel_lim_x: -2.5
        decel_lim_y: 0.0
        decel_lim_theta: -3.2
        vx_samples: 20
        vy_samples: 5
        vtheta_samples: 20
        sim_time: 1.7
        linear_granularity: 0.05
        angular_granularity: 0.025
        transform_tolerance: 0.2
        xy_goal_tolerance: 0.25
        trans_stopped_velocity: 0.25
        short_circuit_trajectory_evaluation: True
        stateful: True
        critics: ["RotateToGoal", "Oscillation", "BaseObstacle", "GoalAlign", "PathAlign", "PathDist", "GoalDist"]
        BaseObstacle.scale: 0.2
        PathAlign.scale: 32.0
        PathAlign.forward_point_distance: 0.1
        GoalAlign.scale: 24.0
        GoalAlign.forward_point_distance: 0.1
        PathDist.scale: 32.0
        GoalDist.scale: 24.0
        RotateToGoal.scale: 32.0
        RotateToGoal.slowing_factor: 5.0
        RotateToGoal.lookahead_time: -1.0

```
Does anyone have an idea ? 
Thanks in advance,
Tom

Ps : using ubuntu 22.04 and ros2 humble

SteveMacenski, 2024-05-07:
You just need to fully set the path. Either you can not set any portion of the path of a variable and change **all** that match or set the full path and change only the one. It does not support in between. Specify the full `/local_costmap/local_costmap obstacle_layer.scan.topic` or take a look at the implementation and I welcome a PR to add that feature potentially :-)

https://github.com/open-navigation/navigation2/blob/12c31c69666fc8e5642ca9db041be8f3faf47684/nav2_common/nav2_common/launch/rewritten_yaml.py#L35

T-Lemmel, 2024-05-07:
hooo, ok thanks 👍 

I will not be able to test it until next week, i'll make sure to close the issue then if it solves the problem.
But even looking at the implementation, i don't get why it would work with other parameters such as `odom_topic` (see above), wouldn't i need to specify `controller_server/odom_topic` ?

SteveMacenski, 2024-05-07:
Because the leading `/` makes it a global match not a partial match

T-Lemmel, 2024-05-13:
I'm sorry but i can't seem to make it work, i'm not sure of the syntaxe though, should this one work ? 

[image]

I tried pretty much all combinations of where to place '/' & '.' but none worked

SteveMacenski, 2024-05-13:
Ah, I think because you missed a namespace for the absolute paths: https://github.com/ros-navigation/navigation2/blob/main/nav2_system_tests/src/system/test_system_launch.py#L63

`ros__parameters` exists (and is somewhat annoying)

T-Lemmel, 2024-05-15:
So would `'/local_costmap/local_costmap.ros__parameters.obstacle_layer.scan.topic': robot_name + '/scan',` be the correct one ? It still doesn't work, I'm sorry it's probably a stupid error but can you send the full line please ?