ros-navigation/navigation2 #4315: MPPI dynamic parameter not working with vx_max

Huyhust13, 2024-05-08:
## Bug report

**Required Info:**

- Operating System: Ubuntu 22.04
- ROS2 Version: Humble
- Version or commit hash: 150dd6beb50aee502ada8935cee23d7160a40f57
- DDS implementation: mw_cyclonedds_cpp

#### Expected behavior

I set vx_max = 0.5 in file parameter, then when it running, I change it to 0.3. I use `parameter get` to show param, it shows vx_max = 0.3 but the robot runs with vx = 0.5.

Huyhust13, 2024-05-08:
Thank for your replying @SteveMacenski. 
I'll do test.

Huyhust13, 2024-05-09:
It's correct, it works now @SteveMacenski 
Thank you very much for your help.

closed 2024-05-09 by SteveMacenski

YoungRicky7, 2024-12-11:
> ## Bug report
> **Required Info:**
> 
> * Operating System: Ubuntu 22.04
> * ROS2 Version: Humble
> * Version or commit hash: [150dd6b](https://github.com/ros-navigation/navigation2/commit/150dd6beb50aee502ada8935cee23d7160a40f57)
> * DDS implementation: mw_cyclonedds_cpp
> 
> #### Expected behavior
> I set vx_max = 0.5 in file parameter, then when it is running, I change the vx_max to 0.3. I use `parameter get` to show param, it shows vx_max = 0.3 but the robot runs with vx = 0.5.

How do I change the speed while robot is running? it seems no matter what parameters that I change the speed of robot stays around 0.26m/s