ros-navigation/navigation2 #4318: Can MPPI follow the global path that represents reversing movement of the robot?

bojan-otsaw, 2024-05-09:
Hi ,

I have a global path planner that outputs a global path, and depending of the scenario in same cases the robot should move forward and follow the forward path but in  same scenario even the goal is to move forward path, it needs  to move reverse back and find another alternative path. 

Given this context, my question is, can the current implementation of MPPI controller follow a global path like this? If not, what will be the best way to modify it to achieve this?

Thank you very much.

SteveMacenski, 2024-05-09:
Yes! See @doisyg’s ROSCon talk on this matter: https://vimeo.com/879000809/b32f8ebec4

bojan-otsaw, 2024-05-10:
But my robot is AMR not and AGV . Can not turn on the spot. So is still possible?

SteveMacenski, 2024-05-10:
Please see the video

closed 2024-05-10 by SteveMacenski