ros-navigation/navigation2 #4333: [Humble] nav2-mppi-controller 1.1.14-1 on arm64 installed from binaries crashes on launch

jimdinunzio, 2024-05-13:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 22.04 Jammy arm64

- ROS2 Version:
Humble, installed from binaries

- Version or commit hash:
ros-humble-nav2-mppi-controller/jammy,now 1.1.14-1jammy.20240407.044830 arm64 [installed]

- DDS implementation:
default

#### Steps to reproduce issue
[navigation.txt](https://github.com/ros-navigation/navigation2/files/15288486/navigation.txt)

```
ros2 launch nav2_bringup navigation_launch.py params_file:=navigation.txt 
```

#### Expected behavior
nav2-mppi-controller works as expected

#### Actual behavior
nav2-mppi-controller crashes on launch

[nav2-mppi-controller.log](https://github.com/ros-navigation/navigation2/files/15288465/nav2-mppi-controller.log)

#### Additional information

When built from source commit 8f097af08ced738f4a0797d941a60d834ebc8d80 nav2-mppi-controller does not crash on start up

SteveMacenski, 2024-05-14:
@pepisg do you see this? I believe you're also on Humble ARM if I recall correctly. 

@jimdinunzio please follow the thread in this topic for some possibly relevant information and context about steps forward here: https://github.com/ros-navigation/navigation2/issues/3767. We ran into a similar situation where build farm binaries were causing a problem that locally compiled did not due to build flags available on the cloud provider jobs that weren't on the local systems.

1.  Please compile with debug flags according to the docs here (https://navigation.ros.org/tutorials/docs/get_backtrace.html) and give me the traceback to know more about the crash
2. What happens if you sudo apt purge ros-humble-nav2-mppi-controller and reinstall it?
3. Run `cat /proc/cpuinfo` and give me the output. Compare with https://build.ros2.org/job/Hbin_uJ64__nav2_mppi_controller__ubuntu_jammy_amd64__binary/26/consoleFull#console-section-2 and let me know if there are any notable changes

pepisg, 2024-05-14:
I'm running iron on ARM with nav2 built from source.

I don't use the binaries so in my case the controller was not crashing, but I found the behavior to be quite erratic: the generated trajectories did not seem to go in accordance with how the  critics  should score them (sometimes they wouldn't stick to the path even when only PathFollow was in use), and also the optimal trajectory did not seem to be sampled from the generated trajectories. 

@jimdinunzio does MPPI work normally for you when building from source from the commit you referenced?

SteveMacenski, 2024-05-14:
@pepisg if you did run the binaries real quick just to see, does it crash?

pepisg, 2024-05-16:
Hey! Sorry for the delay, will do next week when I can easily access a robot and let you know

SteveMacenski, 2024-05-16:
Thanks! Your time is appreciated :-)

pepisg, 2024-05-17:
Hey! actually I had time do check it out today. With the params file provided by @jimdinunzio and installing ros-iron-nav2-mppi-controller, the controller did not crash launching nav2_bringup's navigation.launch.py: 

```
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-05-17-15-42-13-898346-0690693-2326
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [controller_server-1]: process started with pid [2329]
[INFO] [smoother_server-2]: process started with pid [2331]
[INFO] [planner_server-3]: process started with pid [2333]
[INFO] [behavior_server-4]: process started with pid [2335]
[INFO] [bt_navigator-5]: process started with pid [2337]
[INFO] [waypoint_follower-6]: process started with pid [2339]
[INFO] [velocity_smoother-7]: process started with pid [2341]
[INFO] [collision_monitor-8]: process started with pid [2343]
[INFO] [lifecycle_manager-9]: process started with pid [2345]
[behavior_server-4] [1715960537.355056146][INFO][printLifecycleNodeNotification][122]: 
[behavior_server-4]     behavior_server lifecycle node launched. 
[behavior_server-4]     Waiting on external lifecycle transitions to activate
[behavior_server-4]     See https://design.ros2.org/articles/node_lifecycle.html for more information.
[collision_monitor-8] [1715960537.358836709][INFO][printLifecycleNodeNotification][122]: 
[collision_monitor-8]   collision_monitor lifecycle node launched. 
[collision_monitor-8]   Waiting on external lifecycle transitions to activate
[collision_monitor-8]   See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-6] [1715960537.379527286][INFO][printLifecycleNodeNotification][122]: 
[waypoint_follower-6]   waypoint_follower lifecycle node launched. 
[waypoint_follower-6]   Waiting on external lifecycle transitions to activate
[waypoint_follower-6]   See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-6] [1715960537.414301536][INFO][WaypointFollower][35]: Creating
[velocity_smoother-7] [1715960537.389353030][INFO][printLifecycleNodeNotification][122]: 
[velocity_smoother-7]   velocity_smoother lifecycle node launched. 
[velocity_smoother-7]   Waiting on external lifecycle transitions to activate
[velocity_smoother-7]   See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-5] [1715960537.609612103][INFO][printLifecycleNodeNotification][122]: 
[bt_navigator-5]        bt_navigator lifecycle node launched. 
[bt_navigator-5]        Waiting on external lifecycle transitions to activate
[bt_navigator-5]        See https://design.ros2.org/articles/node_lifecycle.html for more information.
[lifecycle_manager-9] [1715960537.633238787][INFO][LifecycleManager][38]: Creating
[bt_navigator-5] [1715960537.673909123][INFO][BtNavigator][37]: Creating
[planner_server-3] [1715960537.705070851][INFO][printLifecycleNodeNotification][122]: 
[planner_server-3]      planner_server lifecycle node launched. 
[planner_server-3]      Waiting on external lifecycle transitions to activate
[planner_server-3]      See https://design.ros2.org/articles/node_lifecycle.html for more information.
[smoother_server-2] [1715960537.722996114][INFO][printLifecycleNodeNotification][122]: 
[smoother_server-2]     smoother_server lifecycle node launched. 
[smoother_server-2]     Waiting on external lifecycle transitions to activate
[smoother_server-2]     See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-3] [1715960537.772954921][INFO][PlannerServer][50]: Creating
[controller_server-1] [1715960537.784534124][INFO][printLifecycleNodeNotification][122]: 
[controller_server-1]   controller_server lifecycle node launched. 
[controller_server-1]   Waiting on external lifecycle transitions to activate
[controller_server-1]   See https://design.ros2.org/articles/node_lifecycle.html for more information.
[smoother_server-2] [1715960537.821870767][INFO][SmootherServer][41]: Creating smoother server
[lifecycle_manager-9] [1715960537.829985518][INFO][message][534]: Creating and initializing lifecycle service clients
[planner_server-3] [1715960537.887772631][INFO][printLifecycleNodeNotification][122]: 
[planner_server-3]      global_costmap lifecycle node launched. 
[planner_server-3]      Waiting on external lifecycle transitions to activate
[planner_server-3]      See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-3] [1715960537.939146257][INFO][init][109]: Creating Costmap
[lifecycle_manager-9] [1715960537.939794831][INFO][message][534]: Starting managed nodes bringup...
[lifecycle_manager-9] [1715960537.940532786][INFO][message][534]: Configuring controller_server
[controller_server-1] [1715960538.008648355][INFO][ControllerServer][49]: Creating controller server
[controller_server-1] [1715960538.094687522][INFO][printLifecycleNodeNotification][122]: 
[controller_server-1]   local_costmap lifecycle node launched. 
[controller_server-1]   Waiting on external lifecycle transitions to activate
[controller_server-1]   See https://design.ros2.org/articles/node_lifecycle.html for more information.
[controller_server-1] [1715960538.122432672][INFO][init][109]: Creating Costmap
[controller_server-1] [1715960538.136501785][INFO][on_configure][86]: Configuring controller interface
[controller_server-1] [1715960538.136751844][INFO][on_configure][88]: getting progress checker plugins..
[... cut ...]
[collision_monitor-8] [1715960591.910985512][ERROR][configureSources][374]: Error while getting parameters: parameter 'observation_sources' is not initialized
[lifecycle_manager-9] [1715960591.916989827][ERROR][changeStateForNode][245]: Failed to change state for node: collision_monitor
[lifecycle_manager-9] [1715960591.918109848][ERROR][startup][312]: Failed to bring up all requested nodes. Aborting bringup.
```

The collision monitor didn't activate because there don't seem to be CM params in the params file, but otherwise it looked OK

SteveMacenski, 2024-05-17:
@jimdinunzio can you try iron and let me know if you run into an issue? This is a really good datapoint @pepisg, thanks!

david-wb, 2024-05-23:
I'm not sure if it's related, but for me, inclusion of "Obstacle Critic" plugin causes errors. When I leave it out, it works normally. I'm on Humble.
```
[component_container_isolated-5] [INFO] [1716456486.788960038] [controller_server]: Controller Server has goal_checker  goal checkers available.
[component_container_isolated-5] [INFO] [1716456486.790657891] [controller_server]: Created controller : FollowPath of type nav2_mppi_controller::MPPIController
[component_container_isolated-5] [INFO] [1716456486.796058061] [controller_server]: Controller period is equal to model dt. Control sequence shifting is ON
[component_container_isolated-5] [INFO] [1716456486.799306979] [controller_server]: ConstraintCritic instantiated with 1 power and 4.000000 weight.
[component_container_isolated-5] [INFO] [1716456486.799370389] [controller_server]: Critic loaded : mppi::critics::ConstraintCritic
[component_container_isolated-5] [INFO] [1716456486.801564289] [controller_server]: InflationCostCritic instantiated with 1 power and 500.000000 / 0.015000 weights. Critic will collision check based on footprint cost.
[component_container_isolated-5] [INFO] [1716456486.801615601] [controller_server]: Critic loaded : mppi::critics::CostCritic
[component_container_isolated-5] Magick: abort due to signal 11 (SIGSEGV) "Segmentation Fault"...
[ign-1] [GUI] [Wrn] [Model.hh:98] Unable to deserialize sdf::Model
[ERROR] [component_container_isolated-5]: process has died [pid 19168, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_isolated --ros-args --log-level info --ros-args -r __node:=nav2_container --params-file /tmp/launch_params_7ihy6wdt --params-file /tmp/launch_params_jpp6nfoq -r /tf:=tf -r /tf_static:=tf_static'].
[INFO] [namoros-6]: process started with pid [19299]
```

SteveMacenski, 2024-05-23:
@jimdinunzio Any update?

SteveMacenski, 2024-05-29:
Closing due to lack of response. Happy to open if someone can reproduce.

closed 2024-05-29 by SteveMacenski