ros-navigation/navigation2 #4341: Error codes in NavigateToPose/NavigateThroughPoses

ajtudela, 2024-05-14:
Moving the conversation here to discuss the possibility of adding error codes for NavigateToPose/NavigateThroughPoses actions in a similar way as the other servers (planner, controller, etc.)

I'm thinking of a use case where you call the NavigateToPose action from a high-level system, and I want to know why the robot can't reach the goal from the action client itself.  It would be nice to add or populate some error codes from the "subservers" like tf_error, timeout, invalid path, etc.

aosmw, 2024-06-20:
@SteveMacenski  what is your appetite for the interface change of adding a `string error_msg` to go with the `uint16 error_code` in the nav2_msgs/action/*.action result definitions?

That would allow a pathway to propagate failure messages with details out to the action client.

For backward compatibility would you prefer new actions?

ajtudela, 2024-06-20:
Adding a ``string error_message`` in addition to the current ``int error_code`` it isn't hard to implement. That would help people who are not familiar with error codes and not have to check what each error code means. And it would be a first step to have the error codes for the navigators actions.

I think storing in the blackboard the error code in enum format will simplify the logic for populate it at higher level. For example, storing the last error code ocurred (from the BT node that fails).

Regarding backward compatibility, there are no error codes in humble so, if this feature is implemented, can't be backported to humble.

aosmw, 2024-06-21:
> How do you imagine using those messages?

My use case for error messages is to be able to bubble up reasons that the behavior tree action aborted to a user interface.  I foresee it helping a wider spectrum of people being able to reason about behavior and ultimately lead to them coming up with ideas for behavior tree improvements.

I think, right now, that it is reasonable that the error message strings are not put into the behavior tree blackboard.  (Limited experience so I could be easily swayed on this front).  I agree that behavior tree logic that responds to errors is best done against  against enums, rather than string matching.  If the behavior tree is complete/complex enough to handled the errors encountered then the noise of error messages is not necessarily interesting.    

On the other hand a dedicated published log of error messages interposed with successful behavior transitions could be very useful for explain-ability.  Maybe putting the error message strings into the blackboard would facilitate that. Maybe a ReportError  action or ?decorator? could control what messages are deemed important enough to make it into such a log. 

>> For backward compatibility would you prefer new actions?
> What would you have in mind? 

I agree that dual action definitions like NavigateToPoseWithErrorString/NavigateThroughPosesWithErrorString or NavigateToPose2/NavigateThroughPoses2 would be distateful and either be a naive cut and paste, or possibly use some template logic to detect the presence of and populate an error_message string in an action Result class.

I personally, build from source on humble and track close to main periodically by back porting it to humble myself.  I would like to change the existing action result messages to include the error_message string and can accept that this will not be available in humble/iron.  

You have a wider audience and hence my check in with your appetite for such a change.   My read is that it would not be a back ported change to the interface.

SteveMacenski, 2024-06-21:
> I think, right now, that it is reasonable that the error message strings are not put into the behavior tree blackboard. (Limited experience so I could be easily swayed on this front).

We can always add it later too. We can / probably should follow the same patterns we use for the error IDs just now with messages.  How are the messages proping to your User Interface then?

> I personally, build from source on humble and track close to main periodically by back porting it to humble myself. 

That seems like a reasonable solution for this. Though, if you can add in the `error_msg` to the interfaces **ASAP** and open a PR, I can merge that in before we do a Jazzy cut. That would let this feature be ship-able with Jazzy later on since that's the major ABI breaking change we need to be aware of for backporting. Then, a follow up PR to implement can be done on the schedule that makes the most sense for you. 

Would you be able to do that ASAP? I'm looking to cut Jazzy today... but I could delay to Monday or Tuesday if that makes a difference. 

>  My read is that it would not be a back ported change to the interface.

Correct, due to API/ABI stability of existing users, can't be breaking people's robots out in the wild

aosmw, 2024-06-22:
I just saw this.  I'll have a go now.

aosmw, 2024-06-22:
See https://github.com/ros-navigation/navigation2/pull/4459

aosmw, 2024-06-22:
Here is a second pull request with a first pass at populating the error_msg, and error_code where it was missing (FollowWaypoints)

See https://github.com/ros-navigation/navigation2/pull/4460

It builds but is untested and probably incomplete so I left it as draft.   
You may have comments on the style.

aosmw, 2024-06-23:
Added a second pass to use the error_code and error_msg expected in each action in nav2_simple_navigator.

Not run them yet.

aosmw, 2024-06-23:
> @SteveMacenski  wrote: a while back.
> How are the messages proping to your User Interface then?

Finally figured out that "proping" probably means propagating.

The error messages are currently not propagating to my user interface.  Hence I went searching for why, came across this issue and also felt the pain point.  With these changes I will be able to get the errors to our user interface.

aosmw, 2024-06-23:
And now I am at the nav2_bt_navigator/src/navigators level and staring down the catch(...) and wondering if that is my target.

https://github.com/ros-navigation/navigation2/blob/9fbae3e66ecc3b7315ca7cbb070468a564f5e111/nav2_bt_navigator/src/navigators/navigate_through_poses.cpp#L167

aosmw, 2024-06-24:
Now I am thinking that ResultStatus needs to be the one carrying the error message. 
https://github.com/ros-navigation/navigation2/blob/9fbae3e66ecc3b7315ca7cbb070468a564f5e111/nav2_behaviors/include/nav2_behaviors/timed_behavior.hpp#L52

SteveMacenski, 2024-06-24:
> The error messages are currently not propagating to my user interface. Hence I went searching for why, came across this issue and also felt the pain point. With these changes I will be able to get the errors to our user interface.

This is why I asked how you planned to propagate the error messages through to your interface. You'll see that the individual BT nodes that have actions with error codes populate in the BT blackboard the error codes which are aggregated by the BT Navigator into the action's result message for the highest priority failure number (but realistically there should only be 1 at at time, though possible to have multiple if multiple things go wrong simultaneously).  

If you want Nav2Pose and NavThruPoses to have the error message, then it'll need to propogate through the tree into the Navigator's action result callback. If you track down the `error_id` logic in the BTs, you can see how we did that. It doesn't have anything to do with the feedback exception catching 

> Now I am thinking that ResultStatus needs to be the one carrying the error message.

That's for the behavior server, wrong place. Example
- https://github.com/ros-navigation/navigation2/blob/9fbae3e66ecc3b7315ca7cbb070468a564f5e111/nav2_behavior_tree/include/nav2_behavior_tree/plugins/action/compute_path_to_pose_action.hpp#L88
-

ewak, 2025-01-20:
It's ready from my end.

SteveMacenski, 2025-01-28:
I'm a little confused, I believe this is @aosmw's PR and the last that they mentioned, it was not ready yet: https://github.com/ros-navigation/navigation2/pull/4460#issuecomment-2547305951

I'm not sure which PR that @ewak has that does this (and if its public?)

ewak, 2025-01-28:
Sorry, @ewak == @aosmw.  

I am currently between jobs (Melbourne, Australia for anyone interested). Going forward will just use my own account/identity.  I can still push to the aosmw PR to keep all the comments together or I can push a new one.  I don't have any further changes to make.  There are *ALOT* of comments and threads on that PR.  From my perspective they are all resolved. I assume I leave it to @SteveMacenski to mark comments resolved.  

If you think you are ready to take on the disruption the changes to nav2_msgs then I will rebase it on main again.

Or do you want a fresh PR from @ewak?

aosmw, 2025-01-28:
For what its worth, here is a confirmation that @ewak == @aosmw

ewak, 2025-02-02:
@tonynajjar I have rebased the error_msg work (https://github.com/ros-navigation/navigation2/pull/4460) onto origin/main again.  Are you or anyone else in a position to/willing to try it out?

SteveMacenski, 2025-02-03:
> Sorry, @ewak == @aosmw.

Ah, got it! I will add it back to my queue to look at and re-review in detail. But, if you're telling me that you've tested this and have it working (i.e. error codes from the servers are shown to propagate all the way up and get reset back to `NONE` between missions) in your application, I can't imagine I'd have much more to say other than nitpicks or small edge case gaps

SteveMacenski, 2025-02-13:
PR to be merged imminently within the next couple of days / weeks #4460

ajtudela, 2025-02-13:
Awesome work @ewak !!

closed 2025-03-04 by SteveMacenski

SteveMacenski, 2025-04-17:
It is an API breaking change across a large number of individual packages and nodes, this is not going to be backportable. However, Jazzy now builds using the `main` branch so you can use the `main` branch directly and obtain it!