ros-navigation/navigation2 #4343: Hi, I am building the package of nav2_waypoint_follower and facing that errors,

Muteeb125, 2024-05-15:
Starting >>> nav2_waypoint_follower
--- stderr: nav2_waypoint_follower                              
/usr/bin/ld: ../libwaypoint_follower_core.so: undefined reference to `rosidl_action_type_support_t const* rosidl_typesupport_cpp::get_action_type_support_handle<nav2_msgs::action::FollowGPSWaypoints>()'
/usr/bin/ld: ../libwaypoint_follower_core.so: undefined reference to `nav2_util::setSoftRealTimePriority()'
collect2: error: ld returned 1 exit status
gmake[2]: *** [test/CMakeFiles/test_dynamic_parameters.dir/build.make:857: test/test_dynamic_parameters] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:345: test/CMakeFiles/test_dynamic_parameters.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: libwaypoint_follower_core.so: undefined reference to `rosidl_action_type_support_t const* rosidl_typesupport_cpp::get_action_type_support_handle<nav2_msgs::action::FollowGPSWaypoints>()'
/usr/bin/ld: libwaypoint_follower_core.so: undefined reference to `nav2_util::setSoftRealTimePriority()'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/waypoint_follower.dir/build.make:855: waypoint_follower] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:182: CMakeFiles/waypoint_follower.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< nav2_waypoint_follower [0.84s, exited with code 2]

Summary: 9 packages finished [3.33s]
  1 package failed: nav2_waypoint_follower
  1 package had stderr output: nav2_waypoint_follower
[image]

padhupradheep, 2024-05-15:
Hey, could you please fill in the template that was given? We mainly require the Distro and the Nav2 version that you are using.

SteveMacenski, 2024-05-15:
Please reopen filling in the template properly with the relevant information.

closed 2024-05-15 by SteveMacenski