ros-navigation/navigation2 #4345: New "Configure" transition for the LifecycleManager

jonipol, 2024-05-15:
Hello!

## Feature request

The current implementation of the LifecycleManager only allows to transition all the nodes from _unconfigured_ to _active_ which causes some issues in certain cases at startup for the system we are working on. If one would add the transition from _unconfigured_ to _inactive_ as "CONFIGURE" would you approve such contribution? Or is there some problem with that which I have not been able to see?

#### Feature description

Add a CONFIGURE transition to the LifecycleManager which transitions all the managed nodes from _unconfigured_ to _inactive_ state.

#### Implementation considerations

Requires:
- Adding a new command to _ManageLifecycleNodes.srv_ eg. `uint8 CONFIGURE = 5`
- Handle the case in the lifecycle_manager.cpp

SteveMacenski, 2024-05-15:
Seems reasonable enough - take a look at the lifecycle manager and lifecycle manager client, I think that you'll see its easy that we can exposure the full spectrum of lifecycle states in addition to the "composed" transitions that we do for a 1-button startup. I think that would be a great contribution and I would be happy to review / merge that PR :smile: 

You'd just have to call the manage nodes service 2x for bring up / shutdown using the rclcpp lifecycle transition types rather than our "startup" and "shutdown". 

Just make sure to expose them all so that you can bring up and shut down through the appropriate state transitions -- not just configure, but also activate, deactivate, etc.

SteveMacenski, 2024-05-28:
PR to be merged imminently

closed 2024-05-28 by SteveMacenski