ros-navigation/navigation2 #4350: Nav2 Stalling on Multiple Robots?

WallerJonathan, 2024-05-16:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 22.04 Jammy Jellyfish
- ROS2 Version:
  - ROS2 Humble
- Version or commit hash:
  - Latest Version
- DDS implementation:
  - Default

#### Steps to reproduce issue
This is the code I am currently looking at (everything else in the file is not run yet)
```
def main():
    rclpy.init()

    naviturt = BasicNavigator()

    # Set initial pose
    xStart = float(input("Starting x position (float): "))
    yStart = float(input("Starting y position (float): "))
    init_pose = naviturt.getPoseStamped([xStart, yStart], TurtleBot4Directions.NORTH)
    naviturt.setInitialPose(init_pose)

    # Wait until Nav2 starts up
    naviturt.waitUntilNav2Active()
```
1. Make into a build
2. Use namespacing to make the first robot (I am using the turtlebot4 gazebo simulator for this) ```ros2 launch turtlebot4_ignition_bringup turtlebot4_ignition.launch.py namespace:=/robot1 localization:=true rviz:=true```
3. Use namespacing to make the second robot (again I am using the turtlebot4 gazebo simulator for this) ```ros2 launch turtlebot4_ignition_bringup turtlebot4_spawn.launch.py namespace:=/robot2 localization:=true rviz:=true```
4. Run the build file; I ran ```ros2 run my_nav my_nav_py_node --ros-args -r __ns:=/robot2```

#### Expected behavior
I expect the second robot to wait until nav2 is active, then start with the tasks

#### Actual behavior
It freezes up while waiting for Nav2. The output runs forever unless you use a keyboard interrupt:

```Starting x position (float): 1
Starting y position (float): 0
[INFO] [1715887737.224566729] [robot2.basic_navigator]: Publishing Initial Pose
[INFO] [1715887740.263714789] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1715887741.267756369] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1715887742.272033874] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1715887743.276700730] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1715887744.280826984] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1715887745.285144201] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1715887746.289668470] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1715887747.293806072] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1715887748.298023824] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1715887749.301946025] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1715887750.306116626] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1715887751.310751901] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1715887752.315370022] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1715887753.319347344] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
[INFO] [1715887754.324028083] [robot2.basic_navigator]: bt_navigator/get_state service not available, waiting...
^CTraceback (most recent call last):
  File "/home/biorobotics/turtlebot4_ws/install/my_nav/lib/my_nav/my_nav_py_node", line 33, in <module>
    sys.exit(load_entry_point('my-nav', 'console_scripts', 'my_nav_py_node')())
  File "/home/biorobotics/turtlebot4_ws/build/my_nav/my_nav/my_nav_py_node.py", line 72, in main
    naviturt.waitUntilNav2Active()
  File "/opt/ros/humble/lib/python3.10/site-packages/nav2_simple_commander/robot_navigator.py", line 316, in waitUntilNav2Active
    self._waitForNodeToActivate(navigator)
  File "/opt/ros/humble/lib/python3.10/site-packages/nav2_simple_commander/robot_navigator.py", line 537, in _waitForNodeToActivate
    while not state_client.wait_for_service(timeout_sec=1.0):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 177, in wait_for_service
    time.sleep(sleep_time)
KeyboardInterrupt
[ros2run]: Interrupt
```

#### Additional information
I am thinking this may be a namespacing problem? The thing I can't figure out is why it seems to work sometimes with the first robot spawned but never the second. Also, I do not think this is a turtlebot4 specific problem, but if it is please let me know!

Thanks!

SteveMacenski, 2024-05-16:
> [robot2.basic_navigator]: bt_navigator/get_state service not available

Almost certainly. You can see that `bt_navigator` isn't namespaced. What if you pass into `waitUntilNav2Active` the namespaced navigator/localizer types? They are used in the implementation.

```

    def waitUntilNav2Active(self, navigator='bt_navigator', localizer='amcl'):
        """Block until the full navigation system is up and running."""
        if localizer != 'robot_localization':  # non-lifecycle node
            self._waitForNodeToActivate(localizer)
        if localizer == 'amcl':
            self._waitForInitialPose()
        self._waitForNodeToActivate(navigator)
        self.info('Nav2 is ready for use!')
        return
```

If that fixes things, then we could store the namespace in the constructor and use it here automatically. Should be easy:

```
self.namespace = namespace 

... 

 self._waitForNodeToActivate(namespace + '/' + localizer)
 self._waitForNodeToActivate(namespace + '/' + navigator)

```

WallerJonathan, 2024-05-17:
Thanks for the reply!

Unfortunately I don't think it worked (unless I did it wrong). Here is the modified code:
```
def main():
    rclpy.init()

    naviturt = BasicNavigator()

    # Set initial pose
    xStart = float(input("Starting x position (float): "))
    yStart = float(input("Starting y position (float): "))
    init_pose = naviturt.getPoseStamped([xStart, yStart], TurtleBot4Directions.NORTH)
    naviturt.setInitialPose(init_pose)

    # Wait until Nav2 starts up
    naviturt.waitUntilNav2Active(navigator='/robot2/bt_navigator', localizer='/robot2/amcl')
```

One of my runs gave me this as the terminal output:
```
ubuntu@unity:~/tb4_ws$ ros2 run my_nav my_nav_py_node --ros-args -r __ns:=/robot2
Starting x position (float): 0
Starting y position (float): 1
[INFO] [1715949631.921310175] [robot2.basic_navigator]: Publishing Initial Pose
[INFO] [1715949634.957696762] [robot2.basic_navigator]: /robot2/bt_navigator/get_state service not available, waiting...
[INFO] [1715949635.961778416] [robot2.basic_navigator]: /robot2/bt_navigator/get_state service not available, waiting...
[INFO] [1715949636.965818513] [robot2.basic_navigator]: /robot2/bt_navigator/get_state service not available, waiting...
[INFO] [1715949637.969824188] [robot2.basic_navigator]: /robot2/bt_navigator/get_state service not available, waiting...
[INFO] [1715949638.974061734] [robot2.basic_navigator]: /robot2/bt_navigator/get_state service not available, waiting...
[INFO] [1715949639.977766634] [robot2.basic_navigator]: /robot2/bt_navigator/get_state service not available, waiting...
[INFO] [1715949640.981943741] [robot2.basic_navigator]: /robot2/bt_navigator/get_state service not available, waiting...
[INFO] [1715949641.986084632] [robot2.basic_navigator]: /robot2/bt_navigator/get_state service not available, waiting...
[INFO] [1715949642.990370890] [robot2.basic_navigator]: /robot2/bt_navigator/get_state service not available, waiting...
[INFO] [1715949643.995334885] [robot2.basic_navigator]: /robot2/bt_navigator/get_state service not available, waiting...
[INFO] [1715949644.999524696] [robot2.basic_navigator]: /robot2/bt_navigator/get_state service not available, waiting...
^CTraceback (most recent call last):
  File "/home/ubuntu/tb4_ws/install/my_nav/lib/my_nav/my_nav_py_node", line 33, in <module>
    sys.exit(load_entry_point('my-nav', 'console_scripts', 'my_nav_py_node')())
  File "/home/ubuntu/tb4_ws/build/my_nav/my_nav/my_nav_py_node.py", line 72, in main
    naviturt.waitUntilNav2Active(navigator='/robot2/bt_navigator', localizer='/robot2/amcl')
  File "/opt/ros/humble/lib/python3.10/site-packages/nav2_simple_commander/robot_navigator.py", line 316, in waitUntilNav2Active
    self._waitForNodeToActivate(navigator)
  File "/opt/ros/humble/lib/python3.10/site-packages/nav2_simple_commander/robot_navigator.py", line 537, in _waitForNodeToActivate
    while not state_client.wait_for_service(timeout_sec=1.0):
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/client.py", line 177, in wait_for_service
    time.sleep(sleep_time)
KeyboardInterrupt
[ros2run]: Interrupt
```
I ran the exact same commands again and I got a slightly different output that looks like this:
```
ubuntu@unity:~/tb4_ws$ ros2 run my_nav my_nav_py_node --ros-args -r __ns:=/robot2
Starting x position (float): 0
Starting y position (float): 1
[INFO] [1715950555.372274157] [robot2.basic_navigator]: Publishing Initial Pose
[INFO] [1715950556.437064355] [robot2.basic_navigator]: /robot2/amcl/get_state service not available, waiting...
[INFO] [1715950557.439939156] [robot2.basic_navigator]: /robot2/amcl/get_state service not available, waiting...
[INFO] [1715950561.209691728] [robot2.basic_navigator]: robot2/bt_navigator/get_state service not available, waiting...
[INFO] [1715950562.213891973] [robot2.basic_navigator]: robot2/bt_navigator/get_state service not available, waiting...
[INFO] [1715950563.218917243] [robot2.basic_navigator]: robot2/bt_navigator/get_state service not available, waiting...
[INFO] [1715950564.223296188] [robot2.basic_navigator]: robot2/bt_navigator/get_state service not available, waiting...
[INFO] [1715950565.227448403] [robot2.basic_navigator]: robot2/bt_navigator/get_state service not available, waiting...
^CTraceback (most recent call last):
  File "/home/ubuntu/tb4_ws/install/my_nav/lib/my_nav/my_nav_py_node", line 33, in <module>
    sys.exit(load_entry_point('my-nav', 'console_scripts', 'my_nav_py_node')())
  File "/home/ubuntu/tb4_ws/build/my_nav/my_nav/my_nav_py_node.py", line 72, in main
    naviturt.waitUntilNav2Active(navigator='/robot2/bt_navigator', localizer='/robot2/amcl')
  File "/opt/ros/humble/lib/python3.10/site-packages/nav2_simple_commander/robot_navigator.py", line 316, in waitUntilNav2Active
    self._waitForNodeToActivate(navigator)
[... 157 lines cut ...]
[lifecycle_manager-8] [INFO] [1715949713.165744652] [robot2.lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-8] [INFO] [1715949713.215867395] [robot2.lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-8] [INFO] [1715949713.265998054] [robot2.lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[lifecycle_manager-8] [INFO] [1715949713.316118132] [robot2.lifecycle_manager_navigation]: Waiting for service controller_server/get_state...
[... cut ...]
[INFO] [lifecycle_manager-8]: sending signal 'SIGKILL' to process[lifecycle_manager-8]
[ERROR] [lifecycle_manager-8]: process has died [pid 22624, exit code -9, cmd '/opt/ros/humble/lib/nav2_lifecycle_manager/lifecycle_manager --ros-args --log-level info --ros-args -r __node:=lifecycle_manager_navigation -r __ns:=/robot2 --params-file /tmp/launch_params_r341kn8k --params-file /tmp/launch_params_ibsnqp1y --params-file /tmp/launch_params_9r493_pm -r /robot2/global_costmap/scan:=/robot2/scan -r /robot2/local_costmap/scan:=/robot2/scan'].
```

Thanks for the help!

WallerJonathan, 2024-05-17:
I did not mean to close this, that is my bad!

SteveMacenski, 2024-05-17:
I can't comment on the TB4 stuff - but did you try this with our multirobot bringup? The TB4 package might not be working properly... What's your `ros2 topic list` look like - is everything properly namespaced and nothing in the root namespace? 

>  Waiting for service controller_server/get_state...

This is making me think its just another instance of the same problem.

WallerJonathan, 2024-05-20:
Running `ros2 topic list | nl` gives me a list of topics that has what I believe I need:

```
ubuntu@unity:~$ ros2 topic list | nl
     1	/clock
     2	/diagnostics
     3	/parameter_events
     4	/robot1/amcl/transition_event
     5	/robot1/amcl_pose
     6	/robot1/battery_state
     7	/robot1/behavior_server/transition_event
     8	/robot1/behavior_tree_log
     9	/robot1/bond
    10	/robot1/bt_navigator/transition_event
    11	/robot1/bumper_contact
    12	/robot1/clicked_point
    13	/robot1/cmd_audio
    14	/robot1/cmd_lightring
    15	/robot1/cmd_vel
    16	/robot1/cmd_vel_nav
    17	/robot1/cmd_vel_teleop
    18	/robot1/controller_server/transition_event
    19	/robot1/cost_cloud
    20	/robot1/diffdrive_controller/cmd_vel_unstamped
    21	/robot1/diffdrive_controller/transition_event
    22	/robot1/dock_status
    23	/robot1/downsampled_costmap
    24	/robot1/downsampled_costmap_updates
    25	/robot1/dynamic_joint_states
    26	/robot1/evaluation
    27	/robot1/function_calls
    28	/robot1/global_costmap/costmap
    29	/robot1/global_costmap/costmap_raw
    30	/robot1/global_costmap/costmap_updates
    31	/robot1/global_costmap/footprint
    32	/robot1/global_costmap/global_costmap/transition_event
    33	/robot1/global_costmap/published_footprint
    34	/robot1/global_costmap/voxel_marked_cloud
    35	/robot1/goal_pose
    36	/robot1/hazard_detection
    37	/robot1/initialpose
    38	/robot1/interface_buttons
    39	/robot1/ip
    40	/robot1/ir_intensity
    41	/robot1/ir_opcode
    42	/robot1/joint_state_broadcaster/transition_event
    43	/robot1/joint_states
    44	/robot1/joy
    45	/robot1/kidnap_status
    46	/robot1/local_costmap/clearing_endpoints
    47	/robot1/local_costmap/costmap
    48	/robot1/local_costmap/costmap_raw
    49	/robot1/local_costmap/costmap_updates
    50	/robot1/local_costmap/footprint
    51	/robot1/local_costmap/local_costmap/transition_event
    52	/robot1/local_costmap/published_footprint
    53	/robot1/local_costmap/voxel_grid
    54	/robot1/local_costmap/voxel_marked_cloud
    55	/robot1/local_plan
    56	/robot1/map
    57	/robot1/map_server/transition_event
    58	/robot1/map_updates
    59	/robot1/marker
    60	/robot1/mobile_base/sensors/bumper_pointcloud
    61	/robot1/mouse
    62	/robot1/oakd/rgb/preview/camera_info
    63	/robot1/oakd/rgb/preview/depth
    64	/robot1/oakd/rgb/preview/depth/points
    65	/robot1/oakd/rgb/preview/image_raw
    66	/robot1/odom
    67	/robot1/particle_cloud
    68	/robot1/plan
    69	/robot1/plan_smoothed
    70	/robot1/planner_server/transition_event
    71	/robot1/preempt_teleop
    72	/robot1/received_global_plan
    73	/robot1/robot_description
    74	/robot1/scan
    75	/robot1/sim_ground_truth_dock_pose
    76	/robot1/sim_ground_truth_pose
    77	/robot1/slip_status
    78	/robot1/smoother_server/transition_event
    79	/robot1/speed_limit
    80	/robot1/standard_dock_description
    81	/robot1/stop_status
    82	/robot1/tf
    83	/robot1/tf_static
    84	/robot1/transformed_global_plan
    85	/robot1/velocity_smoother/transition_event
    86	/robot1/waypoint_follower/transition_event
    87	/robot1/waypoints
    88	/robot1/wheel_status
    89	/robot1/wheel_ticks
    90	/robot1/wheel_vels
    91	/robot2/amcl/transition_event
    92	/robot2/amcl_pose
    93	/robot2/battery_state
    94	/robot2/behavior_server/transition_event
    95	/robot2/bond
    96	/robot2/bt_navigator/transition_event
    97	/robot2/bumper_contact
    98	/robot2/clicked_point
    99	/robot2/cmd_audio
   100	/robot2/cmd_lightring
   101	/robot2/cmd_vel
   102	/robot2/controller_server/transition_event
   103	/robot2/diffdrive_controller/cmd_vel_unstamped
   104	/robot2/diffdrive_controller/transition_event
   105	/robot2/dock_status
   106	/robot2/downsampled_costmap
   107	/robot2/downsampled_costmap_updates
   108	/robot2/dynamic_joint_states
   109	/robot2/function_calls
   110	/robot2/global_costmap/costmap
   111	/robot2/global_costmap/costmap_updates
   112	/robot2/global_costmap/global_costmap/transition_event
   113	/robot2/global_costmap/voxel_marked_cloud
   114	/robot2/hazard_detection
   115	/robot2/initialpose
   116	/robot2/interface_buttons
   117	/robot2/ip
   118	/robot2/ir_intensity
   119	/robot2/ir_opcode
   120	/robot2/joint_states
   121	/robot2/joy
   122	/robot2/kidnap_status
   123	/robot2/local_costmap/costmap
   124	/robot2/local_costmap/costmap_updates
   125	/robot2/local_costmap/local_costmap/transition_event
   126	/robot2/local_costmap/published_footprint
   127	/robot2/local_costmap/voxel_marked_cloud
   128	/robot2/local_plan
   129	/robot2/map
   130	/robot2/map_server/transition_event
   131	/robot2/map_updates
   132	/robot2/mobile_base/sensors/bumper_pointcloud
   133	/robot2/mouse
   134	/robot2/odom
   135	/robot2/particle_cloud
   136	/robot2/plan
   137	/robot2/planner_server/transition_event
   138	/robot2/robot_description
   139	/robot2/scan
   140	/robot2/sim_ground_truth_dock_pose
   141	/robot2/sim_ground_truth_pose
   142	/robot2/slip_status
   143	/robot2/smoother_server/transition_event
   144	/robot2/standard_dock_description
   145	/robot2/stop_status
   146	/robot2/tf
   147	/robot2/tf_static
   148	/robot2/waypoint_follower/transition_event
   149	/robot2/waypoints
   150	/robot2/wheel_status
   151	/robot2/wheel_ticks
   152	/robot2/wheel_vels
   153	/rosout
```

In addition, when I run `ros2 service list` I can see the services active (shortened so it is easy to read):

```
/robot1/bt_navigator/get_state
/robot1/controller_server/get_state
/robot2/bt_navigator/get_state
/robot2/bt_navigator/get_state
```

I agree that it is probably something to do with Turtlebot4 and not as much NAV2. An interesting thing is that about ten or so topics are not seen in /robot2 including `16	/robot1/cmd_vel_nav`; could this possibly be part of the problem?

Again, thanks for the help.

SteveMacenski, 2024-05-20:
In that case, this line: 

> [robot2.lifecycle_manager_navigation]: Waiting for service controller_server/get_state...

Points to the likely issue being that their eq of https://github.com/ros-navigation/navigation2/blob/main/nav2_bringup/launch/navigation_launch.py#L43 isn't respecting namespaces. Its looking for `controller_server/get_state`, not `<robot>/controller_server/get_state`.

I think you should file this with the TB4 folks. The namespace issue in the `wait until active` seems like a good direct contribution back to Nav2 however, if you wanted to open a PR to add using `self.namespace`!

WallerJonathan, 2024-05-29:
A few updates so far:

It seems that the TB4 team has other work at the moment, and they are not able to answer the tickets.

I decided to look into this more myself as the project I am working on requires multiple NAV2 modules working at once. I found that using the bringup_launch.py from the nav2_bringup package allows me to bring up multiple modules at the same time! It seems their launch file may have some issues, although I am not sure what.

The problem I have now run into is that the local costmap appears blank from Nav2. It seems to be publishing a blank costmap. I checked through https://github.com/ros-navigation/navigation2/issues/3034, but their solution does not seem to work in my case. I also tried changing the parameter file to one that works with TB4, but that doesn't seem to change anything.

Nav2 still works on both robots, it is just missing the costmap which means the robots can't see dynamic obstacles. I am not sure what code/terminal outputs you might need, so just let me know and I will get back to you.

As always, thanks for your help!

SteveMacenski, 2024-05-29:
Great, I think that means we can close this? The issues are with the TB4 launch files and our stuff here works. Happy to respond here though if things come up!

closed 2024-05-29 by SteveMacenski