ros-navigation/navigation2 #4351: Error: looking up target frame: Lookup would require extrapolation into the past.

Razzi86, 2024-05-16:
## Bug report

**Required Info:**

- Operating System:
  Ubuntu 22.04
- ROS2 Version:
  - Ros2 Humble, apt install
- Version or commit hash:
  - ros-humble-navigation2	1.1.14-1jammy.20240407.041637
- DDS implementation:
  - rmw_fastrtps_cpp

#### Expected behavior
I'm running nav2 with a four wheel robot I built. I used a depth camera to make a 3d map using SLAM in rtab-map, then I used the heights of the points to determine the costs of the costmap. My costmap is designed so that everywhere there is a -1, replace it with a 0. When I select a 2D Goal Pose for the robot it determines a valid path and starts it, but in the middle of the path it stops and spins, and gives an error I've tried to troubleshoot but seem to be going down an endless rabbit hole. (Nav2 works properly once every 20 times I set a path)

What stands out to me in the error, i'm most concerned about the first one:
```
[planner_server-15] [ERROR] [1715892386.509665059] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the past.  Requested time 274.088000 but the earliest data is at time 275.472000, when looking up transform from frame [odom] to frame [map]
```
```
[bt_navigator-17] [ERROR] [1715892386.681814539] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown goal response, ignoring...
[controller_server-13] [INFO] [1715892386.762765270] [controller_server]: Passing new path to controller.
[controller_server-23] [INFO] [1715892386.762862315] [controller_server]: Passing new path to controller.
[controller_server-23] [ERROR] [1715892386.763046115] [controller_server]: Invalid path, Path is empty.
[controller_server-23] [WARN] [1715892386.763086627] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[controller_server-13] [ERROR] [1715892386.762970324] [controller_server]: Invalid path, Path is empty.
[controller_server-13] [WARN] [1715892386.763018383] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[planner_server-25] [INFO] [1715892386.781807607] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
```
```[bt_navigator-17] [WARN] [1715892386.903302498] [BehaviorTreeEngine]: Behavior Tree tick rate 50.00 was exceeded!
```
```[behavior_server-16] [INFO] [1715892387.022015769] [behavior_server]: Running spin
[behavior_server-26] [INFO] [1715892387.022013814] [behavior_server]: Running spin
[behavior_server-16] [INFO] [1715892387.022117480] [behavior_server]: Turning 1.57 for spin behavior.
[behavior_server-26] [INFO] [1715892387.022117511] [behavior_server]: Turning 1.57 for spin behavior.
```

Full error:
```[planner_server-15] [ERROR] [1715892386.509665059] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the past.  Requested time 274.088000 but the earliest data is at time 275.472000, when looking up transform from frame [odom] to frame [map]
[planner_server-15] 
[planner_server-15] [WARN] [1715892386.509687193] [planner_server]: Could not transform the start or goal pose in the costmap frame
[planner_server-15] [WARN] [1715892386.509713069] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[bt_navigator-27] [ERROR] [1715892386.541655619] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown result response, ignoring...
[controller_server-13] [INFO] [1715892386.562068921] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[controller_server-23] [INFO] [1715892386.562087721] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[bt_navigator-17] [ERROR] [1715892386.562262400] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown result response, ignoring...
[controller_server-13] [INFO] [1715892386.562422479] [controller_server]: Received a goal, begin computing control effort.
[controller_server-23] [INFO] [1715892386.562519482] [controller_server]: Received a goal, begin computing control effort.
[rtabmap-4] [INFO] [1715892386.586850915] [rtabmap]: rtabmap (115): Rate=0.00s, Limit=0.000s, Conversion=0.0003s, RTAB-Map=0.0366s, Maps update=0.0000s pub=0.0005s (local map=0, WM=0)
[bt_navigator-17] [ERROR] [1715892386.601917320] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown goal response, ignoring...
[planner_server-25] [INFO] [1715892386.642010511] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-15] [INFO] [1715892386.642017687] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[bt_navigator-17] [ERROR] [1715892386.642231097] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown result response, ignoring...
[bt_navigator-17] [ERROR] [1715892386.642417881] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown goal response, ignoring...
[bt_navigator-17] [ERROR] [1715892386.681814539] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown goal response, ignoring...
[controller_server-13] [INFO] [1715892386.762765270] [controller_server]: Passing new path to controller.
[controller_server-23] [INFO] [1715892386.762862315] [controller_server]: Passing new path to controller.
[controller_server-23] [ERROR] [1715892386.763046115] [controller_server]: Invalid path, Path is empty.
[controller_server-23] [WARN] [1715892386.763086627] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[controller_server-13] [ERROR] [1715892386.762970324] [controller_server]: Invalid path, Path is empty.
[controller_server-13] [WARN] [1715892386.763018383] [controller_server]: [follow_path] [ActionServer] Aborting handle.
[planner_server-25] [INFO] [1715892386.781807607] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-15] [INFO] [1715892386.781842609] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[bt_navigator-27] [ERROR] [1715892386.782101042] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown result response, ignoring...
[rtabmap-4] [INFO] [1715892386.787149774] [rtabmap]: rtabmap (116): Rate=0.00s, Limit=0.000s, Conversion=0.0005s, RTAB-Map=0.0334s, Maps update=0.0000s pub=0.0005s (local map=0, WM=0)
[bt_navigator-27] [ERROR] [1715892386.821657018] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown goal response, ignoring...
[bt_navigator-17] [ERROR] [1715892386.861843676] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown result response, ignoring...
[bt_navigator-17] [ERROR] [1715892386.882202505] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown cancel response, ignoring...
[bt_navigator-17] [ERROR] [1715892386.902261923] [bt_navigator_navigate_to_pose_rclcpp_node]: Failed to get result for compute_path_to_pose in node halt!
[controller_server-23] [INFO] [1715892386.902405486] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[controller_server-13] [INFO] [1715892386.902411573] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[planner_server-25] [INFO] [1715892386.902637344] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-15] [INFO] [1715892386.902641688] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-25] [WARN] [1715892386.903094563] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-15] [WARN] [1715892386.903127992] [planner_server]: [compute_path_to_pose] [ActionServer] Aborting handle.
[bt_navigator-17] [ERROR] [1715892386.903230755] [bt_navigator_navigate_to_pose_rclcpp_node.rclcpp_action]: unknown goal response, ignoring...
[bt_navigator-17] [WARN] [1715892386.903302498] [BehaviorTreeEngine]: Behavior Tree tick rate 50.00 was exceeded!
[rtabmap-4] [INFO] [1715892386.990738761] [rtabmap]: rtabmap (117): Rate=0.00s, Limit=0.000s, Conversion=0.0003s, RTAB-Map=0.0351s, Maps update=0.0000s pub=0.0007s (local map=0, WM=0)
[behavior_server-16] [INFO] [1715892387.022015769] [behavior_server]: Running spin
[behavior_server-26] [INFO] [1715892387.022013814] [behavior_server]: Running spin
[behavior_server-16] [INFO] [1715892387.022117480] [behavior_server]: Turning 1.57 for spin behavior.
[... cut ...]
[planner_server-15] [ERROR] [1715892387.103443263] [transformPoseInTargetFrame]: Extrapolation Error looking up target frame: Lookup would require extrapolation into the past.  Requested time 274.088000 but the earliest data is at time 276.065000, when looking up transform from frame [odom] to frame [map]

```

#### Actual behavior
The robot doesn't follow the full path, instead it stops and spins and seems to fail:
[image]

my nav2_params.yaml: https://github.com/Razzi86/ana_bot/blob/main/src/ana/config/nav2_params.yaml

closed 2024-05-16 by SteveMacenski

Razzi86, 2024-05-17:
Solved:

global_costmap's global_frame was /odom, and local_costmap's global_frame was /map. /map didn't publish anything (with my prototype setup). After changing /map to /odom it started working!

SteveMacenski, 2024-05-17:
Great! Glad it was comparatively an easy one!