ros-navigation/navigation2 #4355: [Collision_monitor] Time=0 step is not processed

BriceRenaudeau, 2024-05-20:
## Bug report

In the approach polygon process, the algorithm loops over `simulation_time_step` to find the collision time.
`time` start at 0.0 but the `projectState(simulation_time_step_, pose, vel);` is call directly therefor the time 0.0 polygon is not checked.

https://github.com/ros-navigation/navigation2/blob/bf291d704dbb1c2572575dd0620b12f816407233/nav2_collision_monitor/src/polygon.cpp#L243-L256

This leads to a possible flaw in the safety. If the `simulation_time_step` or the velocity is too big, close obstacles in the polygon are not detected and the robot is not stopped.

**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Iron
- Version or commit hash:
  - sources
- DDS implementation:
  - Cyclonne

#### Steps to reproduce issue
- Put a huge value in `simulation_time_step` like 1s
- Start a simulation with collision_monitor enabled
- Put an obstacle close to the robot
- Move toward the obstacle at full speed (teleop)

#### Expected behavior
The robot shouldn't move if there is an obstacle in its polygon

#### Actual behavior
The robot moves even if there is an obstacle in its polygon

#### Additional information
- We can add a check before the simulation loop.
- The `time` in the collision check actually starts at simulation_time_step_ and not 0.0.

tonynajjar, 2024-05-20:
This makes sense to me, I think the most straightforward fix would be to add something like

```
    if (getPointsInside(collision_points) >= min_points_) {
      return 0.0;
    }
```

at the beginning of `getCollisionTime`. What do you think?

BriceRenaudeau, 2024-05-20:
That's what I was thinking about. (plus init `time` at `simulation_time_step_`).

My only concern is that people may have used this bug as a feature to evade when the robot is touching a wall for example.

tonynajjar, 2024-05-20:
> My only concern is that people may have used this bug as a feature to evade when the robot is touching a wall for example.

This should have been implemented with https://github.com/ros-navigation/navigation2/issues/3313 afaik so I don't see the reason to keep this "feature" in that way

tonynajjar, 2024-05-20:
Actually, thinking about this again, I think there is no avoiding the collision monitor misbehaving if `simulation_time_step` is too big. Implementing what we just discussed would avoid moving if the robot is already in collision, but it would not avoid the collision if the robot is close to collision, `simulation_time_step` is big and the velocity sent is big enough.

It might still be worth it to check for collision at time=0 but essentially `simulation_time_step` still need to be small enough to avoid collision reliably

tonynajjar, 2024-05-20:
> A better solution is to make the projectState() AFTER the getPointsInside

Agree! Feel free to open a PR

SteveMacenski, 2024-05-20:
Seems rational, thanks for the thread on the European timezone. I have nothing further to add :-)

closed 2024-05-21 by SteveMacenski