ros-navigation/navigation2 #4359: Errors with **controller_server** using GPS for Navigation

arad2456, 2024-05-21:
#### Problem Description

I am working with ROS2 Humble and trying to implement a navigation algorithm that integrates GPS measurements. I successfully ran a simulation in Gazebo using the [nav2_gps_waypoint_follower_demo](https://github.com/ros-navigation/navigation2_tutorials/tree/master/nav2_gps_waypoint_follower_demo) with the turtlebot, after adjusting some configurations. When I tried to adapt this approach for my custom robot using a launch file, I encountered the following errors during the **nav2_bringup** process:
```
[controller_server-10] [ERROR] [1716328246.091536224] [controller_server]: Couldn't load critics! Caught exception: No critics defined for FollowPath
[controller_server-10] [ERROR] [1716328246.096419625] [controller_server]: Caught exception in callback for transition 10
[controller_server-10] [ERROR] [1716328246.096489476] [controller_server]: Original error: No critics defined for FollowPath
[controller_server-10] [WARN] [1716328246.096507160] [controller_server]: Error occurred while doing error handling.
[controller_server-10] [FATAL] [1716328246.096518289] [controller_server]: Lifecycle node controller_server does not have error state implemented
[lifecycle_manager-17] [ERROR] [1716328246.097113046] [lifecycle_manager_navigation]: Failed to change state for node: controller_server
[lifecycle_manager-17] [ERROR] [1716328246.097165270] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```

As indicated, the **nav2_bringup** process fails to launch, and thus, the navigation fails. I would appreciate any assistance you can provide. 

**Package Download**
[nav2_gps_waypoint_follower_demo.zip](https://github.com/ros-navigation/navigation2/files/15395929/nav2_gps_waypoint_follower_demo.zip)

For reproducibility, I have uploaded the package containing all necessary configuration files, including the problematic launch file, _run_nav2_gps.launch.py_, and the launch file from the documentation that worked, _gps_waypoint_follower.launch.py_.  In the file  _gps_waypoint_follower.launch.py_ you have to first change the topic of the map in rviz to a different topic, and only then the navigation works. 

#### Launch and Urdf Files
This is the launch file _gps_waypoint_follower.launch.py_:

```
# Copyright (c) 2018 Intel Corporation
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

import os

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.substitutions import LaunchConfiguration, Command
from launch.actions import IncludeLaunchDescription, DeclareLaunchArgument
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.conditions import IfCondition
from nav2_common.launch import RewrittenYaml
from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare

def generate_launch_description():
    # Get the launch directory
    bringup_dir = get_package_share_directory('nav2_bringup')
    gps_wpf_dir = get_package_share_directory("nav2_gps_waypoint_follower_demo")
    launch_dir = os.path.join(gps_wpf_dir, 'launch')
    params_dir = os.path.join(gps_wpf_dir, "config")
    nav2_params = os.path.join(params_dir, "nav2_no_map_params.yaml")
    configured_params = RewrittenYaml(
        source_file=nav2_params, root_key="", param_rewrites="", convert_types=True
    )

    pkg_share = get_package_share_directory('nav2_gps_waypoint_follower_demo')
    pkg_gazebo_ros = get_package_share_directory('gazebo_ros')
    path_lafette = os.path.join(pkg_share, 'urdf', 'Lafette_small_urdf.urdf')

    use_rviz = LaunchConfiguration('use_rviz')

    declare_use_rviz_cmd = DeclareLaunchArgument(
        'use_rviz',
[... 520 lines cut ...]
        </scan>
        <range>
          <min>0.120000</min>
          <max>3.5</max>
          <resolution>0.015000</resolution>
        </range>
        <noise>
          <type>gaussian</type>
          <mean>0.0</mean>
          <stddev>0.01</stddev>
        </noise>
      </ray>
      <plugin name="scan" filename="libgazebo_ros_ray_sensor.so">
        <ros>
          <remapping>~/out:=scan</remapping>
        </ros>
        <output_type>sensor_msgs/LaserScan</output_type>
        <frame_name>lidar_link</frame_name>
      </plugin>
    </sensor>
  </gazebo>
  

</robot>
```

SteveMacenski, 2024-05-21:
Please ask on robotics stack exchange but your error seems clear to me from your logs'

> [controller_server-10] [ERROR] [1716328246.096489476] [controller_server]: Original error: No critics defined for FollowPath

You need to pass in parameters. This is likely an issue with your launch file.

closed 2024-05-21 by SteveMacenski

arad2456, 2024-05-21:
> Please ask on robotics stack exchange but your error seems clear to me from your logs'
> 
> > [controller_server-10] [ERROR] [1716328246.096489476] [controller_server]: Original error: No critics defined for FollowPath
> 
> You need to pass in parameters. This is likely an issue with your launch file.

I tried on different occasions to ask three different questions on Robotics Stack Exchange, but from my experience you rarely get an answer there if you ask a question which is not completly basic. I know this forum is for problems with the package, and I understand that you don't have or need to help me with the problem in my code. Yet, I would be extremely thankful if you could help me here. I tried to pass in parameters which also contain the critiques. This is the calling for **nav2_bringup** from the launch file:
```
    navigation2_cmd = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(os.path.join(bringup_dir, "launch", "navigation_launch.py")),
        launch_arguments={
            "use_sim_time": "True",
            "params_file": configured_params,
            "autostart": "True",
        }.items(),
    )
```

And this is the part with the critics in the configuration file when calling the controller_server:

```
...
controller_server:
  ros__parameters:
   ...
    FollowPath:
  ...
      critics: ["RotateToGoal", "Oscillation", "BaseObstacle", "GoalAlign", "PathAlign", "PathDist", "GoalDist"]
   
```

SteveMacenski, 2024-05-21:
Unfortunately this is not an alternative for Q&A, that's what robotics stack exchange (or ROS Answers back when) is for. This is an issue tracker for project bugs and issues, not to help for first time setup. We as maintainers do not have the availability to help with every user's unique problems in getting up and running the first time (which often just has us pointing back to our own `nav2_bringup` files or other projects using Nav2 as successful examples).

arad2456, 2024-05-21:
Ok, I understand, thank you anyway.