ros-navigation/navigation2 #4363: Vector polygon - Collision Monitor is not available for ROS2 humble

samarth-eic, 2024-05-23:
While implementing collision monitor for our AMR in ROS2 humble we were able to use collision monitor, but it has a drawback 

1. whenever there is obstacle in front of robot it just stop despite having enough space to change the path and move forward
2.  It stays in stopping condition till the recovery mode is not running.  Which is not allowing the robot to move with any of the valid path.
After reviewing the documentation and other source for solving this we found that vector polygon can help us with these drawbacks, but unfortunately there is not implementation of vector polygon on ROS2 humble. Is there anything which can be done to solve these issue or drawbacks, or is there any plan for giving implementation for humble.

tonynajjar, 2024-05-23:
> vector polygon

I assume you mean the Velocity polygon added [here](https://github.com/ros-navigation/navigation2/pull/3708)

Unfortunately it cannot be backported to Humble as per this [comment](https://github.com/ros-navigation/navigation2/issues/3313#issuecomment-2048526708). I haven't tried but I think it wouldn't be so hard for you to backport it in a fork.

samarth-eic, 2024-05-23:
@tonynajjar Thanks for your effort and time, for which ROS version does velocity polygons are available and also does backport it in a fork will work or not ?

tonynajjar, 2024-05-23:
> for which ROS version does velocity polygons are available

https://docs.nav2.org/migration/Iron.html#add-velocitypolygon-in-collision-monitor

> backport it in a fork will work or not

Likely but you'd need to try to be sure

closed 2024-05-23 by tonynajjar

MichaelT64, 2024-05-30:
Hi,
I had the same problem and made a fork from Main and tried to adopt it for humble with some CPP and Python expression magic: [https://github.com/MichaelT64/navigation2/commits/rolling-humble/](https://github.com/MichaelT64/navigation2/commits/rolling-humble/)
The CollisionMonitor works for me, due to lack  of time and knowledge I was not able to make deeper tests of the navigation itself for now.
Maybe it helps you. 

Michael

samarth-eic, 2024-05-30:
Thanks for your effort, we were still stuck there maybe we can check and let you know :)