ros-navigation/navigation2 #4370: [for help] `navigateThroughPoses` doesn't work (humble)

GoesM, 2024-05-27:
## Bug report

**Required Info:**

- Operating System:
  -  ubuntu 22.04
- ROS2 Version:
  -  humble
- Version or commit hash:
  -   the latest
- DDS implementation:
  -  defaulted

#### Steps to reproduce issue

my command: 
```
ros2 action send_goal /follow_path nav2_msgs/action/FollowPath "
controller_id: 'FollowPath'
goal_checker_id: 'general_goal_checker'
path:
  header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: 'map'
  poses:
    - header:
        frame_id: 'map'
      pose:
        position:
          x: -2.0
          y: 1.5
          z: 0.0
        orientation:
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
    - header:
        frame_id: 'map'
      pose:
        position:
          x: 1.0
          y: -1.5
          z: 0.0
        orientation:
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
    - header:
        frame_id: 'map'
      pose:
        position:
          x: -1.0
          y: 2.0
          z: 0.0
        orientation:
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0 "
```

#### Expected behavior

#### Actual behavior
the navigation2 regard my path as an empty path (following it log): 
```
[controller_server-9] [INFO] [1716797013.770443479] [controller_server]: Received a goal, begin computing control effort.
[controller_server-9] [WARN] [1716797013.770569341] [controller_server]: Resulting plan has 0 poses in it.
[controller_server-9] [WARN] [1716797013.820611688] [controller_server]: Resulting plan has 0 poses in it.
[controller_server-9] [WARN] [1716797013.870615594] [controller_server]: Resulting plan has 0 poses in it.
[controller_server-9] [WARN] [1716797013.920622037] [controller_server]: Received plan with zero length
[controller_server-9] [WARN] [1716797013.970616956] [controller_server]: Received plan with zero length
[controller_server-9] [WARN] [1716797014.020608404] [controller_server]: Received plan with zero length
[controller_server-9] [WARN] [1716797014.070613054] [controller_server]: Received plan with zero length
[controller_server-9] [WARN] [1716797014.120619206] [controller_server]: Received plan with zero length
[controller_server-9] [ERROR] [1716797014.120640876] [controller_server]: Controller patience exceeded
[controller_server-9] [WARN] [1716797014.120653814] [controller_server]: [follow_path] [ActionServer] Aborting handle.
```

#### Additional information

I'm not sure whether my command was wrong or something went wrong in navigation2.

SteveMacenski, 2024-05-28:
> I'm not sure whether my command was wrong

I'm pretty sure its your command since that is the same API that is used by the BT and the Python3 API to send the paths to the controller server over the action interface. 

I see there's a trailing space after your last orientation - try removing that? Maybe there's a problem in serializing your path and the CLI clips the entire field that failed to serialize (i.e. the poses list) and thus is empty?

padhupradheep, 2024-05-29:
I ran the same command in the terminal and tested on our robot, seems to work fine.. 

@GoesM could you please share your setup and some information on how to reproduce the issue?

GoesM, 2024-05-29:
### my steps to reproduce the issue

I setup my navigation2-humble by following steps:
#### install 
```sh
cd ros_ws
git clone https://github.com/ros-navigation/navigation2.git -b humble
```
#### setup
```sh
export CC=/usr/bin/clang
export CXX=/usr/bin/clang++
colcon build \
	--cmake-clean-cache \
	--cmake-args \
		-DBUILD_TESTING=OFF \
		-DCMAKE_CXX_FLAGS="${CMAKE_CXX_FLAGS} \
			-w -Wno-error -Wno-everything" \
		-DCMAKE_C_FLAGS="${CMAKE_C_FLAGS}  \
			-w -Wno-error -Wno-format-security"

```
#### launch
```sh
export ROS_LOG_DIR="ros_log"
source install/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=False use_rviz:=True use_composition:=False
```

#### send goal
```sh
ros2 action send_goal /follow_path nav2_msgs/action/FollowPath "
controller_id: 'FollowPath'
goal_checker_id: 'general_goal_checker'
path:
  header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: 'map'
  poses:
    - header:
        frame_id: 'map'
      pose:
        position:
          x: -2.0
          y: 1.5
          z: 0.0
        orientation:
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
    - header:
        frame_id: 'map'
      pose:
        position:
          x: 1.0
          y: -1.5
          z: 0.0
        orientation:
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
    - header:
        frame_id: 'map'
      pose:
        position:
          x: -1.0
          y: 2.0
          z: 0.0
        orientation:
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0 "
```

GoesM, 2024-05-29:
@padhupradheep thanks for your focus on the issue.  ^_^ 

> I ran the same command in the terminal and tested on our robot, seems to work fine..

uh huh , i failed all the time in my environment..   Did you test also within `humble` branch ?

SteveMacenski, 2024-05-29:
I see this too, but I can't explain it. If its not happening in the Python or BT which uses the Python and C++ Action APIs respectively, I expect its an issue with the CLI parsing (or how its populated?) and not something in Nav2. I also see this in Iron/Rolling on 22.04 (not rolling 24.04, can't test that yet). I know the CLI populating is really really finnicky, so it wouldn't surprise me if there's something just not quite formatted as it expects that is causing the issue. 

I usually do the ` ros2 action send_goal /server msg "` and *tab tab* to complete the message template and then fill that in since I know it can be a problem. That looks different from yours, so worth a try

padhupradheep, 2024-05-30:
Okay, I noticed that the coordinates that @GoesM is giving is outside the map. That's the reason why we have a resulting plan with 0 poses. 

@GoesM: I hope you are not using anyother special world. I tried the following coordinates, and it worked fine for me. Only thing is I couldn't visualize the global plan for some weird reason. 

```
ros2 action send_goal /follow_path nav2_msgs/action/FollowPath "
controller_id: 'FollowPath'
goal_checker_id: 'general_goal_checker'
path:
  header:
    stamp:
      sec: 0
      nanosec: 0
    frame_id: 'map'
  poses:
    - header:
        frame_id: 'map'
      pose:
        position:
          x: -1.73
          y: -0.523
          z: 0.0
        orientation:
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
    - header:
        frame_id: 'map'
      pose:
        position:
          x: -1.18
          y: -0.59
          z: 0.0
        orientation:
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0
    - header:
        frame_id: 'map'
      pose:
        position:
          x: -0.60
          y: -0.519
          z: 0.0
        orientation:
          x: 0.0
          y: 0.0
          z: 0.0
          w: 1.0 "
```

Note that, I have all the packages installed as binaries and not built from source on 22.04

GoesM, 2024-05-30:
@padhupradheep Thanks a lot ! You're right ! It's because  some poses are outside the map so that nav2 drop it.

Greatly thanks for all of your reply. No problem about this issue anymore.

closed 2024-05-30 by GoesM