ros-navigation/navigation2 #4376: Understand MPPI Critics and parameters

EnzoGhisoni, 2024-05-29:
We are trying to use nav2 with the MPPI plugin as a local trajectory planner in our controller_server. The robot is a differential robot running ROS2 Humble on Ubuntu 22.04 and the version 1.1.14 of nav2_mppi_controller. 
By reading the [readme.md](https://github.com/ros-navigation/navigation2/blob/main/nav2_mppi_controller/README.md) from the repository and the [configuration guide](https://docs.nav2.org/configuration/packages/configuring-mppic.html) from the documentation, we still have difficulties to understand multiple points in order to tune our MPPI:
	- What are the difference between Path Align Critic and Path Angle Critic ?
	- What are their impact on the controller policy?

The understanding of offset_from_furthest parameter used by different critics remains also unclear. Does the parameter have the same function in the different critics ?
	- In PathAlignCritic, it seems that if the reached path point is smaller than offset_from_furthest, these paths are not considered (nav2_mppi_controller/src/critics/path_align_critic.cpp line 58):
        ```
        if (path_segments_count < offset_from_furthest_) {
           return;
        }
        ```
	- In PathAngleCritics and PathAlignCritics, the parameter is used to compute offseted_idx which will be used to select dynamic obstacles on path.
Thanks in advance!

SteveMacenski, 2024-05-29:
> What are the difference between Path Align Critic and Path Angle Critic ?

They are a bit subtle so this I think is a fair question. `PathAlign` aligns the robot to the path by looking at the integrated distance between the reference path and the trajectory and scoring negatively for the sum total discrepancy. Thus, it tries to align the trajectories with the reference path. `PathAngle` instead looks at the angle wrt the robot and a point on the path some `N` points ahead and scores negatively if the angle is 'large'. This has the benefit of additionally scoring highly when you're doing a sharp turn or some other similar move that is difficult to achieve requiring some penalization for significant movement angularly that is out of the existing optimal trajectory's distribution. 

Really, the PathAlign is doing all the work, but the PathAngle really helps you get out of some sticky situations where you're doing a sharp turn or otherwise are misaligned and need to get back on the right path. Its there to cover edge cases primarily and when you remove it, you'll see failure rates / total task times increase as it tries to do some soft optimizer resets to sample trajectories from another distribution. 

> The understanding of offset_from_furthest parameter used by different critics remains also unclear. Does the parameter have the same function in the different critics ?

No, but they all have the same meaning of picking a point offset from the furthest reference path point obtained by any trajectory in the batch. What they do with that point can be different. `PathAlign` "passes" when we're moving slow w.r.t. our expected speed so that we don't try to over optimize for exact alignment while getting started tracking a path which creates a more natural behavior. Some of the critics have "pass" conditions where its best they not be considered in that context for natural / fluid behavior

EnzoGhisoni, 2024-06-03:
Thanks for your explanation and your time.