ros-navigation/navigation2 #4379: set dynamic param `z_rand` of `nav2_amcl` would lead to a UAF bug

GoesM, 2024-05-30:
## Bug report

**Required Info:**

- Operating System:
  -  Ubuntu22.04
- ROS2 Version:
  -  humble
- Version or commit hash:
  -  the latest
- DDS implementation:
  -  defaulted

#### Steps to reproduce issue

Launch the navigation2 normally, as following steps:
```sh
#!/bin/bash
export ASAN_OPTIONS=halt_on_error=0:new_delete_type_mismatch=0:detect_leaks=0:log_pah=asan
source install/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/humble/share/turtlebot3_gazebo/models
ros2 launch nav2_bringup tb3_simulation_launch.py headless:=True use_rviz:=False use_composition:=False 
```

Learning about how the dynamic-parameter works , I had a try on it . 
```sh
ros2 param set /amcl z_rand 0.5
```

[notice] whatever the value be (0.5 or anyother double value43) , the UAF occurs all the time. 

#### Expected behavior
no crash occurs

#### Actual behavior
The ASAN reporting a **heap-user-after-free** bug to me as following, and the `nav2_amcl` stop its work.

```
=================================================================
==145868==ERROR: AddressSanitizer: heap-use-after-free on address 0x6170000aaab0 at pc 0x730fff32b5a4 bp 0x7fff30eba470 sp 0x7fff30eba468
READ of size 8 at 0x6170000aaab0 thread T0
    #0 0x730fff32b5a3 in message_filters::Signal1<sensor_msgs::msg::LaserScan_<std::allocator<void> > >::removeCallback(std::shared_ptr<message_filters::CallbackHelper1<sensor_msgs::msg::LaserScan_<std::allocator<void> > > > const&) (/home/***/nav2_humble/install/nav2_amcl/lib/libamcl_core.so+0x52b5a3) (BuildId: e96a3822dfb41386b5409b544e4bc74088297548)
    #1 0x730fff32e80a in std::_Function_handler<void (), std::_Bind<void (message_filters::Signal1<sensor_msgs::msg::LaserScan_<std::allocator<void> > >::* (message_filters::Signal1<sensor_msgs::msg::LaserScan_<std::allocator<void> > >*, std::shared_ptr<message_filters::CallbackHelper1<sensor_msgs::msg::LaserScan_<std::allocator<void> > > >))(std::shared_ptr<message_filters::CallbackHelper1<sensor_msgs::msg::LaserScan_<std::allocator<void> > > > const&)> >::_M_invoke(std::_Any_data const&) (/home/***/nav2_humble/install/nav2_amcl/lib/libamcl_core.so+0x52e80a) (BuildId: e96a3822dfb41386b5409b544e4bc74088297548)
    #2 0x730fff3273c0 in tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>::~MessageFilter() (/home/***/nav2_humble/install/nav2_amcl/lib/libamcl_core.so+0x5273c0) (BuildId: e96a3822dfb41386b5409b544e4bc74088297548)
    #3 0x730fff3281c0 in tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>::~MessageFilter() (/home/***/nav2_humble/install/nav2_amcl/lib/libamcl_core.so+0x5281c0) (BuildId: e96a3822dfb41386b5409b544e4bc74088297548)
    #4 0x730fff1c36e4 in std::__uniq_ptr_impl<tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>, std::default_delete<tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer> > >::reset(tf2_ros::MessageFilter<sensor_msgs::msg::LaserScan_<std::allocator<void> >, tf2_ros::Buffer>*) (/home/***/nav2_humble/install/nav2_amcl/lib/libamcl_core.so+0x3c36e4) (BuildId: e96a3822dfb41386b5409b544e4bc74088297548)
    #5 0x730fff156ee9 in nav2_amcl::AmclNode::initMessageFilters() (/home/***/nav2_humble/install/nav2_amcl/lib/libamcl_core.so+0x356ee9) (BuildId: e96a3822dfb41386b5409b544e4bc74088297548)
    #6 0x730fff163f1e in nav2_amcl::AmclNode::dynamicParametersCallback(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >) (/home/***/nav2_humble/install/nav2_amcl/lib/libamcl_core.so+0x363f1e) (BuildId: e96a3822dfb41386b5409b544e4bc74088297548)
    #7 0x730fff23743e in std::__invoke_result<rcl_interfaces::msg::SetParametersResult_<std::allocator<void> > (nav2_amcl::AmclNode::*&)(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >), nav2_amcl::AmclNode*&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&>::type std::__invoke<rcl_interfaces::msg::SetParametersResult_<std::allocator<void> > (nav2_amcl::AmclNode::*&)(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >), nav2_amcl::AmclNode*&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&>(rcl_interfaces::msg::SetParametersResult_<std::allocator<void> > (nav2_amcl::AmclNode::*&)(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >), nav2_amcl::AmclNode*&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&) (/home/***/nav2_humble/install/nav2_amcl/lib/libamcl_core.so+0x43743e) (BuildId: e96a3822dfb41386b5409b544e4bc74088297548)
    #8 0x730fff2372a1 in rcl_interfaces::msg::SetParametersResult_<std::allocator<void> > std::__invoke_impl<rcl_interfaces::msg::SetParametersResult_<std::allocator<void> >, std::_Bind<rcl_interfaces::msg::SetParametersResult_<std::allocator<void> > (nav2_amcl::AmclNode::* (nav2_amcl::AmclNode*, std::_Placeholder<1>))(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >)>&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&>(std::__invoke_other, std::_Bind<rcl_interfaces::msg::SetParametersResult_<std::allocator<void> > (nav2_amcl::AmclNode::* (nav2_amcl::AmclNode*, std::_Placeholder<1>))(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >)>&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&) (/home/***/nav2_humble/install/nav2_amcl/lib/libamcl_core.so+0x4372a1) (BuildId: e96a3822dfb41386b5409b544e4bc74088297548)
    #9 0x730fff2370d3 in std::_Function_handler<rcl_interfaces::msg::SetParametersResult_<std::allocator<void> > (std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&), std::_Bind<rcl_interfaces::msg::SetParametersResult_<std::allocator<void> > (nav2_amcl::AmclNode::* (nav2_amcl::AmclNode*, std::_Placeholder<1>))(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> >)> >::_M_invoke(std::_Any_data const&, std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&) (/home/***/nav2_humble/install/nav2_amcl/lib/libamcl_core.so+0x4370d3) (BuildId: e96a3822dfb41386b5409b544e4bc74088297548)
    #10 0x7310002601a1  (/opt/ros/humble/lib/librclcpp.so+0x10e1a1) (BuildId: 4cca8a387f3c93d38a0567a8efc7cba9106f5d9a)
    #11 0x731000264cd1 in rclcpp::node_interfaces::NodeParameters::set_parameters_atomically(std::vector<rclcpp::Parameter, std::allocator<rclcpp::Parameter> > const&) (/opt/ros/humble/lib/librclcpp.so+0x112cd1) (BuildId: 4cca8a387f3c93d38a0567a8efc7cba9106f5d9a)
    #12 0x7310002ddafc  (/opt/ros/humble/lib/librclcpp.so+0x18bafc) (BuildId: 4cca8a387f3c93d38a0567a8efc7cba9106f5d9a)
    #13 0x73100029b107  (/opt/ros/humble/lib/librclcpp.so+0x149107) (BuildId: 4cca8a387f3c93d38a0567a8efc7cba9106f5d9a)
    #14 0x7310002a96ae  (/opt/ros/humble/lib/librclcpp.so+0x1576ae) (BuildId: 4cca8a387f3c93d38a0567a8efc7cba9106f5d9a)
    #15 0x73100023c2a5  (/opt/ros/humble/lib/librclcpp.so+0xea2a5) (BuildId: 4cca8a387f3c93d38a0567a8efc7cba9106f5d9a)
    #16 0x731000239c89 in rclcpp::Executor::execute_service(std::shared_ptr<rclcpp::ServiceBase>) (/opt/ros/humble/lib/librclcpp.so+0xe7c89) (BuildId: 4cca8a387f3c93d38a0567a8efc7cba9106f5d9a)
    #17 0x731000239ff5 in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&) (/opt/ros/humble/lib/librclcpp.so+0xe7ff5) (BuildId: 4cca8a387f3c93d38a0567a8efc7cba9106f5d9a)
    #18 0x7310002418af in rclcpp::executors::SingleThreadedExecutor::spin() (/opt/ros/humble/lib/librclcpp.so+0xef8af) (BuildId: 4cca8a387f3c93d38a0567a8efc7cba9106f5d9a)
    #19 0x731000241ac4 in rclcpp::spin(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>) (/opt/ros/humble/lib/librclcpp.so+0xefac4) (BuildId: 4cca8a387f3c93d38a0567a8efc7cba9106f5d9a)
    #20 0x5e3a5905f941 in main (/home/***/nav2_humble/install/nav2_amcl/lib/nav2_amcl/amcl+0xe6941) (BuildId: 068ea3ea0211b05161e789e94dd1668c91ef8430)
    #21 0x730ffe429d8f in __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16
    #22 0x730ffe429e3f in __libc_start_main csu/../csu/libc-start.c:392:3
    #23 0x5e3a58f9f504 in _start (/home/***/nav2_humble/install/nav2_amcl/lib/nav2_amcl/amcl+0x26504) (BuildId: 068ea3ea0211b05161e789e94dd1668c91ef8430)

0x6170000aaab0 is located 48 bytes inside of 680-byte region [0x6170000aaa80,0x6170000aad28)
freed by thread T0 here:
[... cut ...]
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==145868==ABORTING
```

#### Additional information

----

it seems that the callback function for laser_scan_message is stiil executing while dynamic parameter changes, and they have some data race.

SteveMacenski, 2024-05-30:
That appears to be the case! I think adding a mutex lock should square that up. The laser scan sub is in a different callback group on another executor https://github.com/ros-navigation/navigation2/blob/main/nav2_amcl/src/amcl_node.cpp#L1510-L1513 so they're not synchronized

GoesM, 2024-05-31:
well, I think it's a better way to fix it but not adding a mutex lock, because of following facts:

Let's notice that:

https://github.com/ros-navigation/navigation2/blob/30e2cde4997eec8664e8bd8b77caf717c61b1945/nav2_amcl/src/amcl_node.cpp#L1281-L1283

some dynamic parameter setting could let  `reinit_laser = true`, and then do following code lines:

https://github.com/ros-navigation/navigation2/blob/cf3dd553c2164c2a586f0f66628f22d43264d35f/nav2_amcl/src/amcl_node.cpp#L1355-L1364

However, we could find here's a lack of `laser_scan_filter_ .reset()` before `initMessageFilters()`.

As we know, the callback functions called by `lase_scan_filter_` would access the memory , which would be freed during `initMessageFilters()` , so that a use-after-free bug occurs.

SteveMacenski, 2024-05-31:
So you're suggesting that we reset the message filter at the start of the dynamic parameter callback and then re-initialize it at the end to kill the extra thread? I think that would work in practice, but I think the mutex is a bit cleaner and in line with what we've done elsewhere in the stack for consistency. 

Also constantly creating / destroying the data writer could uncover some DDS vendor implementation issues since that's really exercising the stability of their solution. I don't know that we'd find any bugs, but that would certainly be a good way to trigger them if there was some memory leaks. So in the back of my mind I'd rather not test it if we have an alternative that seems reasonable. The dynamic parameter callbacks are quick so I think this should be fine to mutex lock them :-)

GoesM, 2024-06-01:
> So you're suggesting that we reset the message filter at the start of the dynamic parameter callback and then re-initialize it at the end to kill the extra thread? 

I'm considering that  the current design approach of nav2_amcl seems to work indeed in this way.

After amcl receiving the instruction to change dynamic parameters, the related threads are closed : 

https://github.com/ros-navigation/navigation2/blob/cf3dd553c2164c2a586f0f66628f22d43264d35f/nav2_amcl/src/amcl_node.cpp#L1355-L1364

and then restarted within the new configuration by `initMessageFilters();`

https://github.com/ros-navigation/navigation2/blob/cf3dd553c2164c2a586f0f66628f22d43264d35f/nav2_amcl/src/amcl_node.cpp#L1497-L1517

***

Additionally, I've had a try on it , by adding the code `laser_scan_filter_.reset();` as following: 

```cpp
  // Re-initialize the lasers and it's filters
  if (reinit_laser) {
    lasers_.clear();
    lasers_update_.clear();
    frame_to_laser_.clear();
    laser_scan_connection_.disconnect();
    laser_scan_filter_.reset();
    laser_scan_sub_.reset();

    initMessageFilters();
  }
```

And then these commands for changing dynamic parameters would work correctly.

***

> So in the back of my mind I'd rather not test it if we have an alternative that seems reasonable. The dynamic parameter callbacks are quick so I think this should be fine to mutex lock them :-)

Sorry, I might not fully understand this method. Do you mean changing the current design, removing all the code that closes the threads, and instead using a locking method?  

I feel that I need to understand your suggestion more clearly, and then proceed with my future work based on your recommendation. ^_^

SteveMacenski, 2024-06-03:
What you say makes sense actually, lets do that!

closed 2024-06-05 by SteveMacenski