ros-navigation/navigation2 #4384: velocity smoother deadband velocity as an interval

T-Lemmel, 2024-05-31:
**Required Info:**

- Operating System:
  - Ubuntu 22.04
- ROS2 Version:
  - Humble
----
## Feature request

A deadband velocity parameter for the velocity smoother node that defines an interval in between velocities can't be, rather then minimums

#### Feature description

Hello, I am working on a real robot that has linear and angular velocities minimum, for the robot to move, i need :

- angular speeds ∈ [-1rad/s, -0.3rad/s] ∪ [0.3rad/s, 1rad/s] 
- linear speeds ∈ [-1.2m/s, -0.3m/s] ∪ [0.3m/s, 1.2m/s]

Outside of these values the robot doesn't move at all due to constructor code and safety reasons. But the cmd_vel i receive is most of the time out of these values (between -0.3 and 0.3). Since it's trying to have a smooth transition phase.

I tried using the velocity_smoother node with deadband_velocity: [0.3, 0.3, 0.3]

but this means all velocities under 0.3 rad/s or m/s will not be sent, doing so i can't ever have negative values. I would want to define a band between -0.3 and 0.3 that would be where velocities can't be.

Thanks a lot for the Help and the work on Ros/Ros2 in general

SteveMacenski, 2024-05-31:
You could use MPPI with the deadband critic to penalize actions outside of those ranges or create easily your own critics for DWB that do the same. 

However this is a bit of a specialized problem that may require a bit of a specialized solution with some leg work on your part. I think MPPI/DWB can do what you need, but you may benefit from something a bit simpler / different if you can't control the robot's speed continuously.

T-Lemmel, 2024-06-05:
I tried it and you were right, RPP makes it work really well !
The angular velocities threshold is actually low enough for it to work without oscillations. A bit of tuning is required but no need for homemade critics or a new feature.

Thanks a lot :+1:

T-Lemmel, 2024-06-11:
After a bit of testing i would actually need a more dynamic behavior. Ideally i would need the controller to be able to adapt the path to dynamic obstacles that are seen on the local costmap. 

From what i understand the RPP wont achieve that since it stricly follows the global plan. So i would want to try using MPPI/DWB with a custom critic to ensure velocities are within the appropriate ranges.

But there is something i don't understand, where is the local costmap used ? I thought the local_plan was an adapted version of the global_plan taking into account obstacles that were not seen when the path was computed. But i only see my local_costmap being subscribed to by the behavior_tree and never be the controller or the planner. 

Is this a configuration problem from me or am i misunderstanding how it works ?

SteveMacenski, 2024-06-11:
The local plan is typically just the segment of the global plan transformed and pruned into the local and odometry frames. The trajectory published by the algorithms for the velocity and/or time-series trajectory being executed will be published by various other debugging topics depending on the algorithm (i.e. RPP/DWB publish out its arcs; MPPI publishes out its trajectry). Its generally **bad** to subscribe to at least the MPPI one because the candidates are **huge** and the publication of all those samples, even when pruned, is alot of data and slows down the actual controller in visualization. Its only for debugging use.

If you need the dynamic behavior and large deadbands, I'd suggest looking at creating the deadband critics for DWB/MPPI depending on which you want to use. There may be other specialized algorithms that you should look into literature regarding. 

The isn't something unfortunately I can give you a recipe for "here's how to do it" or even a "here's how I would do it" that I would be confident will work very well. I'd personally need to experiment with a few ways of doing it, analyze where they are good/bad and then come to the final solution direction to refine. The MPPI critic is where I'd start since MPPI is very powerful and no matter what, a critic will be part of a MPPI-based solution, even if not itself sufficient

T-Lemmel, 2024-06-11:
Ok so the local_plan is actually just a portion of the global plan transformed and pruned ? But it does not take into account the local dynamic obstacles ? 
Then how to get the path being adapted to thoses ? I thought that was done be the local_planner in ros1 which was replaced be the controller in ros2.

Will those dynamic obstacles be used to compute the path by the MPPI using ObstaclesCritic ? How is it possible without the controller_server subscribing to the local_costmap ?

I'll make sure to keep you up to date to the solutions i find ! 

PS : Just to be sure i'm clear, i am using a global_costmap that only has the static obstacles ( inflation layer and static layer ) and a local_costmap that only has the dynamic ones (inflation layer and obstacle layer). 

The way i thought it works is this one : but it might be outdated/i misinterpreted

[image]

SteveMacenski, 2024-06-11:
> Ok so the local_plan is actually just a portion of the global plan transformed and pruned ? But it does not take into account the local dynamic obstacles ?

Correct. See the debugging topics for the specific algorithms that publish their computed trajectories as I mentioned above. 

> Will those dynamic obstacles be used to compute the path by the MPPI using ObstaclesCritic ? How is it possible without the controller_server subscribing to the local_costmap ?

See the MPPI docs, there are multiple options for collision avoidance critics. Obstacles are one, Cost is another. The local costmap is inside of the controller server.  

> PS : Just to be sure i'm clear, i am using a global_costmap that only has the static obstacles ( inflation layer and static layer ) and a local_costmap that only has the dynamic ones (inflation layer and obstacle layer).

Usually how I set it up too :-) 

BTW that diagram is ROS 1 and bares little resemblance to Nav2. See https://docs.nav2.org/

T-Lemmel, 2024-06-11:
Ok perfect, i think i get it !!

So the reason RPP was generating path through local dynamic obstacles is because RPP doesn't take them into account (hence "pure pursuit" ), but MPPI with cost/obstacles critic will ? 

Thanks a lot :-)

closed 2024-06-11 by SteveMacenski